arendst/Tasmota #8115: Wiki links are all dead

rolandow, 2020-04-10:
Last week I installed Tasmota on a smart plug. It's really disappointing (and annoying, to be honest) to see that all the wiki links are dead. Links found on google, but also references in github issues, all point to the same frontpage on github.

For example in ticket #2219 is ended with "See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.". Clicking it leads nowhere.

Is there not a way to fix this with redirects?

closed 2020-04-10 by close-issue-app[bot]

Jason2866, 2020-04-10:
If you look at the start page here you see https://tasmota.github.io/docs
No, old outdated wiki will NOT be reactivated. Reason Google will still reference when it is still there.

lalo-uy, 2020-04-10:
May be wipe all old wiki and put there a link to new docs.