arendst/Tasmota #8117: Moes Dimmer Switch (US model)

nj2299, 2020-04-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The dimmer range on this device is 150 to 1000 - as per feedback from the device.  When the dimmer range is set, the device does only change when sending the command Dimmer 75 and above.  Below Dimmer 75 (assuming this is percent) the dimmer stays at the lowest setting. Essentially the full dimming range is controlled between 75-100.  Second issue is that the LED stops responding to any status changes or commands from the console.   It does flash red when power cycled.  The White LED seems to stop responding.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X - but only open issues] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Moes Dimmer DSC01
- [ ] Tasmota binary firmware version number used:  8.2.0
  - [ X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tuya Convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

255
15:23:10 RSL: stat/tasmota_BA8FF1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:23:10 RSL: stat/tasmota_BA8FF1/RESULT = {"Module":{"54":"Tuya MCU"}}
15:23:11 RSL: stat/tasmota_BA8FF1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
15:23:29 RSL: tele/tasmota_BA8FF1/STATE = {"Time":"2020-04-10T15:23:29","Uptime":"3T13:55:21","UptimeSec":309321,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":68,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"CNN","BSSId":"C4:71:54:3D:97:90","Channel":11,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:16"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Setup device as TuyaMCU - Module (54)
- enter command in console - TuyaMCU 21,2
- enter command in console - dimmerrange 150,1000

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expect that the Dimmer command would be able to use the full 0-100 range
Expect that the white LED would respond to changes of state as set by the LEDState and LEDpower commands

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The device does connect to the network, and does respond to commands in the console.  Just not as expected.

**(Please, remember to close the issue when the problem has been addressed)**

nj2299, 2020-04-10:
I don’t think I filled the issue template for the check boxes properly.  Sorry about that.

effelle, 2020-04-10:
Please edit it and add the result of  `status 0` and show us what happens when you issue the command in console (set it to weblog 4).

nj2299, 2020-04-11:
Thank you for fixing the format.  I've checked the steps i've taken, and added the output from the status 0 with the weblog 4 set.

Jontron123, 2020-04-12:
Try setting LedTable 0 and see if that helps the dimmer command.

I assume you means a Moes DS01? I have a TreatLife DS01 which is similar to a Moes DS01 and had a similar issue with the dimming range. After turning off LedTable, the dimming range works as expected.

nj2299, 2020-04-14:
You are correct, it is the Moes DS01.  Setting the LedTable 0 fixed the issue with the range!  Thank you!!

Do you also have the issue with the white power LED not responding?

broken, 2020-04-17:
I'm having the same issue with the status light on my TreatLife DS01. It is worth noting that my switch on version 8.1.0 does not have this issue, only the switches on 8.2.0. Thus, I have my suspicion the problem is in the upgrade, but I cannot confirm. I don't want to upgrade the 8.1.0 switches, and I cannot downgrade the 8.2.0 switches even with SetOption78 disabled.

ascillato2, 2020-04-22:
Hi, several fixes has been merger. Please, can you re test with latest version from the development branch? ( thehackbox.org/tasmota ) thanks.

broken, 2020-04-22:
I just updated both of my switches to dev v8.2.0.4 and am still experiencing both issues. The status light does not turn on, and the Dimmer command does not have its full range. The one difference between @nj2299 and I is that my Dimmer command's lower limit is 50 and not 75. This was before and after the update.

nj2299, 2020-04-23:
I've just updated to 8.2.0.4 as well and still have the same issues as before.

closed 2020-05-24 by stale[bot]

philharlow, 2020-05-25:
It looks like I just missed the end of this ticket. Using the Treatlife Dimmer DS01, the ```ledtable 0``` command definitely helped the dimming function. Now It's just that white led issue. Can we compare changes between 8.1.0 and 8.2.0?

broken, 2020-06-24:
A similar issue has been reopend with #8577