arendst/Tasmota #8122: WiFi retry interval

wolfster-W3, 2020-04-10:
Device: sonoff Basic R3 runs Tasmota 8.2.0

**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
My Wi-Fi Access Point, the sonoff is connected to, will be regularly turned off for several days at a time. During this time, the device continuously attempts to reconnect to the Wi-Fi at an interval of about 9 seconds, whereby the temperature of the ESP 8285 chip increases by about 11°C.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Can the interval for the connection attempts be set to 30 minutes for example, when the AP is not available?

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Would appreciate any instructions on how I can change behaviour myself.

**Additional context**  
_Add any other context or screenshots about the feature request here._

wificonfig 5 (Wait)

Console output:
21:28:52 WIF: Connecting to AP1 TestW in mode 11N as tasmota...
21:29:01 WIF: Connect failed as AP cannot be reached
21:29:01 WIF: Connecting to AP1 TestW in mode 11N as tasmota...
21:29:10 WIF: Connect failed as AP cannot be reached
21:29:10 WIF: Connecting to AP1 TestW in mode 11N as tasmota...

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-10:
>Can the interval for the connection attempts be set to 30 minutes for example, when the AP is not available?

Yes, you can change in the code the following line:

https://github.com/arendst/Tasmota/blob/ac06abffe80fbf879371c9d1bd95f51a92f8aef8/tasmota/support_wifi.ino#L38

BUT... if you want to decrease the temperature, the increased waiting time won't work, because the arduino core and SDK (that Tasmota uses) can't use the SLEEP feature while not connected to wifi. All resources are put on wifi reconnection. So your device will wait more time but the temperature is not going to be lower.

You can use instead the **_deepsleep_** feature. When you are going to turn off your router, you can make a rule (triggered by a timer) to make your device going to deepsleep. (for deepsleep, see the docs.)

If you need further assistance with timers, rules or deepsleep, please, ask here or address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

wolfster-W3, 2020-04-11:
Changed WIFI_RETRY_OFFSET_SEC to 120 seconds for testing purposes, but no luck. It still tries to reconnect every 9 seconds, even though it was already set to 12 seconds in the original file.

ascillato2, 2020-04-13:
Hi, have you tried deepsleep? That will lower the temperature of the device as you want, but the device will be unavailable while it is under deepsleep.

When you are going to turn off your router, you can make a rule (triggered by a timer or by a ping command) to make your device going to deepsleep. (for deepsleep, see the docs.) If you need further assistance with timers, rules or deepsleep, please, ask here or address this to the Tasmota Support Chat.

closed 2020-04-13 by ascillato2