arendst/Tasmota #8123: Add Date to Log string (Console etc)

BBBits, 2020-04-10:
**Have you looked for this feature in other issues and in the docs?**  
briefly
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
no
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
prepend current log strings that already have timestamp to include Date 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
none
**Additional context**  
_Add any other context or screenshots about the feature request here._
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-11:
Tasmota already sends the full date and time stamp by MQTT.

In the console is not very useful due to the buffer is very small and you can see just last few entries. If you keep the brower open, you will see more but that is just your browser's cache, not Tasmota.

If you set a SYSLOG (as explained in the docs) you will have the length of logs that you need. The data in the syslog has already the date and time, and you can have there the logs of more than just one Tasmota device if you want.

BBBits, 2020-04-11:
I have NO interest in MQTT.
Why have ONLY a time stamp in the Log String ?

BBBits, 2020-04-11:
guess I have to live in my own world and leave my ridiculous thoughts to myself.

closed 2020-04-11 by BBBits

BBBits, 2020-04-11:
Not that you are interested  ,but It's configurable in my builds.
[image]

DittelHome, 2024-04-06:
Sorry, i know, the thread is 4 years old, but the problem is the same.
I use Tasmota 13.4.0 Sonoff Basic, the "prepend log with date ." checkbox . is missing.
Any help is welcome

barbudor, 2024-04-06:
That doesn't exist, so you won't find it anywhere

DittelHome, 2024-04-06:
Thanks for yor info.
OK, that would have been too nice...

sfromis, 2024-04-06:
You do have the date in the state payloads each TelePeriod. I can also recommend using `SetOption52` to have those date/time values include the time zone offset.

DittelHome, 2024-04-07:
Thanks sfromis, i have had disbaled the telemetrieperiode in the past, because i do not need this information via mqtt.
The log is significantly more inflated with these telemessages. Just a date stamp would be much more efficiently...