arendst/Tasmota #8124: Add On & Off To KNX Relay Event

BBBits, 2020-04-10:
**Have you looked for this feature in other issues and in the docs?**  
briefly
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
no
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
When using KNX for a Scene or a specific function , rules can work but adding an On & Off Makes it more versatile .
I looked at adding On/Off functionality to the KNX send but it was a lot of work.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
none
**Additional context**  
_Add any other context or screenshots about the feature request here._
you might also note I have changed the layout of the page so it will fit in portrait on my phone
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-10:
>Describe the solution you'd like
>A clear and concise description of what you want to happen.
>When using KNX for a Scene or a specific function , rules can work but adding an On & Off Makes it more versatile .

Sorry, I don't understand your request. Please, can you explain?

ascillato, 2020-04-10:
If what you want is to do a specific task no matter what you receive from your KNX network, you can use a rule ( https://tasmota.github.io/docs/KNX/#5-using-rules )

Set a group address in the **_group address to receive data from_** to KNX RX 1 (for example) and another address to KNX RX 2, and then make a rule like:

* If your KNX device is sending a **_command_**:

```lua
rule1 1
rule1 on event#knxrx_cmnd1 do power1 1 endon on event#knxrx_cmnd2 do power1 0 endon
```

* If your KNX device is sending a **_value_**:

```lua
rule1 1
rule1 on event#knxrx_val1 do power1 1 endon on event#knxrx_val2 do power1 0 endon
```

BBBits, 2020-04-10:
Hi, an Example .
Have 5 devices in a room & a few buttons at the door (or anywhere)
each button simply sends 3-3-[button number]
each device in the room can respond with an On or an Off to create a Scene for that button
you can probably achieve this with scripting etc but an On & Off in the Gui is Simple
Does that clarify what I am on about ?

ascillato, 2020-04-10:
If what you want is to do a specific task, no matter what you receive from your KNX network, you can use a rule ( https://tasmota.github.io/docs/KNX/#5-using-rules )

Set a group address in the **_group address to receive data from_** to KNX RX 1 (for example) and another address to KNX RX 2, and then make a rule like:

* If your KNX device is sending a **_command_**:

```lua
rule1 1
rule1 on event#knxrx_cmnd1 do power1 1 endon on event#knxrx_cmnd2 do power1 0 endon
```

* If your KNX device is sending a **_value_**:

```lua
rule1 1
rule1 on event#knxrx_val1 do power1 1 endon on event#knxrx_val2 do power1 0 endon
```

ascillato, 2020-04-11:
@BBBits 

This rule solves your use-case? The rule trigger `event#knxrx_cmnd1` was meant exactly for those cases.

closed 2020-04-11 by BBBits