arendst/Tasmota #8125: KNX logging problem

BBBits, 2020-04-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
While debugging my builds , noted that KNX logging was not working correctly.
problem is in KNX_CB_Action routine.
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Log Without Mod 
2020-04-11 13:37:37 KNX: Received from 5.1.90 Command Write: 
2020-04-11 13:37:37 RSL: RESULT = {"POWER":"ON"}
2020-04-11 13:37:37 RSL: POWER = ON
2020-04-11 13:38:26 KNX: Received from 5.1.90 Command Write:  to Output 1
2020-04-11 13:38:26 RSL: RESULT = {"POWER":"OFF"}
2020-04-11 13:38:26 RSL: POWER = OFF

Log with Mod
2020-04-11 13:43:59 KNX: Received from 5.1.90 Command Write: 1 to Output 1
2020-04-11 13:43:59 RSL: RESULT = {"POWER":"ON"}
2020-04-11 13:43:59 RSL: POWER = ON
2020-04-11 13:44:09 KNX: Received from 5.1.90 Command Write: 0 to Output 1
2020-04-11 13:44:09 RSL: RESULT = {"POWER":"OFF"}
2020-04-11 13:44:09 RSL: POWER = OFF

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
check log entries for knx reception

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
void KNX_CB_Action(message_t const &msg, void *arg)
{
  device_parameters_t *chan = (device_parameters_t *)arg;
  if (!(Settings.flag.knx_enabled)) { return; }
 
  char tempchar[33];

  if (msg.data_len == 1) {
    // COMMAND
    //BAZ
    sprintf(tempchar,"%d",msg.data[0]);
    //END BAZ
    /* Was Not Displaying the Message data with the code below
    tempchar[0] = msg.data[0]; // Could just add '0' but figured best be safe ( tempchar[0] = msg.data[0] + '0' )
    tempchar[1] = '\0';
    */
  }  else  {

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-11:
You code change is right. Do you want to provide a Pull Request?

BBBits, 2020-04-11:
no

ascillato, 2020-04-11:
Ok, I will do it later. Thanks for pointing this issue. :+1:

ascillato2, 2020-04-11:
Proposed PR with the fix done at #8126

Thanks :+1:

PR #8126 (2020-04-11): KNX: Fixed Console Log
## Description:

Fix console log.

Bug finded and solved by @BBBits

**Related issue (if applicable):** fixes #8125

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core Tasmota_core_stage
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2020-04-11 by ascillato2