arendst/Tasmota #8132: Tool for creating custom builds

MarkCiliaVincenti, 2020-04-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Creating custom builds is not clear enough

**Describe the solution you'd like**  
How possible would it be to have a tool with a checkbox list where one chooses the features they need, click a button and get a .bin file?

**Describe alternatives you've considered**  
N/A

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

MarkCiliaVincenti, 2020-04-11:
https://github.com/benzino77/tasmocompiler

closed 2020-04-11 by MarkCiliaVincenti

ascillato, 2020-04-11:
Duplicated issue.

Already answered in old issues and in your previous issue (https://github.com/arendst/Tasmota/issues/8131)

-----------------------------------------------------------

Your request already exists and it is explained in the docs.

Please search for TasmoCompiler in the docs.

Direct link to the compiler: https://github.com/benzino77/tasmocompiler