arendst/Tasmota #8134: Access Tasmota via Serial on a Multi-drop/half-duplex/RS485 type network - How to poll individual devices?

Cloolalang, 2020-04-11:
**Have you looked for this feature in other issues and in the docs?**  YES

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

WIFI/MQTT is great but limited in range, alternatives include Modbus & RS485 on cables or on RF.
For RF there are a few cheap UART radios (HC12, Ebyte Lora etc).

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

We can access Tasmota now with Serial which is a good alternative to WIFI/MQTT.
But to be able to have a system with more than one Tasmota device on the network with serial, we need a way to poll/address each device separately to moderate traffic.
ModBus does this by using device addresses and a master/slave setup.

If we can communicate on the serial using each command starting with the device topic, then this could serve as a selective device polling method. 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Modbus RTU

**Additional context**  
_Add any other context or screenshots about the feature request here._

Here is a typical Tasmota serial console session (using Termnite via USB port):
 (serial logging =2)

To Tasmota >>     Power
From Tasmota<<  00:00:08 RSL: stat/tasmota/RESULT = {"POWER":"OFF"}
From Tasmota<<  00:00:08 RSL: stat/tasmota/POWER = OFF

If we have tasmota ignore all serial commands except those which are prefixed with the device topic (say Tas123)

<To Tasmota      Tas123 Power
From Tasmota>  00:00:08 RSL: stat/Tas123/RESULT = {"POWER":"OFF"}
From Tasmota>  00:00:08 RSL: stat/Tas123/POWER = OFF

Then we can setup polling/Acking of multiple Tasmota devices on a simple network

I am sorry If I missed something whereby this can be done by a setconfig or rule, it would be great to be able to get Tasmota on an RF network.

Also, WiFi Manager resets and serial logging time-out would need to be stopped.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-11:
Nice idea but many new devices use the serial hardware to connect a local mcu controlling their hardware making it unusable for modbus.

Cloolalang, 2020-04-11:
Hey! thanks for taking a look.
Im thinking of Sonoff basic, as its so cheap.
My idea is to offer a simple add-on HC12 Uart radio to a sonoff basic to get that much-needed range.
Im using Hassio with Modbus, and its working well at 500m+.My modbus slave is just a pro-mini with 4 input and 4 output registers.
I guess Im looking for a quick and dirty MAC layer for Tasmota serial to allow a pseudo-multi-drop arrangement.
Maybe I just add a little packet-forwarder at the HC12 with a unique address.
Cheers Theo.
P

Jason2866, 2020-04-11:
Adding additional hardware to sonoff basic is not a good idea because of the very very weak power supply. Most of the time this ends in a unreliable device.

Cloolalang, 2020-04-11:
Good Point Jason2866,
So yes Sonoff-SV with dedicated PSU (and lipo UPS).
After a little researching, I see that Mysensors have an quite well-developed RF networking side.
They are using called serial-2.x
https://www.mysensors.org/download/serial_api_20
P

ascillato, 2020-04-11:
> How to poll individual devices?

Just an idea. As it is Tasmota now, you can use EVENTS on Rules.

* **Device1**:
```lua
rule1 on event#lamp1 do power1 %value% endon
```
* **Device2**:
```lua
rule1 on event#lamp2 do power1 %value% endon
```
* **Device3**:
```lua
rule1 on event#lamp3 do power %value% endon
```

Then if by serial you send to ALL devices the command `EVENT LAMP1=ON`, the only one that will _power on_ will be the **Device1**.

Cloolalang, 2020-04-11:
Arrh yes!
Thanks Adrian.
I see a new, esoteric,  rules/EVENTS-driven sub-protocol!
P

Cloolalang, 2020-04-12:
This is great fun..
So yes this rules polling half works for a multi-drop serial network except that I still get this response: 

datagrams sent and received with Adrian's rules:

1)
Sent:
EVENT LAMP1=ON   

Reply:
00:01:27 CMD: EVENT LAMP1=ON 
00:01:27 RSL: stat/tasmota/RESULT = {"Event":"Done"}
00:01:27 RUL: EVENT#LAMP1 performs "power1 ON"
00:01:27 RSL: stat/tasmota/RESULT = {"POWER":"ON"}
00:01:27 RSL: stat/tasmota/POWER = ON

However,  when its a different "ID"" say, LAMP99 I get:

2)
Sent:
EVENT LAMP99=ON

Reply:
00:02:26 CMD: EVENT LAMP99=ON 
00:02:26 RSL: stat/tasmota/RESULT = {"Event":"Done"}

So, if this was on a multi-drop, these reply datagrams from all the devices would collide, it would be better if Tasmota could be made to kill this CMD and RSL logging response.

I had a play with SerialLog setting but could not see how to do that..

P

Cloolalang, 2020-04-13:
I will come back t this later.
I thought of perhaps setting up a crude time-division way to manage the potential collisions..each Tasmota devise could have a delay rule of different periods..?
Thanks for your feedback guys.
Stay safe.
P

closed 2020-04-13 by Cloolalang

Cloolalang, 2020-06-01:
SLIP 
  I found this:

https://playground.arduino.cc/Code/SerialIP/
https://github.com/krzychb/esp-just-slip

So could SLIP be implemented into Tasmota over software serial?  
My idea goes like:

Remote NODEMCU With Tasmota
>MQTT inside  TCP/IP inside SLIP over serial.
>serial to UART Radio (HC12 NiceRF etc).
>UART radio base radio to serial
>SLIP>TCP/IP>MQTT>WIFI on Base NODEMCU
>WIFI to Ethernet to Controller (Home assisitant)

P

meingraham, 2020-06-01:
Have you looked at https://tasmota.github.io/docs/TasmotaSlave/? Put a sketch on your slave to support the SLIP comms using software serial and the slave then communicates to the Tasmota device on a separate Rx/Tx pair on the slave connected to the ESP82xx (running Tasmota) hardware UART.

Cloolalang, 2020-06-01:
Hey Meingraham! Thanks for taking a look at this.
Interesting.  
Can the slave be a Tasmota ESP266 and pass the MQTT over serial?

meingraham, 2020-06-01:
> pass the MQTT

Can you explain? If you mean to "forward" the message that is received via MQTT by Tasmota to the slave, yes. Your sketch on the slave handles receiving the text and processes it how you need.

Cloolalang, 2020-06-01:
Hey, Im looking for an alternative way of Tasmota connecting to an MQTT broker. Now we have TCP/IP over wifi.
I want to connect my Tasmota device MQTT session over TCP/IP over serial/UART so I can use UART radios like the HC12 or Nice RF Uart Modems instead of WIFI.
p

Cloolalang, 2020-06-02:
I think I will try it with ESPEasy and Hassio Telnet.
https://www.letscontrolit.com/wiki/index.php/Ser2Net