arendst/Tasmota #8136: Hm10 LYWSD03 battery level not working properly

Truddel, 2020-04-11:
### PROBLEM DESCRIPTION
Hi togehter, with big interest i have added my HM10 with LYWSD03 sensors to Tasmota. The only thing is the battery level which is only 100% or 79% rotating. It seems here is something wrong in the moment. Maybe someone is interested in debugging? If any help is required by me I can help. 

Before Tasmota had support for these sensors i have used my RPI with the MiTemperature script from JSBergbau (https://github.com/JsBergbau/MiTemperature2). He had similar issues reading the battery level and found out the LYWSD03 reporting the level as voltage also. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WemosD1Mini with Hm10
- [x] Tasmota binary firmware version number used: 8.2.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
17:22:34 MQT: stat/BLEHub/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:22:34 MQT: stat/BLEHub/RESULT = {"Module":{"0":"Generic"}}
17:22:35 MQT: stat/BLEHub/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"194":"HM10 RX"},"GPIO13":{"0":"None"},"GPIO14":{"195":"HM10 TX"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Battery level in percent or voltage displayed and transmitted correctly.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-04-11:
I will take a look.

Staars, 2020-04-11:
Just a short question to be sure about your version.
Can you use command HM10PAGE?

Because since this commit, the mentioned bug should have been fixed.

79 is simply "O" from a somehow delayed and then unexpected return message from the HM10, like "OK+CONN".

Truddel, 2020-04-11:
Hi, yes i have installed the latest Version 8.2.0.3 with Hm10Page working. 
Pleased let me know If you need any further informations.

Staars, 2020-04-11:
Good.
My assumption is, that you receive a „AT“ and not a „AT+something“ in that moment.

If this is true, a change of line 633 to:
`if(_buf[0]==0x4f && _buf[1]==0x4b) return false; // "OK“

should solve it.

Can you self-compile this for testing? If not I can send you a binary.

Truddel, 2020-04-11:
Hi, i have added  ` if(_buf[0]==0x4f && _buf[1]==0x4b) return;` without return false; because got error due to void and returning.. Same issue with this adjusted line. Most of the sensors show 100% one is showing 79%

Staars, 2020-04-12:
This is very strange, because this line must return a boolean. Your modification can not work.

 I wanted to do some work on this driver anyway where this can be addressed.

Reading the voltage is not hard, but will add up to the code size. I think this is not a good idea.

Truddel, 2020-04-12:
I have chaged this way and compiling was possible: 
  ```
//if(_buf[0]==0x4f && _buf[1]==0x4b && _buf[2]==0x2b) return; // "OK+"
  if(_buf[0]==0x4f && _buf[1]==0x4b) return; // "OK"
```
When compiling this way I get the following error message:
```
  //if(_buf[0]==0x4f && _buf[1]==0x4b && _buf[2]==0x2b) return; // "OK+"
  if(_buf[0]==0x4f && _buf[1]==0x4b) return false; // "OK"
```

```
Compiling .pioenvs\tasmota-sensors\src\tasmota.ino.cpp.o
In file included from C:\users\basti\.platformio\packages\framework-arduinoespressif8266@src-f440165f854f8a9f10351afcd0aa2ed5\cores\esp8266/Arduino.h:29:0,
                 from c:\users\basti\appdata\local\temp\tmp47hxkc:1:
E:/VSCode/Tasmota-development/tasmota/xsns_62_MI_HM10.ino: In function 'void HM10readTLMF(char*)':
E:/VSCode/Tasmota-development/tasmota/xsns_62_MI_HM10.ino:634:45: error: return-statement with a value, in function returning 'void' [-fpermissive]
   if(_buf[0]==0x4f && _buf[1]==0x4b) return false; // "OK"
                                             ^
*** [.pioenvs\tasmota-sensors\src\tasmota.ino.cpp.o] Error 1

```

> 
> I wanted to do some work on this driver anyway where this can be addressed.

sounds good, if you need someone for testing with multiple LYWSD03 just let me know..
> 
> Reading the voltage is not hard, but will add up to the code size. I think this is not a good idea.

Ok it's up to you. If it is working with percentage this is fine for me as well.

Truddel, 2020-04-12:
Not directly related to the issue but maybe you can consider about that:
One thing when you do some work on the driver:
I am not sure how this is handled but I assume when a reading is not successful the data from the last reading is still there and will be sent by mqtt, right? Maybe you can implement kind of expiry of the data or show reading unsuccessful when one of the sensors is out of range or has another difficulty to report the right sensor data.

Staars, 2020-04-12:
I am sorry, I had given you the false line (doing such things from a mobile phone is quite error prone 😁 )

Anyway, I did a new PR that hopefully solves the battery problem (at least it will fail with a smaller memory consumption 😀 ).

The expiry of the data is a problem, that I already thought about too. I think there are roughly two concepts:
1. Do not report "expired" data.
2. Add a timestamp, for instance with an UTC-timestamp with every sensor entry.

As always it is probably not the problem to implement something, but to find a good compromise with regards to memory constraints.

Truddel, 2020-04-12:
Thanks for your work! Now every sensor is showing me 100%. The 79% measurements are gone now. 
When I compare measuring the sensors with the RPI Script I get values like the following:

Sensor which is in use since approx 2 months now: 
```
Temperature: 26.3
Humidity: 33
Battery voltage: 2.737
Battery level: 64
```

Sensor which is in use since yesterday: 
```
Temperature: 25.5
Humidity: 39
Battery voltage: 3.044
Battery level: 94

```
Tasmota is showing 100% for both of them.

You get with tasmota other values like 100% with the LYWSD03?

I think we should consider about a concept filtering out "old expired" values, so the user can decide how to deal with it. I think a UTC timestamp would be the best. Sure we need to maintain the memory as good as possible but in this case I think we can spent a little for it. Otherwise you get multiple time the same values and could not be sure if the temperature is not moving or if it is an old measurement.

Staars, 2020-04-12:
Thanks for reporting.
I never had doubts about the battery values, because they are the same, that an „usual“ app like BLE-scanner shows for the battery. Maybe the sensor shows very rough approximations here.
I will do some internal tests to see, how costly (with regards to memory) a more precise solution will be. Basically we can read out everything.
I agree with the UTC-timestamp. This should be the most versatile solution and the memory consumption will not go through the roof.

But if the primary problem of this issue is solved please close it.

Truddel, 2020-04-12:
I double checked with BLE Scanner right now. You are right. I get 99% for all of my sensors. I think the standard BLE charactarization is nonsense here and i think it would be better to use the battery level which is transmitted in every measurement as well.  I think at MiTemperature from JSBergbau they are doing the same.

Yeah i know we need to have a look for the memory consumption here as well. 

Thanks again for your great job.

closed 2020-04-12 by Truddel