arendst/Tasmota #8142: Energy Today value never resets if time isn't set up

belidzs, 2020-04-11:
### PROBLEM DESCRIPTION
I have a Sonoff Pow module and ever since my iot devices were put into a firewalled quarantine they weren't able to reach any ntp service anymore, resulting their time set back to epoch 0 after a power loss.

It turns out almost everything works as it should even without the correct time set, except
* the energy today timer never reset, it was constantly growing
* I was still able to manually reset the energy today counter
* energy total behaved normally
* interestingly, energy yesterday also behaved normally (it was reset each day)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

23:33:24 MQT: sonoff-pow/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
23:33:24 MQT: sonoff-pow/stat/RESULT = {"Module":{"6":"Sonoff Pow"}}
23:33:25 MQT: sonoff-pow/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"130":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"132":"HLWBL CF1"},"GPIO14":{"133":"HLW8012 CF"},"GPIO15":{"52":"Led1"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

23:33:56 MQT: sonoff-pow/stat/STATUS = {"Status":{"Module":6,"FriendlyName":["Water heater"],"Topic":"sonoff-pow","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
23:33:56 MQT: sonoff-pow/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:11:32","StartupUTC":"2020-04-11T21:22:24","Sleep":50,"CfgHolder":4617,"BootCount":77,"BCResetTime":"2020-04-11T22:43:42","SaveCount":564,"SaveAddress":"F6000"}}
23:33:56 MQT: sonoff-pow/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"333/699"}}
23:33:56 MQT: sonoff-pow/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hk-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8029","2805C8000100068000005AFF000000000000","00000201","00000000"]}}
23:33:56 MQT: sonoff-pow/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:33:56 MQT: sonoff-pow/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-pow-1534","IPAddress":"192.168.1.156","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:EA:65:FE","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:33:56 MQT: sonoff-pow/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EA65FE","MqttUser":"iot","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:33:56 MQT: sonoff-pow/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-04-11T21:33:56","Local":"2020-04-11T23:33:56","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+02:00","Sunrise":"06:01","Sunset":"19:27"}}
23:33:56 MQT: sonoff-pow/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:33:56 MQT: sonoff-pow/stat/STATUS10 = {"StatusSNS":{"Time":"2020-04-11T23:33:56","ENERGY":{"TotalStartTime":"2018-10-30T21:25:56","Total":176.389,"Yesterday":1.909,"Today":0.574,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":245,"Current":0.000}}}
23:33:56 MQT: sonoff-pow/stat/STATUS11 = {"StatusSTS":{"Time":"2020-04-11T23:33:56","Uptime":"0T00:11:32","UptimeSec":692,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"hk-iot","BSSId":"A0:F3:C1:99:41:27","Channel":13,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Setup correct time
1. Use device for some time to get some statistics
1. Block device from being able to sync time
1. Cycle power
1. Use device
1. Energy today counter will not reset anymore
1. Energy yesterday counter will still reset daily

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

* Energy today should also reset daily

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

belidzs, 2020-04-11:
Forgot to mention, after setting the correct time everything started to work again.

I assume I just had to set it to a time later than `TotalStartTime`, but I haven't tested this theory.

```
23:42:36 MQT: sonoff-pow/tele/SENSOR = {"Time":"2020-04-11T23:42:36","ENERGY":{"TotalStartTime":"2018-10-30T21:25:56","Total":176.494,"Yesterday":1.909,"Today":0.679,"Period":52,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":245,"Current":0.000}}
```

Jason2866, 2020-04-11:
Please close Thx

belidzs, 2020-04-11:
@Jason2866 I really don't see why should I close this. This behavior is very confusing and it clearly looks like a bug to me.

After all this is why I have done all the lengthy research required to open this issue even though I have already fixed it for myself.

At least a warning message should be in the log suggesting that there's something wrong with the time as the current time shouldn't precede the value of TotalStartTime. Or the Energy Today should reset anyway. Or the Energy Yesterday also shouldn't reset so there's at least some consistency in the behavior.

Jason2866, 2020-04-12:
Because many functions in Tasmota do not work if there is no correct time.
It is like bringing car to service and make a issue because it does not drive without fuel in the tank.

belidzs, 2020-04-12:
@Jason2866 cars have a "fuel tank empty" light while tasmota doesn't warn you at all if this happens.

If you want a full analogy this is like the car would drive further without gas (energy yesterday resets correctly) but the battery would mysteriously run flat (energy today keeps growing).

arendst, 2020-04-12:
Without valid walltime (is time after 20160101 used to know real time has been detected) the following code will never execute:
```
    if (RtcTime.valid) {
      if (LocalTime() == Midnight()) {
        Settings.energy_kWhyesterday = RtcSettings.energy_kWhtoday;

        RtcSettings.energy_kWhtotal += RtcSettings.energy_kWhtoday;
        Settings.energy_kWhtotal = RtcSettings.energy_kWhtotal;
        Energy.kWhtoday = 0;
        Energy.kWhtoday_offset = 0;
        RtcSettings.energy_kWhtoday = 0;
        Energy.start_energy = 0;

        Energy.kWhtoday_delta = 0;
        Energy.period = Energy.kWhtoday;
        EnergyUpdateToday();
#if defined(USE_ENERGY_MARGIN_DETECTION) && defined(USE_ENERGY_POWER_LIMIT)
        Energy.max_energy_state  = 3;
#endif  // USE_ENERGY_POWER_LIMIT
      }
#if defined(USE_ENERGY_MARGIN_DETECTION) && defined(USE_ENERGY_POWER_LIMIT)
      if ((RtcTime.hour == Settings.energy_max_energy_start) && (3 == Energy.max_energy_state )) {
        Energy.max_energy_state  = 0;
      }
#endif  // USE_ENERGY_POWER_LIMIT

    }
```
You'll miss functionality without valid wall time. As there is no link to walltime midnight the today energy values would be reset just 24 hours after restart as that would be the only time reference.

So in order to get good results a valid walltime is needed for energy monitoring just as your Home Automation software would do.

belidzs, 2020-04-12:
@arendst I don't dispute the fact that having the correct time set up should be (is already) a requirement.

However what I suggested was to provide a clear warning message in case the time is incorrectly set so people will be able to fix this issue quickly. Currently the time isn't displayed neither on the main webpage nor the information page so unless you specifically monitor the console and look for the system time you'll probably never suspect this.

If it was up to me I wouldn't even send out any power statistics until the current time is newer than `20160101`

ascillato2, 2020-04-12:
> a warning message should be in the log suggesting that there's something wrong with the time

So far, the NTP Sync routine don't give any  error if it can't sync when it tries every hour. It makes sense to show an error in the console every hour if it can't sync.

And if there is no time synced, the Energy Yesterday shouldn't reset either.

@arendst Theo, what do you think of adding that error message in the log? If you agree, I can add it.

ascillato2, 2020-04-20:
Closing as the correct time is required for today energy calculation. Thanks for reporting.

closed 2020-04-20 by ascillato2

wernerhp, 2020-10-31:
Thanks for reporting this @belidzs 
I also noticed my Today does not reset and couldn't figure out why.  Time was not synced with no error.  Thanks for adding the NTP warning in the new version @ascillato2