arendst/Tasmota #8149: Set Global Temp and Humidity via MQTT or Serial command

gururise, 2020-04-12:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
The CCS811 VOC and eCO2 sensor can be compensated using current temperature and humidity and thus provide more accurate sensor readings.  This capability already exists in Tasmota; however, as far as I can tell, it requires the use of another physically connected sensor that can provide temperature and humidity data.  

I have the situation where I have multiple IOT devices each connected to one or two sensors.  The ESP-12 I have connected to my CCS811 sensor does not have another physically connected sensor that can provide temperature or humidity data; however, I do have this data available from other nearby tasmota devices that I would like to pass via MQTT message.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Global Temperature and Humidity data should be able to be passed into Tasmota via MQTT or Serial command if a temp/humidity sensor is not physically connected.  That way sensors that can use this data (such as CCS811 or SGP30) can provide more accurate readings if other nearby sensors can provide such data.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

None.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-04-12:
To be honest, I think, it's not the task of Tasmota to fulfill the tasks of an overlaying system or some conditions  that can easily be changed.
If you need additional sensors, you should switch to a device, that gives you more GPIOs to add additional sensors.

gururise, 2020-04-12:
My opinion is it's probably a simple enough change that it would be worth it.  Already numerous functionality can be controlled via MQTT messages, why not setting global temp/humidity?

If I get a bit of free time I may submit a PR for this...

schrej-zz, 2020-04-13:
> **Have you looked for this feature in other issues and in the docs?**
> Yes.
>
> **Is your feature request related to a problem? Please describe.**
> _A clear and concise description of what the problem [...]

But he talked about sensors NOT being physically connected. Or do I missunderstand anything?

device111, 2020-04-13:
you can ask Theo. at this time you can set hum and temp offset commands but not temp and hum itself. the Code for doing this is not so big.

Jason2866, 2020-04-13:
@schrej-zz  Why using more sensors than needed? The values are there from one! sensor connected to a other Tasmota device. IF it is easily to add with just a little code size increase it is a nice feature.
Maybe with a little support via rules

closed 2020-04-13 by ascillato2