arendst/Tasmota #8163: Tasmota + restart homeassistant = wrong status

bartplessers, 2020-04-13:
### PROBLEM DESCRIPTION

Hello,

I'm running homeassistant with 3 tasmota devices.
In hass, they are discovered with following entity ID's:
  - switch.tasmota
  - switch.tasmota_2
  - switch.tasmota_3

and displayed as
  - Tasmota
  - Tasmota
  - Tasmota

I know I can override the display name in homeassistant, but I decided to configure my tasmota devices manually as follows:
```

# =================================================
# SONOFF
# TASMOTA firmware
# =================================================
switch:
  - platform: mqtt
    name: "Tasmota-105"
    state_topic: "tasmota-105/stat/RESULT"  
    value_template: "{{ value_json.POWER }}"
    command_topic: "tasmota-105/cmnd/POWER"
    payload_on: "ON"
    payload_off: "OFF"
    availability_topic: "tasmota-105/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false

```
I configured lovelace
  - switch.tasmota_105
  
This works fine. I can control my Tasmota device with homeassistant.

However, after restarting homeassistant, I noticed that the status of de tasmota devices is NOT correctly displayed.
In example: 
  - Tasmota-105 has status "ON"
  - I restart homeassistant
  - Tasmota-105 has status of "OFF" in the gui (but is "ON" in reality)

The original discoverd "switch.tasmota_2" (wich is the same as my manually configured "switch.tasmota_105") has correct behaviour: after restarting homeassistant, the switch stays on "ON"

Question: how can I configure my tasmota device manually so that it's status is showed correctly after restarting homeassistant?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basis
- [x] Tasmota binary firmware version number used: 8.2.0(tasmota), 2020-03-20T14:45:23
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: FTDI FT232
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:29:59 CMD: Backlog Template; Module; GPIO 255
21:29:59 MQT: tasmota-105/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:29:59 MQT: tasmota-105/stat/RESULT = {"Module":{"8":"Sonoff S2X"}}
21:30:00 MQT: tasmota-105/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:31:34 MQT: tasmota-105/stat/STATUS = {"Status":{"Module":8,"FriendlyName":["Tasmota"],"Topic":"%hostname%","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:31:34 MQT: tasmota-105/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T00:35:37","StartupUTC":"2020-04-12T19:55:57","Sleep":50,"CfgHolder":4617,"BootCount":46,"BCResetTime":"2020-04-12T20:56:05","SaveCount":346,"SaveAddress":"F5000"}}
21:31:34 MQT: tasmota-105/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":6,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"341/699"}}
21:31:35 MQT: tasmota-105/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BEARFORCE-IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
21:31:35 MQT: tasmota-105/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:31:35 MQT: tasmota-105/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-105","IPAddress":"192.168.3.105","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"5C:CF:7F:7A:1C:D4","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
21:31:35 MQT: tasmota-105/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.bearforce.org","MqttPort":1883,"MqttClientMask":"tasmota-105","MqttClient":"tasmota-105","MqttUser":"tasmota-105","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:31:35 MQT: tasmota-105/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-04-13T20:31:35","Local":"2020-04-13T21:31:35","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:02","Sunset":"19:39"}}
21:31:35 MQT: tasmota-105/stat/STATUS10 = {"StatusSNS":{"Time":"2020-04-13T21:31:35"}}
21:31:35 MQT: tasmota-105/stat/STATUS11 = {"StatusSTS":{"Time":"2020-04-13T21:31:35","Uptime":"1T00:35:38","UptimeSec":88538,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BEARFORCE-IoT","BSSId":"F0:79:59:D4:11:24","Channel":6,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
homeassistant reflects the correct status of tasmota device after restart

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-04-13:
https://tasmota.github.io/docs/Home-Assistant/ 
Read about `Sync Power State`. 

If you have any other questions please address them to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Thanks.

closed 2020-04-13 by effelle

bartplessers, 2020-04-19:
Hi @effelle,

thanx for pointing me to the documentation.
However, the this solved not my problem.
The real problem was ... a typo in my config :-(

my original state_topic was
`
    state_topic: "tasmota-105/stat/RESULT"  
`

the correct one:
`
    state_topic: "tasmota-105/tele/STATE"

`

full manual config
```

# =================================================
# SONOFF
# TASMOTA firmware
# =================================================
switch:
  - platform: mqtt
    name: "Tasmota-105"
    state_topic: "tasmota-105/tele/STATE"
    value_template: "{{ value_json.POWER }}"
    command_topic: "tasmota-105/cmnd/POWER"
    payload_on: "ON"
    payload_off: "OFF"
    availability_topic: "tasmota-105/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false

```

bartplessers, 2020-04-20:
@effelle : I'm not sure about my "solution". If it is the right way to set things up, then the documentation is wrong...

From documentation, manual config
https://tasmota.github.io/docs/Home-Assistant/

```
switch:
  - platform: mqtt
    name: "Tasmota Switch"
    state_topic: "stat/tasmota/RESULT"  
    value_template: "{{ value_json.POWER }}"
    command_topic: "cmnd/tasmota/POWER"
    payload_on: "ON"
    payload_off: "OFF"
    availability_topic: "tele/tasmota/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false

```

But this seems not to work, as I figured out above.
So is that a "bug" in the documentation, and should it be
`
    state_topic: "tele/tasmota/STATE"  

`

?