arendst/Tasmota #8167: Tasmota Lite not saving configuration

ilgrank, 2020-04-14:
### PROBLEM DESCRIPTION
On some older 4Mbit (512Kbytes) device, using Tasmota Lite (the standard version won't fit) the configuration isn't saved.
I have read that Minimal is intended to behave this way, but Lite shouldn't as far as I know.
I have tried both inputting Wifi by hand an by restoring a working config. Same result.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Expressif ESP8266 ESP-12 4Mbit
- [x] Tasmota binary firmware version number used: 8.2.0 Standard (Tasmotizer reports it won't fit) so  switched to 8.2.0 Lite
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: Tasmotizer 1.1
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
Can't issue commands as the module reboots in AP mode.
Phisical Serial console log:
```
00:00:37 UPL: File Config_tasmota_CC0A18_2584_8.2.0(1).dmp ...
00:00:37 UPL: Successful 4096 bytes. Restarting
00:00:38 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
v372a3ec2
~ld
   00:00:00 CFG: Use defaults
00:00:00 Project tasmota Tasmota Version 8.2.0(lite)-STAGE
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on tasmota_9F8554-1364 with IP address 192.168.4.1
00:00:41 WIF: Hostname %s-%04d, SSId1 Rome2, SSId2 , CORS
00:00:43 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
v372a3ec2
~ld
   00:00:00 CFG: Use defaults
00:00:00 Project tasmota Tasmota Version 8.2.0(lite)-STAGE
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on tasmota_9F8554-1364 with IP address 192.168.4.1

```
### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash an ESP-12 4mbit with 8.2.0 lite using Tasmotizer 1.1 with Erase Flash option set.
Configure the Wifi credentials.

### EXPECTED BEHAVIOUR
Configuration should be saved.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-14:
Pls provide status 0 output

ilgrank, 2020-04-14:
how can I provide Status 0 output? I can't go beyond serial output..

arendst, 2020-04-14:
To flash in <1MB do not use Tasmotizer as it uses a wrong memory model. Use another tool like esptool as documented.

You may also hit the max image size as the configuration is saved at the end of your 0.5MB flash leaving only 462k for an image so tasmota-lite v8.2.0 is too large already.

You'll have to compile a smaller one yourselves.

ilgrank, 2020-04-14:
I had tried with ESP Easy Flasher (which, in turn, uses Esptool in background) but flashing with it produced just a boot loop.
Just to be sure, I used also the guide here: 
https://tasmota.github.io/docs/Getting-Started/#esptool-executable
and it produced the same result (ie: bootloop)

As for compiling a version (which is beyond my skills anyways) what else could be removed that hasn'b been already from the Lite version?

ascillato2, 2020-04-14:
Tasmota is not supported on an old 512KB device. Sorry.
To make Tasmota works on that, you need to compile with a memory map of 512KB. The default and minimum of Tasmota is 1MB:
https://github.com/arendst/Tasmota/blob/e3503ab5b82aecac3068ead0640d134570522f2c/platformio.ini#L60
And also you may need to modify Tasmota to use less flash for settings.

ilgrank, 2020-04-14:
My s setup is not supported from the releases, I'll try to build it.
Thanks @ascillato2  and @arendst !

closed 2020-04-14 by ilgrank

suda, 2020-12-30:
I managed to squeeze Tasmota Lite on the 512KB module following the advice by @ascillato2 
The exact changes were:

`platformio.ini`
```ini
[common]
board                       = esp01
board_build.ldscript        = eagle.flash.512k.ld
```

Also had to trim down the binary even further by disabling emulation and turning on Home Assistant in my case in the `tasmota/tasmota_configurations.h` file in the `tasmota-lite` section (the one between `#ifdef FIRMWARE_LITE` and `#endif  // FIRMWARE_LITE`). I ended up with 89% flash usage and the config was saved correctly.
Hope this helps @ilgrank

ascillato, 2020-12-30:
Thanks for sharing this information.

ilgrank, 2021-01-03:
thanks @suda !
I've since disposed my 512kb modules, but I think this could be of general interest
@ascillato : in fact, a 512kb version (tastmota ultralite?) could be useful to all the people still owning 512kb modules.
Happy new year to everyone :)

suda, 2021-01-04:
I have no idea how many people are there with that exact need but I can imagine adding `ultralite` would be an extra burden for the devs to maintain. Maybe a note in the docs would be enough? Anyway, in the meantime, I wrote the steps (and attached binary) in [a blog post](https://suda.pl/squeezing-tasmota-on-512kb-esp8266-module/).

aniston, 2021-06-11:
Hello @suda , i used your tasmota-9.2.0-ultralite.bin and it worked well for relays, thanks for the compilation.
any chances of getting a DS18B20 driver in the ultralite build ? and a current version of Tasmota v9.4.0 ?

thanks anyways for sharing the ultralite build, that works. I did try compiling myself but did not get even running correctly with Platform.io hence asking around if it's possible for a DS18B20 driver inside.

regards,
aniston.