arendst/Tasmota #8170: no password, but HTTP status 200

GrendelWWU, 2020-04-14:
### PROBLEM DESCRIPTION
when sending a http request to a tasmota device, that has a web password, you'll always get a code 200. if you omit the password or use the wrong one, you'll get 200, too.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [-] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP01
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [-] Self-compiled
    - [-] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer, later: OTA via hackbox
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:42:34 CMD: Backlog Template; Module; GPIO 255
10:42:35 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:42:35 RSL: RESULT = {"Module":{"18":"Generic"}}
10:42:35 RSL: RESULT = {"GPIO0":{"4":"DS18x20"},"GPIO1":{"0":"None"},"GPIO2":{"158":"LedLinki"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [-] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:43:33 CMD: status 0
10:43:33 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["Thermo-01"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:43:33 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:05","StartupUTC":"2020-04-14T08:34:28","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2020-03-22T13:00:09","SaveCount":87,"SaveAddress":"FA000"}}
10:43:33 RSL: STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020.03.20 14:54:51","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"414/699"}}
10:43:33 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["geleglu-point-01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","01000000","00000000"]}}
10:43:33 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:43:33 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-7079","IPAddress":"10.11.12.222","Gateway":"10.11.12.254","Subnetmask":"255.255.255.0","DNSServer":"10.11.12.254","Mac":"2C:F4:32:7A:7B:A7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:43:33 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-04-14T08:43:33","Local":"2020-04-14T10:43:33","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"07:00","Sunset":"20:40"}}
10:43:33 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-04-14T10:43:33","DS18B20":{"Id":"01191A3CF510","Temperature":23.2},"TempUnit":"C"}}
10:43:33 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-04-14T10:43:33","Uptime":"0T00:09:05","UptimeSec":545,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"geleglu-point-01","BSSId":"E0:28:6D:6D:A5:9F","Channel":6,"RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [-] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
just set a webpassword and sent a request like
http://10.11.12.222/cm?user=admin&password=gehheim&cmnd=Status%208
when the password isn't gehheim but anything else, you'll get code 200. 200 means OK.

### EXPECTED BEHAVIOUR
The text output is okay, but the HTTP status may not be 200.
I would expect "Error 401: Unauthorized"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-04-14:
Hi,

Just tested and your command works as expected. If the device has a password and the password is ok, it returns
```lua
{"StatusSNS":{"Time":"2020-04-14T13:09:24"}}
```
And if the password is wrong, it returns a message telling you that.
```lua
{"WARNING":"Need user=<username>&password=<password>"}
```lua

GrendelWWU, 2020-04-14:
This is the text. But the http status code is still 200. It has to be 401.

ascillato2, 2020-04-14:
@arendst ? What do you think?

arendst, 2020-04-14:
I'll see if I can change it with minimal code.

arendst, 2020-04-14:
What's the impact of not having 401?

GrendelWWU, 2020-04-14:
I'm retrieving the answer from Tasmota devices for a weather station with commands like http://10.11.12.30/cm?user=admin&password=geheim&cmnd=Status%208. This weather station is also equipped with an 8266 and I'm first reading the status. Then, when it's 200, I'm using the JSON parser. It's sending weird things, cause it doesn't expect this return values. So, it's better getting code 401.

ascillato2, 2020-04-17:
Closing as this request has been added by Theo.

Thanks.

closed 2020-04-17 by ascillato2