arendst/Tasmota #8173: Add view Analog data in script/

AlexTransit, 2020-04-14:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
the ability to read ADC data in a script

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
data can be obtained in rules
Analog#A0div10

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2020-04-14:
you can catch all json tele period messages in the >T section
define a variable in >D an fill it in >T
\>D
volt=0
\>T
volt=ANALOG#A0

\>S
print voltage=%volt%

ascillato2, 2020-04-14:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-14 by ascillato2

AlexTransit, 2020-04-14:
Thank you so much