arendst/Tasmota #8178: SwitchTopic 0 ignored if Rule definied

bennydente, 2020-04-14:
### PROBLEM DESCRIPTION
I have definied Rule1 for switch1 and Rule2 for switch2 to write their state on specified topic on mqtt. every switch update the right topic with its state. but also send message in %topic%/stat/SWITCH1
and %topic%/stat/SWITCH2 ignorig SwitchTopic 0 directive.

### REQUESTED INFORMATION
- [ ] Device used (e.g., Sonoff Basic): nodemcu v3
- [ ] Tasmota binary firmware version number used: 8.2.0 Pre-compiled

### TO REPRODUCE
set this rules.
Rule1 on Switch1#state do Publish %topic%/stat/SWITCH1_STAT %value% endon
Rule2 on Switch2#state do Publish %topic%/stat/SWITCH2_STAT %value% endon

Observe MQTT message in
%topic%/stat/SWITCH2
%topic%/stat/SWITCH1
%topic%/stat/SWITCH1_STAT
%topic%/stat/SWITCH2_STAT

closed 2020-04-14 by close-issue-app[bot]

ascillato, 2020-04-14:
As designed. This is explained in the docs

bennydente, 2020-04-14:
Thanks. I have read docs. But I understood that only message defined in rule is send to MQTT..

ascillato, 2020-04-14:
Ok, so that means that there is room for docs improvement. Please, if you want, click on IMPROVE THIS ARTICLE on the top right of the docs' page and make a better clarification. Thanks.

bennydente, 2020-04-15:
yes i can, but i think this is not a good design maybe. why ignoring SwitchTopic 0 config?

bennydente, 2020-04-15:
mate "blakadder" thinks the docs are correct https://github.com/tasmota/docs/pull/265 . in my opinion the logic of firmware is not correct but, anyway... oook.. thanks.

blakadder, 2020-04-15:
is the design not correct or the docs are not correct?

bennydente, 2020-04-15:
The design. Because if i set the Rule 1 to avoid Power control, the system writes the topic as set in the Rule, but it also working like SwitchTopic 1 directive is set, even if the SwitchTopic directive is set to 0.