arendst/Tasmota #8188: encrypt mqtt_passwd and webui passwd

excxapp, 2020-04-15:
**Have you looked for this feature in other issues and in the docs?**  
if we backup data, the passwd is unencrypted, so can encrypt passwd for safety,
this feature can deploy by self complie.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
backup data id ver danger
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
user passwd can encrypt.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-15:
Not enough data. Pls fill in the template completely.

excxapp, 2020-04-15:
ok, I filled.

excxapp, 2020-04-17:
I have encrypt mqtt_passwd and add globalsign root ca cert  .
can i pr?

ascillato2, 2020-05-06:
> can i pr?

yes, please. Thanks

excxapp, 2020-05-13:
ok,i will clean some code and pr

closed 2020-06-12 by stale[bot]