arendst/Tasmota #8190: Webui - Use of "Back to xx" text string option to go back previous menu

bovirus, 2020-04-15:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Is not a problem but improvemnt.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

In Configuration menu latest option to go back to main menu is "Main menu".
I suggest to rename it as "Back to main menu" (more clear).

In configuration page is you select an option (es. WiFi configuration", the latest option is "Configuration" (to go back previous menu".
I suggest to rename it as "Back to Configuration".

Generally speaking when you have latest option to go back to previous menu I suggest it to rename as "Back to xxx". 

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-15:
You can always change the text using the language translation files. Currently it's standard GUI practice.

bovirus, 2020-04-15:
@arendst 

Thanks for the answer.
I'd like to know if there different strings for two differenmt GUI items.
Example from "Configuration" string.

is there a string for "Configuration" in main menu and a strings for current "Configuration" (back to configuration)?

How can recognzie the strings id used in webui to back to ptevious menu?

Please check screenshots.
Sorry but =i'm using IT web interface

The strings "Configurazioen" used in 1/2/374 screenshots is teh same string id or is there different string id for the same string "Configurazione" used in different webui section?

[image]
[image]
[image]
[image]
I bleiev that also for English UI 

"Back to main menu" is more clear then current "Main Menu" in "Configuration" webui page
"Back to Configuration" is more clear than "Configuration" in "Wifi Configuration" webui page.

arendst, 2020-04-15:
I see your point regarding the configuration sub-menus. Still as designed.

As the main goal was (and is) to add functionality with at less frills as possible keeping the code as small as possible this was never a main issue. Remember ebery character is a byte of wated code space.

bovirus, 2020-04-15:
@arendst 

I know very well.

On the same time if the webui could be improve and make more clear as function with a very little change I believe that could be suefull for all users (expeccialy for novice).

On my Sonoff device free memory is 26KB, then I beleieve that you could spend 50 bytes to make these changes.

You have only add a string ID like

BACK_CONFIGURATION "Back to Configuration"

and use it the specifc webui context.

bovirus, 2020-04-17:
@effelle 
@ascillato2 

What do you think about changing when need "Back to xx" for a better UI knowledge?

ascillato, 2020-04-17:
Hi, your idea is right. I agree, but I think is more important to save as much memory as we can.

bovirus, 2020-04-17:
@ascillato 
As I mentioned the free memory is about 46 KB and the sapce for text string requetsd could be 30 bytes.
I believe that 30bytes are not so big difference.

ascillato2, 2020-04-17:
Thread continues in https://github.com/arendst/Tasmota/pull/8211

closed 2020-04-17 by ascillato2