arendst/Tasmota #8192: zigbee2tasmota

Dr4g0nM4st3r, 2020-04-15:
Great work with the zigbee module!
I miss the support of the Philipps Hue Motion Sensor, can you add it?

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-04-15:
Sure, is it this device? https://www.zigbee2mqtt.io/devices/9290012607.html

Chances are it is already supported. If not please set `weblog 3` and paste here the logs

Dr4g0nM4st3r, 2020-04-15:
this is the complete output with weblog 4.
when pairing i see the last line, but that's all that is sent. with zigbee2mqtt this sensor worked.

`18:53:33 WIF: Checking connection...
18:53:41 ZIG: Bytes follow_read_metric = 3268
18:53:41 ZIG: {"ZbZNPReceived":"45CACDB6705B2903018817000000"}
18:53:41 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x0017880103295B70","ShortAddr":"0xB6CD","ParentNetwork":"0x0000"}}
18:53:41 ZIG: Bytes follow_read_metric = 3268
18:53:41 ZIG: {"ZbZNPReceived":"45C1CDB6CDB6705B29030188170080"}
18:53:41 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x0017880103295B70","ShortAddr":"0xB6CD","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
18:53:41 ZIG: ZbZNPSent 2505CDB6CDB6
18:53:41 ZIG: Bytes follow_read_metric = 3273
18:53:41 ZIG: {"ZbZNPReceived":"650500"}
18:53:42 ZIG: Bytes follow_read_metric = 3285
18:53:42 ZIG: {"ZbZNPReceived":"4585CDB600CDB6020201"}
18:53:42 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x02","0x01"]}}
18:53:42 ZIG: ZbZNPSent 2401CDB60201000003301E0700030004000500
18:53:42 ZIG: Bytes follow_read_metric = 3290
18:53:42 ZIG: {"ZbZNPReceived":"640100"}
18:53:42 ZIG: Bytes follow_read_metric = 3347
18:53:42 ZIG: {"ZbZNPReceived":"4480000103"}
18:53:42 ZIG: Bytes follow_read_metric = 3347
18:53:42 ZIG: {"ZbZNPReceived":"448100000000CDB60201000F0072BF0100001A18030104000042075068696C6970730500004206534D4C303031CDB61D"}
18:53:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":15,"securityuse":0,"seqnumber":0,"timestamp":114546,"fc":"0x18","manuf":"0x0000","transact":3,"cmdid":"0x01","payload":"04000042075068696C6970730500004206534D4C303031"}}
18:53:42 ZIG: ZbZCLRawReceived: {"0xB6CD":{"0000/0004":"Philips","0000/0005":"SML001"}}
18:53:42 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Manufacturer":"Philips","ModelId":"SML001","Endpoint":2,"LinkQuality":15}}}
18:53:53 WIF: Checking connection...
18:53:58 MQT: tele/Zigbee/STATE = {"Time":"2020-04-15T18:53:58","Uptime":"0T01:30:13","UptimeSec":5413,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Spacedragon","BSSId":"74:83:C2:81:14:A3","Channel":13,"RSSI":100,"Signal":-20,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:54:13 WIF: Checking connection...`

Dr4g0nM4st3r, 2020-04-15:
this is the complete output with weblog 4.
when pairing i see the last line, but that's all that is sent. with zigbee2mqtt this sensor worked.

`18:53:33 WIF: Checking connection...

18:53:41 ZIG: Bytes follow_read_metric = 3268

18:53:41 ZIG: {"ZbZNPReceived":"45CACDB6705B2903018817000000"}

18:53:41 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x0017880103295B70","ShortAddr":"0xB6CD","ParentNetwork":"0x0000"}}

18:53:41 ZIG: Bytes follow_read_metric = 3268

18:53:41 ZIG: {"ZbZNPReceived":"45C1CDB6CDB6705B29030188170080"}

18:53:41 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x0017880103295B70","ShortAddr":"0xB6CD","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}

18:53:41 ZIG: ZbZNPSent 2505CDB6CDB6

18:53:41 ZIG: Bytes follow_read_metric = 3273

18:53:41 ZIG: {"ZbZNPReceived":"650500"}

18:53:42 ZIG: Bytes follow_read_metric = 3285

18:53:42 ZIG: {"ZbZNPReceived":"4585CDB600CDB6020201"}

18:53:42 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x02","0x01"]}}

18:53:42 ZIG: ZbZNPSent 2401CDB60201000003301E0700030004000500

18:53:42 ZIG: Bytes follow_read_metric = 3290

18:53:42 ZIG: {"ZbZNPReceived":"640100"}

18:53:42 ZIG: Bytes follow_read_metric = 3347

18:53:42 ZIG: {"ZbZNPReceived":"4480000103"}

18:53:42 ZIG: Bytes follow_read_metric = 3347

18:53:42 ZIG: {"ZbZNPReceived":"448100000000CDB60201000F0072BF0100001A18030104000042075068696C6970730500004206534D4C303031CDB61D"}

18:53:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":15,"securityuse":0,"seqnumber":0,"timestamp":114546,"fc":"0x18","manuf":"0x0000","transact":3,"cmdid":"0x01","payload":"04000042075068696C6970730500004206534D4C303031"}}

18:53:42 ZIG: ZbZCLRawReceived: {"0xB6CD":{"0000/0004":"Philips","0000/0005":"SML001"}}

18:53:42 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":
{"Device":"0xB6CD","Manufacturer":"Philips","ModelId":"SML001","Endpoint":2,"LinkQuality":15}}}

18:53:53 WIF: Checking connection...

18:53:58 MQT: tele/Zigbee/STATE = {"Time":"2020-04-15T18:53:58","Uptime":"0T01:30:13","UptimeSec":5413,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":13,"RSSI":100,"Signal":-20,"LinkCount":1,"Downtime":"0T00:00:06"}}

18:54:13 WIF: Checking connection...`

s-hadinger, 2020-04-15:
Indeed it does not send anything. Let me check zigbee2mqtt.

s-hadinger, 2020-04-15:
The first thing you need to do is bind the motion sensor to your coordinator, on 4 clusters:

```
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0001"}
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0400"}
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0402"}
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0406"}
```

Zigbee2Mqtt also configures "configureReporting" which is a little more tricky.
Please let me know if you get anything with those commands.

Dr4g0nM4st3r, 2020-04-15:
No, unfortunately nothing has changed. I also restarted ESP8266, but the messages (after taking the battery out and putting it back in the sensor) are the same as before.

s-hadinger, 2020-04-15:
Can you paste the results of the ZbBind commands?

Dr4g0nM4st3r, 2020-04-15:
1.

21:27:03 CMD: ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0001"}
21:27:03 SRC: WebConsole from 10.0.0.2
21:27:03 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0001"}"
21:27:03 ZIG: ZbZNPSent 2521CDB6705B2903018817000201000368AD371D004B120001
21:27:03 MQT: stat/Zigbee/RESULT = {"ZbBind":"Done"}
21:27:03 ZIG: Bytes follow_read_metric = 2213
21:27:03 ZIG: {"ZbZNPReceived":"652100"}
21:27:04 ZIG: Bytes follow_read_metric = 2220
21:27:04 ZIG: {"ZbZNPReceived":"45A1CDB600"}
21:27:04 MQT: tele/Zigbee/RESULT = {"ZbBind":{"Device":"0xB6CD","Name":"HueMotion","Status":0,"StatusMessage":"SUCCESS"}}

2
21:27:36 CMD: ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0400"}
21:27:36 SRC: WebConsole from 10.0.0.2
21:27:36 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0400"}"
21:27:36 ZIG: ZbZNPSent 2521CDB6705B2903018817000200040368AD371D004B120001
21:27:36 MQT: stat/Zigbee/RESULT = {"ZbBind":"Done"}
21:27:36 ZIG: Bytes follow_read_metric = 2225
21:27:36 ZIG: {"ZbZNPReceived":"652100"}
21:27:38 WIF: Checking connection...
21:27:39 ZIG: Bytes follow_read_metric = 2232
21:27:39 ZIG: {"ZbZNPReceived":"45A1CDB600"}
21:27:39 MQT: tele/Zigbee/RESULT = {"ZbBind":{"Device":"0xB6CD","Name":"HueMotion","Status":0,"StatusMessage":"SUCCESS"}}

3
21:28:30 CMD: ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0402"}
21:28:30 SRC: WebConsole from 10.0.0.2
21:28:30 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0402"}"
21:28:30 ZIG: ZbZNPSent 2521CDB6705B2903018817000202040368AD371D004B120001
21:28:31 MQT: stat/Zigbee/RESULT = {"ZbBind":"Done"}
21:28:31 ZIG: Bytes follow_read_metric = 2237
21:28:31 ZIG: {"ZbZNPReceived":"652100"}
21:28:34 ZIG: Bytes follow_read_metric = 2244
21:28:34 ZIG: {"ZbZNPReceived":"45A1CDB600"}
21:28:34 MQT: tele/Zigbee/RESULT = {"ZbBind":{"Device":"0xB6CD","Name":"HueMotion","Status":0,"StatusMessage":"SUCCESS"}}

4
21:28:54 CMD: ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0406"}
21:28:54 SRC: WebConsole from 10.0.0.2
21:28:54 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0406"}"
21:28:54 ZIG: ZbZNPSent 2521CDB6705B2903018817000206040368AD371D004B120001
21:28:54 MQT: stat/Zigbee/RESULT = {"ZbBind":"Done"}
21:28:54 ZIG: Bytes follow_read_metric = 2249
21:28:54 ZIG: {"ZbZNPReceived":"652100"}
21:28:58 WIF: Checking connection...
21:28:59 ZIG: Bytes follow_read_metric = 2256
21:28:59 ZIG: {"ZbZNPReceived":"45A1CDB600"}
21:28:59 MQT: tele/Zigbee/RESULT = {"ZbBind":{"Device":"0xB6CD","Name":"HueMotion","Status":0,"StatusMessage":"SUCCESS"}}

s-hadinger, 2020-04-16:
Thanks, now the second step is to configure the reporting of 4 attributes: battery, temperature, illuminance and occupancy.

Here are the 4 commands, that I made in blind mode. It should work but I'm not completely sure. Please report the responses. You need to do the 4 other ZbBind commands.

```
ZbSend {"Device":"HueMotion","Endpoint":2,"Cluster":"0x0001","Send":"0001_06/00210020FFFF000000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Cluster":"0x0001","Send":"0400_06/00000021FFFF00000000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Cluster":"0x0001","Send":"0402_06/00000029FFFF00000000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Cluster":"0x0001","Send":"0406_06/00000018FFFF000000"}
```

Dr4g0nM4st3r, 2020-04-16:
That looks already very good, temperature, battery and brightness are sent now. Now the "Is motion" message is still missing. As far as I can see, a command has failed there ("StatusMessage": "MALFORMED_COMMAND",)

09:25:13 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0001_06/00210020FFFF000000"}

09:25:13 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}

09:25:13 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":31}}}

09:25:15 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","BatteryPercentage":171,"Endpoint":2,"LinkQuality":34}}}

09:25:25 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0400_06/00000021FFFF00000000"}

09:25:25 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}

09:25:28 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":34}}}

09:25:30 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Illuminance":12096,"Endpoint":2,"LinkQuality":28}}}

09:25:31 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Illuminance":12037,"Endpoint":2,"LinkQuality":31}}}

09:25:31 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Illuminance":12037,"Endpoint":2,"LinkQuality":31}}}

09:25:36 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0402_06/00000029FFFF00000000"}

09:25:36 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}

09:25:37 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":28}}}

09:25:38 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Temperature":19.58,"Endpoint":2,"LinkQuality":31}}}

09:25:44 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/00000018FFFF000000"}

09:25:44 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}

09:25:44 MQT: tele/Zigbee/RESULT = {"ZbResponse":{"Device":"0xB6CD","Name":"HueMotion","Command":"0406!06","Status":128,"StatusMessage":"MALFORMED_COMMAND","Endpoint":2,"LinkQuality":31}}

Dr4g0nM4st3r, 2020-04-16:
Weblog 4

`09:32:30 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0001_06/00210020FFFF000000"}
09:32:30 SRC: WebConsole from 10.0.0.2
09:32:30 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"HueMotion","Endpoint":2,"Send":"0001_06/00210020FFFF000000"}"
09:32:30 ZigbeeZCLSend device: 0xB6CD, group: 0x0000, endpoint:2, cluster:0x0001, cmd:0x06, send:"00210020FFFF000000"
09:32:30 ZbSend: shortaddr 0xB6CD, groupaddr 0x0000, cluster 0x0001, endpoint 0x02, cmd 0x06, data 
09:32:30 ZIG: ZbZNPSent 240202CDB600000000000002000001010009301E0C0000090600210020FFFF000000
09:32:30 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}
09:32:30 ZIG: Bytes follow_read_metric = 21379
09:32:30 ZIG: {"ZbZNPReceived":"640200"}
09:32:35 ZIG: Bytes follow_read_metric = 21414
09:32:35 ZIG: {"ZbZNPReceived":"4480000109"}
09:32:35 ZIG: Bytes follow_read_metric = 21414
09:32:35 ZIG: {"ZbZNPReceived":"448100000100CDB60201000D001F69D100000418090700CDB61D"}
09:32:35 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":13,"securityuse":0,"seqnumber":0,"timestamp":13723935,"fc":"0x18","manuf":"0x0000","transact":9,"cmdid":"0x07","payload":"00"}}
09:32:35 ZIG: ZbZCLRawReceived: {"0xB6CD":{}}
09:32:35 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":13}}}
09:32:39 WIF: Checking connection...
09:32:55 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0400_06/00000021FFFF00000000"}
09:32:55 SRC: WebConsole from 10.0.0.2
09:32:55 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"HueMotion","Endpoint":2,"Send":"0400_06/00000021FFFF00000000"}"
09:32:55 ZigbeeZCLSend device: 0xB6CD, group: 0x0000, endpoint:2, cluster:0x0400, cmd:0x06, send:"00000021FFFF00000000"
09:32:55 ZbSend: shortaddr 0xB6CD, groupaddr 0x0000, cluster 0x0400, endpoint 0x02, cmd 0x06, data 
09:32:55 ZIG: ZbZNPSent 240202CDB60000000000000200000100040A301E0D00000A0600000021FFFF00000000
09:32:55 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}
09:32:55 ZIG: Bytes follow_read_metric = 21419
09:32:55 ZIG: {"ZbZNPReceived":"640200"}
09:32:55 ZIG: Bytes follow_read_metric = 21426
09:32:55 ZIG: {"ZbZNPReceived":"448000010A"}
09:32:55 ZIG: Bytes follow_read_metric = 21454
09:32:55 ZIG: {"ZbZNPReceived":"448100000004CDB60201000500F65CD2000004180A0700CDB61D"}
09:32:55 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1024,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":5,"securityuse":0,"seqnumber":0,"timestamp":13786358,"fc":"0x18","manuf":"0x0000","transact":10,"cmdid":"0x07","payload":"00"}}
09:32:55 ZIG: ZbZCLRawReceived: {"0xB6CD":{}}
09:32:55 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":5}}}
09:32:59 WIF: Checking connection...
09:33:07 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0402_06/00000029FFFF00000000"}
09:33:07 SRC: WebConsole from 10.0.0.2
09:33:07 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"HueMotion","Endpoint":2,"Send":"0402_06/00000029FFFF00000000"}"
09:33:07 ZigbeeZCLSend device: 0xB6CD, group: 0x0000, endpoint:2, cluster:0x0402, cmd:0x06, send:"00000029FFFF00000000"
09:33:07 ZbSend: shortaddr 0xB6CD, groupaddr 0x0000, cluster 0x0402, endpoint 0x02, cmd 0x06, data 
09:33:07 ZIG: ZbZNPSent 240202CDB60000000000000200000102040B301E0D00000B0600000029FFFF00000000
09:33:07 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}
09:33:07 ZIG: Bytes follow_read_metric = 21459
09:33:07 ZIG: {"ZbZNPReceived":"640200"}
09:33:10 ZIG: Bytes follow_read_metric = 21466
09:33:10 ZIG: {"ZbZNPReceived":"448000010B"}
09:33:10 ZIG: Bytes follow_read_metric = 21494
09:33:10 ZIG: {"ZbZNPReceived":"448100000204CDB602010000000413D3000004180B0700CDB61D"}
09:33:10 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1026,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":13832964,"fc":"0x18","manuf":"0x0000","transact":11,"cmdid":"0x07","payload":"00"}}
09:33:10 ZIG: ZbZCLRawReceived: {"0xB6CD":{}}
09:33:10 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":0}}}
09:33:17 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/00000018FFFF000000"}
09:33:17 SRC: WebConsole from 10.0.0.2
09:33:17 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"HueMotion","Endpoint":2,"Send":"0406_06/00000018FFFF000000"}"
09:33:17 ZigbeeZCLSend device: 0xB6CD, group: 0x0000, endpoint:2, cluster:0x0406, cmd:0x06, send:"00000018FFFF000000"
09:33:17 ZbSend: shortaddr 0xB6CD, groupaddr 0x0000, cluster 0x0406, endpoint 0x02, cmd 0x06, data 
09:33:17 ZIG: ZbZNPSent 240202CDB60000000000000200000106040C301E0C00000C0600000018FFFF000000
09:33:17 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}
09:33:17 ZIG: Bytes follow_read_metric = 21499
09:33:17 ZIG: {"ZbZNPReceived":"640200"}
09:33:18 ZIG: Bytes follow_read_metric = 21528
09:33:18 ZIG: {"ZbZNPReceived":"448100000604CDB60201000F00F071D3000005180C0B0680CDB61D"}
09:33:18 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":15,"securityuse":0,"seqnumber":0,"timestamp":13857264,"fc":"0x18","manuf":"0x0000","transact":12,"cmdid":"0x0B","payload":"0680"}}
09:33:18 MQT: tele/Zigbee/RESULT = {"ZbResponse":{"Device":"0xB6CD","Name":"HueMotion","Command":"0406!06","Status":128,"StatusMessage":"MALFORMED_COMMAND","Endpoint":2,"LinkQuality":15}}
09:33:19 WIF: Checking connection...
09:33:24 ZIG: Bytes follow_read_metric = 21535
09:33:24 ZIG: {"ZbZNPReceived":"448000010C"}`

s-hadinger, 2020-04-16:
Can you please try this:

```
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/00000018FFFF0000"}
```

Dr4g0nM4st3r, 2020-04-16:
Now 'Occupancy' also appears, that's what zigbee2mqtt was called. But it seems that the value does not change and no updates are sent. But if something moves in front of the sensor, the LED lights up in red for a short time. Without these definitions, the LED did not react.

`10:58:31 CMD: ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/00000018FFFF0000"}

10:58:31 SRC: WebConsole from 10.0.0.2

10:58:31 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"HueMotion","Endpoint":2,"Send":"0406_06/00000018FFFF0000"}"

10:58:31 ZigbeeZCLSend device: 0xB6CD, group: 0x0000, endpoint:2, cluster:0x0406, cmd:0x06, send:"00000018FFFF0000"

10:58:31 ZbSend: shortaddr 0xB6CD, groupaddr 0x0000, cluster 0x0406, endpoint 0x02, cmd 0x06, data 

10:58:31 ZIG: ZbZNPSent 240202CDB60000000000000200000106040D301E0B00000D0600000018FFFF0000

10:58:31 MQT: stat/Zigbee/RESULT = {"ZbSend":"Done"}

10:58:31 ZIG: Bytes follow_read_metric = 23490

10:58:31 ZIG: {"ZbZNPReceived":"640200"}

10:58:34 ZIG: Bytes follow_read_metric = 23525

10:58:34 ZIG: {"ZbZNPReceived":"448000010D"}

10:58:34 ZIG: Bytes follow_read_metric = 23525

10:58:34 ZIG: {"ZbZNPReceived":"448100000604CDB60201003F002568C8000004180D0700CDB61D"}

10:58:34 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":63,"securityuse":0,"seqnumber":0,"timestamp":13133861,"fc":"0x18","manuf":"0x0000","transact":13,"cmdid":"0x07","payload":"00"}}

10:58:34 ZIG: ZbZCLRawReceived: {"0xB6CD":{}}

10:58:34 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Endpoint":2,"LinkQuality":63}}}

10:58:35 ZIG: Bytes follow_read_metric = 23556

10:58:35 ZIG: {"ZbZNPReceived":"448100000604CDB60201003F005874C800000708040A00001800CDB61D"}

10:58:35 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":63,"securityuse":0,"seqnumber":0,"timestamp":13136984,"fc":"0x08","manuf":"0x0000","transact":4,"cmdid":"0x0A","payload":"00001800"}}

10:58:35 ZIG: ZbZCLRawReceived: {"0xB6CD":{"0406/0000":0}}

10:58:35 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Occupancy":0,"Endpoint":2,"LinkQuality":63}}}

10:58:37 ZIG: Bytes follow_read_metric = 23618

10:58:37 ZIG: {"ZbZNPReceived":"448100000604CDB6020B013C00CD87C800000718050A00001800CDB61D"}

10:58:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":11,"wasbroadcast":1,"LinkQuality":60,"securityuse":0,"seqnumber":0,"timestamp":13141965,"fc":"0x18","manuf":"0x0000","transact":5,"cmdid":"0x0A","payload":"00001800"}}

10:58:37 ZIG: ZbZCLRawReceived: {"0xB6CD":{"0406/0000":0}}

10:58:37 ZIG: Bytes follow_read_metric = 23618

10:58:37 ZIG: {"ZbZNPReceived":"448100000604CDB60201013C00CD87C800000718050A00001800CDB61D"}

10:58:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":1,"LinkQuality":60,"securityuse":0,"seqnumber":0,"timestamp":13141965,"fc":"0x18","manuf":"0x0000","transact":5,"cmdid":"0x0A","payload":"00001800"}}

10:58:37 ZIG: ZbZCLRawReceived: {"0xB6CD":{"0406/0000":0}}

10:58:37 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Occupancy":0,"Endpoint":2,"LinkQuality":60}}}

10:58:37 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Occupancy":0,"Endpoint":2,"LinkQuality":60}}}

10:58:38 ZIG: Bytes follow_read_metric = 23634

10:58:38 ZIG: {"ZbZNPReceived":"45CACDB6705B2903018817000000"}

10:58:38 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x0017880103295B70","ShortAddr":"0xB6CD","ParentNetwork":"0x0000"}}

10:58:38 ZIG: Bytes follow_read_metric = 23651

10:58:38 ZIG: {"ZbZNPReceived":"45C1CDB6CDB6705B29030188170080"}

10:58:38 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x0017880103295B70","ShortAddr":"0xB6CD","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}

10:58:38 ZIG: ZbZNPSent 2505CDB6CDB6

10:58:38 ZIG: Bytes follow_read_metric = 23656

10:58:38 ZIG: {"ZbZNPReceived":"650500"}

10:58:39 WIF: Checking connection...

10:58:39 ZIG: Bytes follow_read_metric = 23668

10:58:39 ZIG: {"ZbZNPReceived":"4585CDB600CDB6020201"}

10:58:39 MQT: tele/Zigbee/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x02","0x01"]}}

10:58:39 ZIG: ZbZNPSent 2401CDB6020100000E301E07000E0004000500

10:58:39 ZIG: Bytes follow_read_metric = 23673

10:58:39 ZIG: {"ZbZNPReceived":"640100"}

10:58:39 ZIG: Bytes follow_read_metric = 23723

10:58:39 ZIG: {"ZbZNPReceived":"448100000000CDB60201003F002CAAC800001A180E0104000042075068696C6970730500004206534D4C303031CDB61D"}

10:58:39 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0xB6CD","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":63,"securityuse":0,"seqnumber":0,"timestamp":13150764,"fc":"0x18","manuf":"0x0000","transact":14,"cmdid":"0x01","payload":"04000042075068696C6970730500004206534D4C303031"}}

10:58:39 ZIG: ZbZCLRawReceived: {"0xB6CD":{"0000/0004":"Philips","0000/0005":"SML001"}}

10:58:39 MQT: tele/Zigbee/SENSOR = {"ZbReceived":{"HueMotion":{"Device":"0xB6CD","Manufacturer":"Philips","ModelId":"SML001","Endpoint":2,"LinkQuality":63}}}

10:58:45 ZIG: Bytes follow_read_metric = 23730

10:58:45 ZIG: {"ZbZNPReceived":"448000010E"}

10:58:59 WIF: Checking connection...
`

s-hadinger, 2020-04-16:
The one more thing zigbee2mqtt is doing is setting the occupancy_timeout parameter. Let me see how to do this. This controls the time between occupancy:true and when the device sends occupancy:false.

But here it seems to never send occupancy:true.

s-hadinger, 2020-04-16:
Can you try this one, it will force to report at least once per minute:

```
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/0000001800003C00"}
```

Dr4g0nM4st3r, 2020-04-16:
this command causes the esp to restart. 
the last log before the crash -> 
`11:29:39 WIF: Checking connection...
11:29:57 ZIG: Bytes follow_read_metric = 24811
11:29:57 ZIG: {"ZbZNPReceived":"448100000604CDB602010044001B3123000007080A0A00001801CDB61D"}
11:29:57 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,`

s-hadinger, 2020-04-16:
Actually this is good news. There was a bug causing a crash when receiving `"Occupancy":1` and it was fixed with #8096. Recompile with the latest version.

Currently a synthetic event is created 90 seconds after the last `"Occupancy":1` event was received. I need to remove this if in between a `"Occupancy":0` was received. So don't be surprised if you receive several of these events.

Dr4g0nM4st3r, 2020-04-16:
I switched to DEV Branch and use your last command. Now its working. thank you 👍

s-hadinger, 2020-04-16:
Ok, for the record, here are the required commands:

```
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0001"}
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0400"}
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0402"}
ZbBind {"Device":"HueMotion","Endpoint":2,"ToDevice":"0x0000","ToEndpoint":1,"Cluster":"0x0406"}

ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0001_06/00210020FFFF000000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0400_06/00000021FFFF00000000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0402_06/00000029FFFF00000000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/0000001800003C00"}
```

Dr4g0nM4st3r, 2020-04-16:
I think I found something else. 
The brightness doesn't seem to update correctly 'Illuminance', here I still have the old initial value.

Dr4g0nM4st3r, 2020-04-16:
And Temperature/BatteryPercentage are also the old values from 4 hours ago.

s-hadinger, 2020-04-16:
Then let's apply the same rule as with zigbee2mqtt (it sets minimal interval to 0 and max to 1 hour):

```
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0001_06/00210020100E30F200"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0400_06/000000210000100E0000"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0402_06/000000290000100E1900"}
ZbSend {"Device":"HueMotion","Endpoint":2,"Send":"0406_06/000000180000100E"}
```

Dr4g0nM4st3r, 2020-04-16:
Now it works, i will test it for 24h and then response here.

closed 2020-04-17 by arendst

uberboffin, 2021-09-13:
I am having trouble with both the Philips indoor (SML001) and outdoor (SML002) sensors. I am able to get them to pair and successfully set the reporting attributes, but after a time the devices start to fail to report (and also flash their internal red LED). They sometimes recover, but this is not always the case.

Typically at the time when the device starts to misbehave the following is reported in the log:
`{"ZbState":{"Status":34,"IEEEAddr":"0x001788010648FCDC","ShortAddr":"0xB504","ParentNetwork":"0x0000","JoinStatus":3,"Decision":0}}`

The problem is made worse if I reboot the bridge, when it can take a considerable time for the devices to start reporting again - very different from all other Zigbee devices I have used with the Sonoff bridge.

I am wondering whether these devices require some sort of heartbeat or acknowledgement to keep operating? If I pair these devices with the Philips hub, they work flawlessly, as expected.

**Before I log an issue, I wanted to know if anyone else uses these sensors and if they have seen this behaviour?**

I am using the latest build (9.5.0.8) and the latest Zigbee firmware (ncp-uart-nsw_6.7.9_115200.ota).

PR #8202 (2020-04-17): Disarm Zigbee timer for Occupancy:0
## Description:

By default Z2T arms a timer of 90 seconds after receiving `"Occupancy":1` and generates a synthetic event `"Occupancy":0`. This is required for Aqara presence sensor.

Philips Hue Motion Sensor does send `"Occupancy":0`. This PR disarms the timer when `"Occupancy":0` is received within 90 seconds.

**Related issue (if applicable):** fixes #8192 

## Checklist:
  - [ ] The pull request is done against the latest dev branch
  - [ ] Only relevant files were touched
  - [ ] Only one feature/fix was added per PR.
  - [ ] The code change is tested [... cut]