arendst/Tasmota #8194: I2C: Sensor HDC1080 dose not show data

BrainGameFactory, 2020-04-15:
### PROBLEM DESCRIPTION
This is a complete new Flash on a new D1 mini.
You can see in the in the Screenshot that the sensor is recognized but dose not show any Data.
I really tried to get working but no success but it coud be my fault.

The sensor is also officially supported:
https://github.com/arendst/Tasmota/releases 
"Add support for HDC1080 Temperature and Humidity sensor by Luis Teixeira (#7888)"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [✓] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [✓] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [✓] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [✓] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [✓] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [✓] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [✓] Tasmota binary firmware version number used: Tasmota v8.2.0 Elliot
  - [✓] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: 
- [✓] Flashing tools used:ESP Easy Flasher
- [✓] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:11:43 RSL: stat/tasmota_C0EA42/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:11:43 RSL: stat/tasmota_C0EA42/RESULT = {"Module":{"0":"Generic"}}
19:11:43 RSL: stat/tasmota_C0EA42/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"5":"I2C SCL"},"GPIO5":{"6":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [✓] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:12:38 RSL: stat/tasmota_C0EA42/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_C0EA42","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:04","StartupUTC":"2020-04-15T16:57:34","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2020-04-15T16:36:46","SaveCount":30,"SaveAddress":"F8000"}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020.03.20 14:54:51","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"315/699"}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-Link_B2CF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS5 = {"StatusNET":{"Hostname":"tasmota_C0EA42-2626","IPAddress":"192.168.1.184","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"EC:FA:BC:C0:EA:42","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C0EA42","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS7 = {"StatusTIM":{"UTC":"2020-04-15T17:12:38","Local":"2020-04-15T19:12:38","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:58","Sunset":"20:42"}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS10 = {"StatusSNS":{"Time":"2020-04-15T19:12:38"}}
19:12:38 RSL: stat/tasmota_C0EA42/STATUS11 = {"StatusSTS":{"Time":"2020-04-15T19:12:38","Uptime":"0T00:15:04","UptimeSec":904,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TP-Link_B2CF","BSSId":"98:DA:C4:73:B2:CF","Channel":10,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE

Connect a HDC1080 Sensor and thats it.

### EXPECTED BEHAVIOUR
Show the Temp. and Hummidity in the Menu.

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-04-15:
This sensor is not present on pre-built binaries.
To have it you must compile your own firmware enabling USE_HDC1080 on `user_config_override.h`.
More information about supported sensors [here](https://github.com/arendst/Tasmota/blob/development/BUILDS.md).

BrainGameFactory, 2020-04-15:
Oh thanks. I knew it's was my fault.
I try it and than I close the issue.

BrainGameFactory, 2020-04-15:
I dont know if i doing something wrong again but with my own compiled version does it also not work. Atleast the I2C Driver for the HDC1080 is now active.
Screenshot:
[image]
https://tasmota.github.io/docs/I2CDevices/ 
"45 | USE_HDC1080 | xsns_65 | HDC1080 | 0x40"

teixeluis, 2020-04-16:
Hi @BrainGameFactory, could you provide again the output of the STATUS 0 command (running your compiled firmware)?

Thanks

BrainGameFactory, 2020-04-16:
@teixeluis Here you go:
20:53:49 CMD: STATUS 0
20:53:49 SRC: WebConsole from 192.168.1.102
20:53:49 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
20:53:49 RSL: stat/tasmota_C0EA42/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_C0EA42","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:37","StartupUTC":"2020-04-16T18:53:12","Sleep":50,"CfgHolder":4617,"BootCount":30,"BCResetTime":"2020-04-15T16:36:46","SaveCount":50,"SaveAddress":"F6000"}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020.04.15 19:55:14","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"315/699"}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-Link_B2CF",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS4 = {"StatusMEM":{"ProgramSize":580,"Free":420,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","000000CD","010013C0","C000F981","0000000C"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,65"}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS5 = {"StatusNET":{"Hostname":"tasmota_C0EA42-2626","IPAddress":"192.168.1.184","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"EC:FA:BC:C0:EA:42","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C0EA42","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS7 = {"StatusTIM":{"UTC":"2020-04-16T18:53:49","Local":"2020-04-16T20:53:49","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:56","Sunset":"20:43"}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS10 = {"StatusSNS":{"Time":"2020-04-16T20:53:49"}}
20:53:49 RSL: stat/tasmota_C0EA42/STATUS11 = {"StatusSTS":{"Time":"2020-04-16T20:53:49","Uptime":"0T00:00:37","UptimeSec":37,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TP-Link_B2CF","BSSId":"98:DA:C4:73:B2:CF","Channel":10,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:06"}}}

flashy02, 2020-04-18:
Hi, I've tested the HDC1080 with the latest Release 8.2.0, but there is a bug in humidity conversion. At the Moment there is the same calculation factor for temp and hum:

#define HDC1080_TEMP_MULT   0.0025177
#define HDC1080_RH_MULT     0.0025177

but it should be:

#define HDC1080_TEMP_MULT   0.0025177
#define HDC1080_RH_MULT     0.0015258

flashy02, 2020-04-18:
@BrainGameFactory ,if you want, I can send you my compiled version which works here on a Sonoff SV. Here I've configured RX/TX as I2C (GPIO1 = SDA and GPIO3 = SCL) for testing.

ascillato2, 2020-04-20:
@flashy02 

Hi, please, can you provide a PR with that correction? Thanks.

BrainGameFactory, 2020-04-20:
@flashy02 
Sorry for the late answer. I would be great if you can send me you version.

PaulchenPlump, 2020-04-23:
I involuntarily have a similar problem: I updated my Sonoff T1 to version 8.2 and since then the display has also disappeared. The sensor is a BME280 and was connected via I2C. The sensor is recognized and worked correctly in older versions. It was updated via the web interface and the OTA-URL "http://thehackbox.org/tasmota/release/tasmota-DE.bin". After a downgrade to 8.1.0 everything works fine again.
Is there a problem with the version or I am doing something wrong?

ascillato2, 2020-04-26:
@PaulchenPlump 

Hi, as the first comment, this sensor is not available on **_tasmota-DE.bin_** nor **_tasmota.bin_**

> This sensor is not present on pre-built binaries.
> To have it you must compile your own firmware enabling USE_HDC1080 on `user_config_override.h`.
> More information about supported sensors [here](https://github.com/arendst/Tasmota/blob/development/BUILDS.md).
> 

@flashy02

Thanks for reporting that conversion issue. It has been fixed.

@BrainGameFactory

If you need help on compiling your own version, please, check the docs (you can use online compilers like gitpod or tasmocompiler) or ask for help on the Tasmota Support Chat.
Thanks.

closed 2020-04-26 by ascillato2