arendst/Tasmota #8197: [REQUEST] Tasmota/Zigbee2Tasmota firmware on the Sonoff Zigbee Bridge (Sonoff ZBBridge)

Hedda, 2020-04-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Please consider porting Tasmota/Zigbee2Tasmota to the new Sonoff Zigbee Bridge (Sonoff ZBBridge)

https://www.itead.cc/sonoff-zbbridge.html

Itead has just launched Sonoff ZBBridge as an inexpensive Sonoff Zigbee Bridge which is based on a similar design as the Sonoff RF Bridge 433

https://www.itead.cc/sonoff-zbbridge.html

According to the teardown on notenoughtech.com it sounds as if it is based on Silicon Labs EFR32 (Mighty Gecko) for Zigbee 3.0 radio module support and ESP8266 for WiFi and bridge/gateway/controller software

https://notenoughtech.com/home-automation/sonoff/sonoff-zigbee-bridge-preview/

From the images it even looks like they are reusing the same injection moulds as for the Sonoff RF Bridge 433 housing/enclosure 

https://sonoff.tech/product/accessories/433-rf-bridge

https://www.itead.cc/sonoff-rf-bridge-433.html

https://www.itead.cc/wiki/Sonoff_RF_Bridge_433

EZSP serial protocol interface that Silicon Labs uses is also well documented and already used by open source projects, see example Home Assistant's ZHA integration component via zigpy and bellows:

https://github.com/zigpy/bellows  

https://www.home-assistant.io/integrations/zha/  

https://tasmota.github.io/docs/Zigbee/

As you know, Tasmota does something similar with Zigbee2Tasmota but by connecting an ESP8266 to a Texas Instrument CC2530 module instead however it too is using a serial communication protocol, see:

https://tasmota.github.io/docs/Zigbee/

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
The alternative today is to build your own Zigbee2Tasmota bridge with a ESP8266 and a TI CC2530 module, but that is not a complete hardware solution and the Zigbee module included in Sonoff ZBBridge is much more powerful than TI CC2530.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

timdonovanuk, 2020-04-16:
Seconded! The actual name of this device is Sonoff BASICZBR3.

Jason2866, 2020-04-16:
Since Sonoff Zigbee Bridge uses a chip where no Open Source Router firmware variant exists,
there will be no support for Zigbee2Tamsota

s-hadinger, 2020-04-16:
I would love to make it possible. I didn't know the protocol was documented and there was an exiting implementation (bellows). Let me try to grab one of those and see if it's possible or not.

Ideally having Z-Stack running on the same chip would be easier, but not possible.

Are you sure the Sonoff Zigbee Bridge uses EZSP protocol?

Hedda, 2020-04-16:
> Seconded! The actual name of this device is Sonoff BASICZBR3.

No the name of the new Sonoff Zigbee Bridge is "Sonoff ZBBridge". 

Sonoff BASICZBR3 is just the Zigbee version if Sonoff BASIC BR3.

Hedda, 2020-04-16:
> Are you sure the Sonoff Zigbee Bridge uses EZSP protocol?

What I am sure of is that the Zigbee radio module is EFR32MG21 (EFR32 Mighty Gecko) from Silicon Labs and that features EZSP Bootloader and has what they call EZSP interface (short for "EmberZNet Serial Protocol")  that seems to be well documented.

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2017/05/25/build_an_ezsp-spiho-2VE8

https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

Not sure about the version of EZSP serial protocol but that is what EFR32 uses by default for Zigbee. I would guess that Itead is using the latest EmberZNet SDK on their EFR32MG21 as it supports a Zigbee 3.0 stack. 

I don't know but would assume that the bellows library is probably using an older version of the EZSP serial protocol interface as it lists devices that are using EmberZNet PRO ZigBee Protocol Stack.

https://github.com/zigpy/bellows

bellows is at least very stable using it through zigpy with Home Assistant's ZHA component

timdonovanuk, 2020-04-16:
Ah sorry, my mistake. I'm not sure why you'd want Tasmota running on what is essentially a hub. Unless I'm misunderstanding, Tasmota is for controlling end point devices (such as BASICZBR3). Isn't this like asking for Tasmota for an Intel Nuc? If you want a more powerful opensource bridge zigbee2mqtt supports more powerful zigbee sticks than just CC253x now.

meingraham, 2020-04-16:
A similar device is the Sonoff RF Bridge... which is a Tasmota supported device. Changing the Wi-Fi portion of the Zigbee bridge would likely expand the capabilities of the OEM device and add all of the Tasmota features (inter-device communications & control, rules, etc.).

hgelpke, 2020-04-16:
Based on the preview video from NotEnoughTech on youtube, it also seems locked to their zigbee devices and doesn't have third party support a la deconz or z2m. Opening it up would add an ideal open source router for cheap.

Hedda, 2020-04-17:
Yes, I want to use Tosmota on Sonoff ZBBridge and use it as a Zigbee bridge similar to Zigbee2mqtt.

It is already possible today to use Tosmota as a Zigbee bridge via the Zigbee2Tosmota subproject by connecting it to a Texas Instruments CC2530 Zigbee module (instead of a sensor) to control Zigbee devices and act as a Zigbee 2 MQTT bridge (practically giving it the same function as Zigbee2mqtt):

See https://tasmota.github.io/docs/Zigbee/ and 
https://www.cnx-software.com/2019/12/23/tasmota-open-source-firmware-now-supports-zigbee-connectivity/

Sonoff ZBBridge is basically a prepackaged hardware solution for that same function. The main difference is that Sonoff ZBBridge does not use a CC2530 (Zigbee HA 1.2 chip from Texas Instruments CC253x series) radio module but instead an EFR32MG21 (Zigbee 3.0 chip from Silicon Labs EFR32 series) radio module. The Zigbee stack on these chips has an API that can be controlled via serial communication but they do use different serial protocol.

So this is really, in essence, a request to a Zigbee2Tosmota subproject to add support for the different serial protocol that Silicon Labs EFR32 series uses to control the Zigbee stack on it.

zigpy organization has open-source Python libraries which could maybe be ported to achieve this?

- https://github.com/zigpy/

bellows is a library that supports serial communication with Silicon Labs chips:
- https://github.com/zigpy/bellows

zigpy-cc & zigpy-znp are libraries that supports serial communication with Texas Instruments chips:
- https://github.com/zigpy/zigpy-cc
- https://github.com/zha-ng/zigpy-znp

They also have libraries which support serial communication with XBee, ZiGate & deCONZ modules:
- https://github.com/zigpy/zigpy-xbee
- https://github.com/zigpy/zigpy-zigate
- https://github.com/zigpy/zigpy-deconz

The zigpy library itself can abstract all these radio libraries for higher-level support in a unified way: 
- https://github.com/zigpy/zigpy

PS: Again, ut is also already possible to use Tasmota as a RF Bridge on the somewhat similar device models called "Sonoff RF Bridge" which let Tasmota act as a RF Bridge for 315MHz and 433MHz RF devices. So these types Sonoff RF bridges devices have also already been 'hacked' to run Tasmota.

See https://tasmota.github.io/docs/Commands/#rf-bridge and https://sonoff.tech/product/accessories/433-rf-bridge

Hedda, 2020-04-17:
> Based on the preview video from NotEnoughTech on youtube, it also seems locked to their zigbee devices and doesn't have third party support a la deconz or z2m. 

That is not true. It should be noted that NotEnoughTech did his review on an early preview version. Itead has already stated that this Zigbee bridge will also support third-party Zigbee devices as well. Sure that might mean that more and more devices get added with more and more future software upgrades, but that is also the case with any Zigbee bridge, including the existing Zigbee2Tosmota subproject, Zigbee2mqtt, and deCONZ software.

Hedda, 2020-04-17:
> Since Sonoff Zigbee Bridge uses a chip where no Open Source Router firmware variant exists,
> there will be no support for Zigbee2Tasmota

Not sure what you mean. Sorry but I don't think you have the full picture about either of these modules. Using these Silicon Labs chips should be the same scenario as using Texas Instruments chip with open source software like Tasmota. 

There is no open-source router firmware for Texas Instruments chips either, instead, you flash them via a firmware which consists of binary blobs and config files. This is the same for the Silicon Labs chips. 

The firmware is usually produced by using an SDK, and again this is normally the same for Texas Instruments chips and Silicon Labs chips.  The SDK pack the binary blobs (which has Zigbee stack) and the config files (that to simplify as just text files with a configuration of how to use the stack).

Then to communicate with these chips you use a UART or serial interface. Here is what is practically different when using the other manufacturer of Zigbee chips as all manufacturers use different serial protocols, however, both these manufacturers do a good job at documenting those protocols.

Anyway, Silicon Labs has a good history of working with open-source software. So I think it is way to early to say "there will be no support for Zigbee2Tasmota" unless you just say so on principle.

Jason2866, 2020-04-17:
@Hedda  from TI the zigbee stack is free availiable. The zigbee stack from Silicon labs is NOT free available. All you said is not proven. You dont know if ITEAD uses EZSP protocol or maybe a variant
or something completely different. I dont understand the comparision between RF Bridge and the Zigbee Bridge. The device do have nothing in common expect the case and using a ESP82xx chip for wifi.  NotEnoughTech just did a unboxing.
To be clear at the moment we just know what chips are inside NOTHING more.
Everything else are just assumptions what might be, could be...
We are going to try to get some. Maybe! something is possible....

Hedda, 2020-04-17:
@Jason2866 Yes, I mean; could be, maybe, is possible, ...but definitely not "there will be no support for Zigbee2Tasmota" unless someone like yourself shoots down the idea on pure principle before proven.

Hedda, 2020-04-20:
One workaround for Home Assistant users could be to simply present the Zigbee radio module as a remote adapter to Home Assistant's ZHA (their Zigbee Home Automation integration component)?

- https://www.home-assistant.io/integrations/zha/

You could use something like ser2net to allow ZHA to remote connect to it via serial over network, like:

- https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Then you need a radio library for zigpy that works with the EZSP protocol used by Silicon Labs EFR32 

- https://github.com/zigpy/zigpy

Hopefully, their bellows radio library should already be compatible or could be made to be if updated

- https://github.com/zigpy/bellows

zigpy devs already do something similar to connect remotely with ESP8266 based "ZiGate Pack WiFi"

- https://zigate.fr/produit/zigate-pack-wifi-v1-3/
- https://github.com/fairecasoimeme/ZiGate/
- https://github.com/zigpy/zigpy-zigate

Jason2866, 2020-04-20:
@Hedda  why? There is no need for a workaround. Zigbee2Tasmota works fine

Hedda, 2020-04-21:
@Jason2866 This is a request for Tasmota support on Sonoff ZBBridge, and no, Zigbee2Tasmota does not currently work fine with the Silicon Labs EFR32 based Zigbee radio module inside the Sonoff ZBBridge. For Tasmota to be able to utilize that Silicon Labs EFR32 based Zigbee radio module inside the Sonoff ZBBridge it would either need to implement a native library for the EZSP (EmberZNet Serial Protocol) serial protocol communication with that or it could as a workaround implement a serial port forwarding feature which could forward the serial port to example the bellows library running on the same computer as Home Assistant with ZHA.

Hedda, 2020-04-22:
Almost any ESP8266 development board (like a NodeMCU or a Wemos D1 Mini) with a Silicon Labs EFR32MG based Zigbee module like the Ebyte E180-ZG120B would just about be about the equivalent to having a hacked Sonoff ZBBridge, so such a setup could be used as a development environment.

E180-ZG120B (and the older E180-ZG120A) modules are sold on eBay and Aliexpress at low prices. 

- https://www.aliexpress.com/wholesale?catId=0&SearchText=%22E180-ZG120B%22

Example: 

- https://www.aliexpress.com/item/4000473346234.html

Ebyte also makes an inexpensive development board called "E180-ZG120B-TB" made for testing it. 

- http://www.ebyte.com/en/product-view-news.aspx?id=896
  - http://www.ebyte.com/en/downpdf.aspx?id=896
- http://www.cdebyte.net/test-board/zigbee-3-0-module-usb-test-board-kits.html

This development is sold for less than $9 US-dollar on Aliexpress or about twice that on eBay UK:

- https://www.aliexpress.com/item/4000538003718.html
- https://www.ebay.co.uk/itm/EFR32-Zigbee-3-0-Test-Board-USB-Port-2-4GHZ-E180-ZG120B-TB-Test-Kit/233517910970

Note! Ebyte is now making two EFR32MG12 based modules called E180-ZG120A and E180-ZG120B

- http://www.ebyte.com/en/product-view-news.aspx?id=842
  - http://www.ebyte.com/en/downpdf.aspx?id=842
  - http://www.ebyte.com/en/pdf-down.aspx?id=161

The downside to using such setup instead of Sonoff ZBBridge is that the E180-ZG120B module will likely not come preloaded with firmware so have to build EmberZNet PRO Zigbee Stack firmware with coordinator device type config and flash firmware to the E180-ZG120B module first.

Jason2866, 2020-04-22:
I know the Ebyte modules. This does not help anything, because we have no access to the
EmberZNet PRO Zigbee Stack. It is only available if you buy the 500$ development board.
There is a chance to get the Ebyte modules going with zigbee2Tasmota (if someone does the work)
This does not help with your request since we still dont know anything about Sonoff Zigbee Bridge
used Zigbee Stack nor can we flash a different stack since there is simply none

Hedda, 2020-04-23:
If there is a will then there is a way. For those interested in a more friendly discussion about the same concept but in an alternative firmware, check out https://github.com/xoseperez/espurna/issues/2224

s-hadinger, 2020-04-23:
Thanks @Hedda 
I have ordered mine from Itead but they announce a shipping date not before May 15. We can continue the discussion when we receive them and take a look at the actual firmware version in the EFR32 MCU.

Hedda, 2020-04-28:
FYI @SillyDay don't own a E180-ZG120B but has tried to build a firmware for it as per discussion here:

https://github.com/Koenkk/zigbee-herdsman/issues/168

SillyDay GitHub repo for EFR32 firmware:

https://github.com/SillyDay/EFR32

Hedda, 2020-05-05:
FYI, as per related request https://github.com/zigpy/zigpy/issues/405 I just learned that there is an existing ESP8266-based networked-attached Zigbee-adapter called "ZiGate Pack WiFi adapter" which has a new v2.0 firmware that archives the requested function of a UART-to-TCP/IP (for Serial-port to WiFi-bridge function) using "ESP-LINK from Jeelab" software, in addition, using ESP-LINK also adds mDNS zeroconf to allow automatic network discovery and configuration:

- https://github.com/jeelabs/esp-link

Description of functions that using ESP-LINK will add to v2.0 firmware for ZiGate Pack WiFi adapter:

https://translate.google.com/translate?sl=fr&tl=en&u=https%3A%2F%2Fzigate.fr%2Fdocumentation%2Fdescription-du-firmware-v2-xx%2F

- ZiGate TCP 9999 to UART WiFi Bridge
- WiFi configuration in client and access point mode
- DHCP client or static IP
- Client syslog
- SNTP client
- mDNS
- MQTT client
- WiFi firmware update (OTA)
- Console for logging and validating the operation of the ZiGate

As I understand it, all ZiGate hardware look to be modular in design and the "ZiGate Pack WiFi adapter" is really just an optional ESP8266 based "dumb" UART-to-TCP/IP (for Serial-port to WiFi-bridge function) for the standard "ZiGate TTL adapter" that allows users to connect to it remotely using TCP/IP over your home LAN (Local Area Network) instead of plugging it directly to your computer via the optional USB adapter. 

- https://zigate.fr/produit/module-wifi-v1-3-compatible-zigate/

Specifically, please see the picture of "ZiGate Pack WiFi adapter" https://zigate.fr/produit/zigate-pack-wifi-v1-3/ compared to the picture of "ZiGate TTL USB adapter" https://zigate.fr/produit/zigate-ttl/

Thus "ZiGate Pack WiFi adapter" allows ZHA users to have a networked Zigbee adapter setup like this:

ZHA <–> zigpy/zigpy-zigate <–> TCP/IP over LAN <–> ZiGate-WiFi <–> UART <–> ZiGate Radio

Suggesting this now as I just learned from @doudz there that the new v2.0 version of the ZiGate Pack WiFi adapter firmware contains "ESP-LINK from Jeelab" software which among other things adds mDNS and  UART WiFi Bridge support over TCP. As I understand, version v1.x of the firmware for the ZiGate Pack WiFi adapter basically only contained a simple UART/serial-port server forwarding service ( serial server software that just acts as a dumb Zigbee to WiFi bridge for zigpy-zigate), while the new version v2.x also has more advanced features (which does not need to used) it still also contain a simple UART/serial-port server forwarding service, but now mDNS also makes it easier to discover the adapter on your local network.

Now it would be awesome if the ZHA integration component for Home Assistant from an end-user perspective supported just as an easy detection and configuration of network networked-attached Zigbee coordinator adapters, like the Sonoff ZBBridge and the ZiGate Pack WiFi adapter. 

I would therefore also suggest using some kind of [Zero-configuration networking (zeroconf)](https://en.wikipedia.org/wiki/Zero-configuration_networking) method, like for example [mDNS](https://en.wikipedia.org/wiki/Multicast_DNS), (as mDNS is already in use in Home Assistant Core), to make the ZHA integration component for Home Assistant automatically detect, connect, and configure compatible networked-attached Zigbee coordinator adapters like the "ZiGate Pack WiFi adapter" as that is otherwise already supported by the zigpy-zigate radio library for zigpy.

s-hadinger, 2020-05-05:
Thanks for this detailed information. However I'm not sure to understand in what way it is of Tasmota concern. The devics above are much more expensive than a Wemos D1 + CC2530. 

I'm still looking forward to playing with Sonoff Zb bridge as soon as they start shipping.

Hedda, 2020-05-05:
For starters, CC2530 based modules do not support Zigbee 3.0 (CC253x only support Zigbee 1.2).

For Zigbee 3.0 you need CC2538, CC26X2, CC13x2 if Texas Instruments, or EFR32 if Silicon Labs.

Then you forgot to add up the total cost for ESP8266 + Zigbee 3.0 module + power-supply + box.

So if comparing apples and oranges then cheapest DIY solution based on Texas Instruments example: 
- Wemos D1 Mini 
- CC2538 (+ CC2592 PA) Zigbee Wireless Module
- USB power supply 
- Enclosure (chassis/box)

You might get a complete CC2538 based DIY solution for a little less but it is going to be hard to compete when the finished hardware that Itead is selling Sonoff ZBBridge just cost $16.90 US-dollars.

CC2538 does, however, contain a less powerful radio than Silabs EFR32MG21 used in Sonoff ZBBridge.

Comparing apples and apples then cheapest DIY solution based on Texas Instruments example: 
- Wemos D1 Mini 
- CC2652R Zigbee Wireless Module
- USB power supply 
- Enclosure (chassi/box)

But again CC2652R still has a less powerful radio than EFR32MG21, so really need a CC2652P or CC1352P module (P for PA = Power Amplifier) to compete with the signal strength of the EFR32MG21.

s-hadinger, 2020-05-06:
I was comparing to the Zigate modules from your last message, not to Sonoff Zigbee bridge.

ascillato2, 2020-05-30:
Closing in favour of https://github.com/arendst/Tasmota/issues/8583

closed 2020-05-30 by ascillato2

shanekuz, 2020-11-11:
Hi everyone, im having an issue with my Sonoff ZBbridge posting MQTT to OPENHAB using the JSONPATH method.

I have a number of devices registered to the bridge but they all send updates on the same MQTT publish path.

When it publishes on this path
tele/tasmota_zbGateway1/SENSOR = {“ZbReceived”:{“0x7294”:{“Device”:“0x7294”,“Name”:" Office_Temp",“Temperature”:23.88,“Humidity”:65.49,“Endpoint”:1,“LinkQuality”:157}}}

I get my temperature updated in openhab and it works fine until another device updates.

tele/tasmota_zbGateway1/SENSOR = {“ZbReceived”:{“0xFF6C”:{“Device”:“0xFF6C”,“Name”:" Wardrobe_PIR",“Illuminance”:6,“Occupancy”:1,“Endpoint”:1,“LinkQuality”:13}}}

Then OPENHAB sets the first value to NULL as it received the JSON payload but it had no entry for that device.

Does anyone have a workaround or maybe can we publish all states of all devices when the gateway publishes anything as a fix?  
Maybe this could be a option you turn on as others will have this issue also im picking.

ascillato, 2020-11-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)