arendst/Tasmota #8207: BH1750 shows errorneous values after a while; restart fixes it

nagyrobi, 2020-04-17:
### PROBLEM DESCRIPTION
BH1750 shows errorneous values after a while; reboot fixes it.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini with BH1750 and VEML6070 on I2C
- [x] Tasmota binary firmware version number used: 8.2.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:09:43 MQT: stat/wemospadlas/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:09:43 MQT: stat/wemospadlas/RESULT = {"Module":{"18":"Generic"}}
15:09:44 MQT: stat/wemospadlas/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
15:10:24 MQT: stat/wemospadlas/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":476,"Rules":"ON System#Boot DO Sensor10 31 ENDON"}
15:10:24 MQT: stat/wemospadlas/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
15:10:24 MQT: stat/wemospadlas/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
15:10:58 MQT: stat/wemospadlas/STATUS = {"Status":{"Module":18,"FriendlyName":["Padlás Sensors"],"Topic":"wemospadlas","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:10:58 MQT: stat/wemospadlas/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:37","StartupUTC":"2020-04-17T14:04:21","Sleep":50,"CfgHolder":4617,"BootCount":51,"BCResetTime":"2020-03-27T14:14:04","SaveCount":84,"SaveAddress":"F6000"}}
15:10:59 MQT: stat/wemospadlas/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(45e9408-sensors)","BuildDateTime":"2020-04-11T15:02:08","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"390/699"}}
15:10:59 MQT: stat/wemospadlas/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ZACUT",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:10:59 MQT: stat/wemospadlas/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000072"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67"}}
15:10:59 MQT: stat/wemospadlas/STATUS5 = {"StatusNET":{"Hostname":"wemospadlas-3578","IPAddress":"172.22.215.38","Gateway":"172.22.215.254","Subnetmask":"255.255.255.0","DNSServer":"172.22.215.254","Mac":"BC:DD:C2:63:ED:FA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:10:59 MQT: stat/wemospadlas/STATUS6 = {"StatusMQT":{"MqttHost":"172.22.215.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_63EDFA","MqttUser":"hamqt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:10:59 MQT: stat/wemospadlas/STATUS7 = {"StatusTIM":{"UTC":"2020-04-17T14:10:59","Local":"2020-04-17T15:10:59","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:54","Sunset":"19:45"}}
15:10:59 MQT: stat/wemospadlas/STATUS10 = {"StatusSNS":{"Time":"2020-04-17T15:10:59","BH1750":{"Illuminance":9999},"VEML6070":{"UvLevel":3025,"UvIndex":4.05,"UvIndexText":"Mid","UvPower":0.101},"DHT12":{"Temperature":280.5,"Humidity":6.3,"DewPoint":149.1},"TempUnit":"C"}}
15:10:59 MQT: stat/wemospadlas/STATUS11 = {"StatusSTS":{"Time":"2020-04-17T15:10:59","Uptime":"0T00:06:38","UptimeSec":398,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"ZACUT","BSSId":"02:9F:C2:A4:1A:28","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
### TO REPRODUCE
Use a BH1750 for longer periods (several days) exposed to direct sunlight.
After a 2-3 days, the sensor starts reporting unreal values, for example in bright sunlight it reports 23lx.
Restarting Tasmota (from the web interface) re-sets the unit in a working state - values are good again.

I suspect there's an issue with continuous operation mode. In the [docs](https://github.com/claws/BH1750/blob/master/README.md) there's a one-time measurement mode which seems to reinitialize the measurement each time it is called. Since there's no actual use of the continuous mode in Tasmota, I'd suggest to try to achieve more stability by triggering to one-time measurements at `TelePeriod`.

### EXPECTED BEHAVIOUR
Have consistent values without needing to reboot.

### SCREENSHOTS
Not required

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-17:
When rules are compiled in every sensor is at least requested to report a value once every three seconds simple to trigger a rule. So changing the BH1750 continues mode to one-time mode is not an option.

There must be a better solution to your unreal values.

nagyrobi, 2020-04-17:
OK, I'm open to more investigation... For the moment I created a rule to automatically restart Tasmota every night.

nagyrobi, 2020-04-18:
What feedback do you need?

device111, 2020-04-19:
can you log the errors from  "Wire.endTransmission()"?

device111, 2020-04-19:
i have a BH1750 and will test for a couple of days..

nagyrobi, 2020-04-19:
Hold your horses. This morning i realized that it might be because I've connected [a second BH1750 on the same i2c bus](https://github.com/arendst/Tasmota/issues/8139) hoping that I'll be able to read them both being on different addresses. 

I suspect the driver swapped from one sensor to the other without and warning. I removed the second sensor now (it was soldered on).

device111, 2020-04-19:
ok, then can be  closed. :-)

nagyrobi, 2020-04-19:
For now I separated the two BH1750 sensors to different Wemos D1 units. I'll compare results, close the issue if after a week no more problematic occurrences.

ascillato2, 2020-04-20:
Great.
Thanks for reporting. So far only one BH1750 is supported. The request for modifying the driver is open.

closed 2020-04-20 by ascillato2