arendst/Tasmota #8215: Sonof rf bridge tasmoto 8.2 no webinterface after the initial setup

dimitrybradt, 2020-04-18:
### PROBLEM DESCRIPTION
Sonof rf bridge tasmoto 8.2 no webinterface after the initial setup

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:59:12 CMD: status 0
11:59:12 RSL: stat/tasmota_E4B421/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"tasmota_E4B421","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:03:46","StartupUTC":"2020-04-18T10:55:26","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-04-18T11:18:12","SaveCount":16,"SaveAddress":"F4000"}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"328/699"}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DenK","DenKBoven"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS5 = {"StatusNET":{"Hostname":"tasmota_E4B421-5153","IPAddress":"10.20.200.247","Gateway":"10.20.0.254","Subnetmask":"255.255.0.0","DNSServer":"10.20.0.254","Mac":"60:01:94:E4:B4:21","Webserver":1,"WifiConfig":4,"WifiPower":17.0}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E4B421","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS7 = {"StatusTIM":{"UTC":"2020-04-18T10:59:12","Local":"2020-04-18T11:59:12","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:52","Sunset":"19:46"}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS10 = {"StatusSNS":{"Time":"2020-04-18T11:59:12"}}
11:59:12 RSL: stat/tasmota_E4B421/STATUS11 = {"StatusSTS":{"Time":"2020-04-18T11:59:12","Uptime":"0T00:03:46","UptimeSec":226,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"DenK","BSSId":"54:B8:0A:C8:F1:20","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:08"}}}
12:00:40 RSL: tele/tasmota_E4B421/STATE = {"Time":"2020-04-18T12:00:40","Uptime":"0T00:05:14","UptimeSec":314,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"DenK","BSSId":"54:B8:0A:C8:F1:20","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:08"}}
12:05:26 APP: Serial logging disabled

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
12:29:26 CMD: backlog rule 1; rule 2; rule3;
12:29:26 RSL: stat/tasmota_E4B421/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:29:26 RSL: stat/tasmota_E4B421/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:29:27 RSL: stat/tasmota_E4B421/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-18:
Do a `reset 6`. All settings will be lost except wifi and mqtt credentials.
After that do a `reset 3` initiates wifi calibration. Power off / on the RF Bridge.
Reconfigure. Web Frontend should work. If not restart your Router

dimitrybradt, 2020-04-18:
Hi, thanks for your reply. i tried. that doesn't help me.. 
any other suggestion?

closed 2020-04-18 by ascillato2