arendst/Tasmota #8217: Sonoff flashed with Tasmota minimal, no Wifi Manager

TroLoos, 2020-04-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hi, by accident I flashed my Sonoff with minimal version instead of lite, my bad...

But it gets worse... I think something went wrong with flashing because I see no tasmota-xxxxx WiFi, instead I have a network with SSID: 3. I can connect to it, and it is my Sonoff as it gives mi IP from range 192.168.4.xxx.

I know that minimal firmware is just a temporary state so I should flash it with full one, but I can't do it as the WiFi Manager is not starting.

I can see that it serves a flashing page under address:
http://SONOFFIP_WITH_WIFIMANAGER_UP_AND_MINIMAL/up

but it asks me for user/pass, which I cannot guess. Any hint what to do next? Serial connection or I can somehow recover?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R3
- [ ] Tasmota binary firmware version number used: 8.2.0 minimal
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Sonoff DIY Mode
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-18:
Do not flash minimal for initial flashing. It is written in the documentation.
You cant configure anything with minimal. Erase flash and flash Tasmota.bin.
If you cant access tx rx Gpio0 3.3V Gnd you have a brick

ascillato2, 2020-04-18:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-18 by ascillato2

TroLoos, 2020-04-18:
All right, it's actually my first contact with Tasmota on Sonoff and it's really hard, really really hard to believe that a working device, that creates access point, and one starting web server running, is a brick :-)

Please let me know if serial flashing is the only way, and if so - how to put it into programming mode since it boots ok this minimal firmware. Thanks!

meingraham, 2020-04-18:
Unfortunately, it's really really really pretty much the case based on thousands of user interactions and countless hours of attempts... by many involved directly in the development of Tasmota (such as Adrian).

Although there is one report of a workaround (non-confirmed) - https://groups.google.com/d/msg/sonoffusers/I5w60R-oNJU/NXAz_b7KAgAJ

TroLoos, 2020-04-18:
> Unfortunately, it's really really really pretty much the case based on thousands of user interactions and countless hours of attempts... by many involved directly in the development of Tasmota (such [...]

Ok, so I can confirm it then :-) WORKS like a charm!

Although, on MacOS the curl command was slightly different:
curl -F "u2=@/path/to/file.bin" http://192.168.4.1/u2

Thanks a lot, you saved me a lot of solder-related trouble ;-) Today's glass of wine is a cheers to you :-)

ebendl, 2020-05-31:
This unfortunatlye 

> Ok, so I can confirm it then :-) WORKS like a charm!
>
> Although, on MacOS the curl command was slightly different:
> curl -F "u2=@/path/to/file.bin" http://192.168.4.1/u2
>
> Thanks a lot, you saved me a [...]

This unfortunately didn't work for me. I had a POW running ESPHome, and wanted to revert back to Tasmota. I proceeded to enable the ESPHome web interface, then decided I'll upload minimal.bin first.

Ended up with a device in AP mode (esp_<xxxxx> SSID) and I could connect to it (192.168.4.1) but nothing was served on port 80, and the curl commands just returned that there was no reponse from the device on port 80. 

Ended up flashing via the serial pins again.

ascillato, 2020-05-31:
> Ended up flashing via the serial pins again.

Yes, because as explained above, Tasmota-minimal is NOT FOR FIRST FLASHING. Tasmota-minimal is ONLY FOR UPDATING TASMOTA on a device that already has Tasmota configured.

As explained in the docs, you need to use tasmota.bin or tasmota-lite.bin if you don't have enough free flash for OTA.

ebendl, 2020-05-31:
Yes, I know that now. This was advice I followed (foolishly) on the Home Assistant forums. 

I was only indicating that the workaround that TroLoos indicated working for them did not work for me. Not sure if TroLoos had went from Tasmota->minimal->Tasmota, or directly Minimal->Tasmota, but going ESPHome->Minimal->Tasmota is not possible using OTA.

TroLoos, 2020-05-31:
In my case I overwrote Sonoff original firmware (the Basic R3 version via DIY mode) with tasmota minimal. That’s how I broke it.

The procedure mentioned allowed my to flash it again with full tasmota firmware.

abmantis, 2020-06-09:
Also did the same thing by mistake. Now I need to solder a Sonoff Mini, since there's no AP :(

Scope666, 2020-06-20:
Just made the same mistake (still learning) ... flashed tasmota-minimal straight from tuya-convert not realizing that it needs the full bin first.  I guess it's solder time.  I tried the posted workaround but it's refusing connections at port 80.  (it will happily answer pings though)      ;)

abmantis, 2020-06-20:
I tried to do the workaround, but I wasn't even able to get a AP up :/

Morphy99, 2020-11-16:
> Unfortunately, it's really really really pretty much the case based on thousands of user interactions and countless hours of attempts... by many involved directly in the development of Tasmota (such [...]

I had this issue after upgrading an old device on 6.x to 8.x and using minimal. Not sure if I didnt follow the upgrade path (was too lazy to check) or the minimal fw didn't work from changing versions. I'm sure I did it like this before as it would complain if I did the full version. Oh well the tip in the link worked for me but I had to modify it slightly. He doesn't explain he's uploading the tasmota.bin file either!

This is what I used under linux:

`curl -F u2=@~/tasmota.bin http://192.168.4.1/u2 `

The tasmota bin file being in the home directory.

Fettkeewl, 2020-12-01:
*Bowing head in shame* I'm also one that did not know the minimal was lacking wifi manager. Used in in Tuya-Convert thinking it would be neat starting from minimal.. should've used lite I guess.

So about the connection refused part.,
I did as above tried from linux and win 10. On borth systems I got connection refused.
**Is there an workaround this "connection refused" ?**

I CAN open the device but it's an smart outdoor plug sealed with triangle screws and plastic hinges which might break upon
opening,.. so I'd rather not if I dont have to :P  

curl -F u2=@~/tasmota.bin http://192.168.4.1/u2       - Connection refused on Ubuntu Desktop
curl -F u2=@C:/tasmota.bin http://192.168.4.1/u2       - Connection refused on Windows 10

**For future fools such as myself, Can the minimal-version serve an simple web interface for OTA upload?**

Morphy99, 2020-12-01:
> _Bowing head in shame_ I'm also one that did not know the minimal was lacking wifi manager. Used in in Tuya-Convert thinking it would be neat starting from minimal.. should've used lite I guess.
> 
> So about the connection refused part.,
> I did as above tried from linux and win 10. On borth systems I got connection refused.
> **Is there an workaround this "connection refused" ?**
> 
> I CAN open the device but it's an smart outdoor plug sealed with triangle screws and plastic hinges which might break upon
> opening,.. so I'd rather not if I dont have to :P
> 
> curl -F u2=@~/tasmota.bin http://192.168.4.1/u2 - Connection refused on Ubuntu Desktop
> curl -F u2=@C:/tasmota.bin http://192.168.4.1/u2 - Connection refused on Windows 10
> 
> **For future fools such as myself, Can the minimal-version serve an simple web interface for OTA upload?**
> 
> Not that it matters but its one of theese
> [image]

Are you connected to the plug's hotspot? Can you ping 192.168.4.1?

Fettkeewl, 2020-12-01:
> Are you connected to the plug's hotspot? Can you ping 192.168.4.1?

Yes pings were working. 
Though all ports closed up to atleast 1000 before i gave up

Morphy99, 2020-12-01:
Sorry can't help then. Looks as though, from what I've read, you'll have to open her up...

edizius, 2021-12-08:
Sorry for reoprning this issue but I have exactly the same issue as @morphy99.
Had Tasmota 6.13, executed reset 2 + restart 1, flashed 8.1 minimal (as this is the smallest minimal I found), then 8.1. lite and then on my other devices tasmota-****AP mode started.

At this module now only ESP_**** is shown, I can 

 connect to it, but no web page is opened and by curl -F u2=@C:/tasmota.bin http://192.168.4.1/u2 - I get Connection refused on Windows 10 like @Fettkeewl also had.

For tasmota.bin on C I used 8.1. lite, is this the correct version? Is only flash by wire solving this issue, or is there something else I can try (module is inside the wall, so I would like to avoid to flash it by wire by every means)

barbudor, 2021-12-08:
You have bricked your device by not following the [Upgrade Path](https://tasmota.github.io/docs/Upgrading/#upgrade-flow) as documented.
From 6.13 you should have been through v6.7.1, v7.2.0, v8.5.1, v9.1.0 and latest.
The numbers on the doc are clickable links to download the right firmware to do the job, no need to search they are just at your hand.

Now, serial flash is your (only) way out.

edizius, 2021-12-09:
Ok, sad, thought this is related to config only (as it also worked for other modules in this way). Now I know... Thanks

edizius, 2021-12-11:
> You have bricked your device by not following the [Upgrade Path](https://tasmota.github.io/docs/Upgrading/#upgrade-flow) as documented. From 6.13 you should have been through v6.7.1, v7.2.0, v8.5.1, [...]

Regarding the upgrade path: no minimal version in between needed by OTA, as it is used directly, correct (already at 5.14?)?
No "reset <X>" nor power cycle in between necessary?

Only to be sure to brick not another device as well (some still on 5.13)

barbudor, 2021-12-11:
If you are using the binaries that are linked behind the number, they are all small enough to be upgraded without minimal.
A power cycle wouldn't harm as well as backuping settings at each step.

edizius, 2021-12-11:
On my selfcompiled version I have only 400kB free space available, and version 5.14 has 501kB.
If I try anyhow, will it brick if it fails? Is flashing 5.14 minimal version prior safe?

barbudor, 2021-12-11:
For sure on a recent version, you would an error message if there is not enough room
I can't tell for so old version.

edizius, 2021-12-12:
Ok, thx for the reply.

I checked now exact version and available space and it is
Program Version	5.13.1a
Program Size	518kB
Free Program Space	420kB
Free Memory	17kB

I checked minimal from 5.14 and it is 438kB, but minimal from 6.7.1 is only 377kB.

Thus, does anyone know if 5.13.1a can already be directly updated to 6.7.1 without risk, as it is "almost 5.14" or might this lead to a semi-brick again?

ascillato, 2021-12-12:
>Thus, does anyone know if 5.13.1a can already be directly updated to 6.7.1 without risk, as it is "almost 5.14" or might this lead to a semi-brick again?

To ensure that everything will work, you should follow the recommended path:
https://tasmota.github.io/docs/Upgrading/#upgrade-flow

barbudor, 2021-12-12:
When upgrading from such an old version, a lot of things could go wrong.
I would avoid the headache and go straight for serial flashing with full erase and reconfigure.
Shouldn't take more time that the 10 steps an normal upgrade path would take even if everything goes well. 
And if it goes south 😱 ...

edizius, 2021-12-13:
These devices are inside the wall, and in this case furthermore behind tiles. Thus, alternative to safe OTA upgrade is only staying on version 5.13.

I wished there is a safe OTA version because I would like to have the shutter control feature.

Jason2866, 2021-12-13:
Flash a test device with the 5.13 version and flash  https://github.com/Jason2866/ESP8266MinimalHTTPUpdater flash with this the 5.13 sonoff basic version and follow the upgrade path.

edizius, 2021-12-13:
OK, thanks, sounds good, so I think I give this a try.
Is this behaving differently (resp. more save) than using any other minimal like 8.1, which is already rather small? Because in this way I got the "briked module problem".

So the statement

> Yes, because as explained above, Tasmota-minimal is NOT FOR FIRST FLASHING. Tasmota-minimal is ONLY FOR UPDATING TASMOTA on a device that already has Tasmota configured.
>
> As explained in the docs, [...]

is not valid for the ESP8266minimalHTTPUpdater
?

Jason2866, 2021-12-13:
ESP8266minimalHTTPUpdater can be flashed when ever you want.
The minimal updater does *not* change anything in Tasmota settings area!
**So if you want to flash a later version of Tasmota you have to follow the upgrade path too!** Flashing ESP8266minimalHTTPUpdater gives you just the possibility to flash a different firmware.

edizius, 2021-12-13:
Awesome! Perfect, thank you. In this case I still do not get why there are (different) minimal versions, and why not just always your binary is used, without the risk of messing around some parameters. 
So, e.g. 5.14->ESP8266minimalHTTPUpdater->6.7.1
Instead of 5.14->6.7.1 minimal -> 6.7.1 and so on.