arendst/Tasmota #8219: PowerRetained not working

koos147, 2020-04-18:
### PROBLEM DESCRIPTION
Retained power state is not working anymore on new versions of Tasmota. it is working on older versions.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Variouse Tuya and Sonoff devices
- [X] Tasmota binary firmware version number used: 8.1.0.2
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Various device specific tools
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
Not a hardware related problem
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
/STATUS = {"Status":{"Module":0,"FriendlyName":["room"],"Topic":"device07","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":1,"PowerRetain":1}}
/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"room","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:01","StartupUTC":"2020-04-18T12:50:04","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":1737,"SaveAddress":"FB000"}}
/STATUS2 = {"StatusFWR":{"Version":"8.1.0.2(tasmota)","BuildDateTime":"2019-12-30T19:07:34","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"380/699"}}
/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["000082E9","2805C8000100060000005A00000000000000","00000000","00000000"]}}
/STATUS4 = {"StatusMEM":{"ProgramSize":452,"Free":548,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F400082","04048000","00000005","000000C0","00000080"],"Drivers":"1,2,3,4,9,10,12,16","Sensors":"1,2,3"}}
/STATUS5 = {"StatusNET":{"Hostname":"device07","IPAddress":"*.*.*.169","Gateway":"*.*.*.1","Subnetmask":"255.255.255.0","DNSServer":"*.*.*.22","Mac":"*********","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
/STATUS6 = {"StatusMQT":{"MqttHost":"*.*.*.60","MqttPort":*,"MqttClientMask":"device07","MqttClient":"device07","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
/STATUS7 = {"StatusTIM":{"UTC":"Sat Apr 18 13:09:05 2020","Local":"Sat Apr 18 14:09:05 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"05:52","Sunset":"19:46"}}
/STATUS10 = {"StatusSNS":{"Time":"2020-04-18T14:09:05"}}
/STATUS11 = {"StatusSTS":{"Time":"2020-04-18T14:09:05","Uptime":"0T00:19:01","UptimeSec":1141,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":10,"Color":"001A","White":10,"CT":500,"Channel":[0,10],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,*******,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
....../POWER = OFF (retained)

```

### TO REPRODUCE
Set option "PowerRetain 1" > "Restart 1" and change the starte. after this check the MQTT topic for the Retained flag

### EXPECTED BEHAVIOUR
that the Power topic (and preferable also the dimmer topic) has a retained flag on  MQTT

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
On older firmware "Sonoff-Tasmota 6.5.0" the function is working as expected

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-04-18:
Check SetOption90. Should be 0 in your case.

koos147, 2020-04-18:
Thanks. i checked it and it is off.

> /RESULT = {"SetOption90":"OFF"}

arendst, 2020-04-18:
That's strange. In my case it works as expected with SetOption90 set to 0:
```
15:27:56 CMD: powerretain
15:27:56 MQT: stat/wemos4/RESULT = {"PowerRetain":"OFF"}
15:28:01 CMD: powerretain 1
15:28:01 MQT: stat/wemos4/RESULT = {"PowerRetain":"ON"}
15:28:04 MQT: stat/wemos4/RESULT = {"POWER":"ON"}
15:28:06 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
15:33:10 CMD: setoption90
15:33:10 MQT: stat/wemos4/RESULT = {"SetOption90":"ON"}
15:33:15 CMD: setoption90 0
15:33:15 MQT: stat/wemos4/RESULT = {"SetOption90":"OFF"}
15:33:23 MQT: stat/wemos4/RESULT = {"POWER":"ON"}
15:33:23 MQT: stat/wemos4/POWER = ON (retained)
15:33:27 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
15:33:27 MQT: stat/wemos4/POWER = OFF (retained)
15:33:37 CMD: powerretain 0
15:33:37 MQT: stat/wemos4/POWER =  (retained)
15:33:37 MQT: stat/wemos4/RESULT = {"PowerRetain":"OFF"}
15:33:39 MQT: stat/wemos4/RESULT = {"POWER":"ON"}
15:33:39 MQT: stat/wemos4/POWER = ON
15:33:48 MQT: stat/wemos4/RESULT = {"POWER":"OFF"}
15:33:48 MQT: stat/wemos4/POWER = OFF
```
If you compile yourselves did you by accident enable define MQTT_NO_RETAIN
```
//#define MQTT_NO_RETAIN                         // Disable all retain flags (including LWT!) if unsupported by broker (eg Losant)
```
if so, it disables ALL retained messages.

ascillato2, 2020-04-20:
Hi, any news on this?

koos147, 2020-04-20:
Sorry for the late reply.
I used the bin files. so nothing changed. (always the latest at the time of installation)

My console in Tasmota shows retained. but on MQTT the retained flag is missing.
on the old Tasmota (or self build firmwares) the reained flag is working so the broker and client app are working fine.

arendst, 2020-04-20:
Cannot reproduce.

- I set ``PowerRetain 1`` and switch the power to send a retained message to the broker.
- I logon with a MQTT client and immediatly receive the retained power message

So the functionality is still there.

ascillato, 2020-04-20:
Sorry, I can't reproduce it either. Retained messages are working in my devices. Seems that there is something wrong in your device. Using latest Tasmota from the precompiled bins, try to wipe out all settings of your Tasmota with reset 5 command and config it again.

ascillato2, 2020-04-21:
Hi,

Any news on this?

ascillato2, 2020-04-22:
Please, when you have time, tell us if a config reset solves your issue. Thanks

closed 2020-04-22 by ascillato2

rafuz, 2020-04-25:
I am experiencing the same behaviour on all my tasmota devices, but only when I remove power from the device.
If I simply restart the device the power state is retained.

I use SetOption0 Off to reduce writes to flash and PowerRetain On to retain power in MQTT.
Anyway when I switch power I don't see a retained message in the broker.
The additional info I can give is that all my devices where flashed with the last Tasmota binary (8.2) and they are configured with SetOption19 On and inverted topic/prefix
SetOption90 is Off
The broker is the Mosquitto Plugin for HomeAssistant.
This happens with all my Tasmotas (POW, Basic, various Tuya based smart plugs)