arendst/Tasmota #8220: PWM Dimmer - Dimmer state not reported

raufis27, 2020-04-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Upgraded from version 8.1 to latest 8.2.0 with support of PWM Dimmer from MJ. 

Rule on Dimmer#State do publish doesn't work anymore. Also if I publish to the ...cmnd/dimmer/Dimmer and check in Tasmota console I see that Dimmer state changed but it doesn't affect dimmer slider or light. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): __MJ Dimmer___
- [X ] Tasmota binary firmware version number used: _8.2_
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:29:05 MQT: /myhome/smart_light/stat/bedroom_dimmer/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:29:05 MQT: /myhome/smart_light/stat/bedroom_dimmer/RESULT = {"Module":{"73":"PWM Dimmer"}}
00:29:05 MQT: /myhome/smart_light/stat/bedroom_dimmer/RESULT = {"GPIO0":{"19":"Button3"},"GPIO1":{"18":"Button2"},"GPIO2":{"0":"None"},"GPIO3":{"59":"Led4i"},"GPIO4":{"158":"LedLinki"},"GPIO5":{"58":"Led3i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"57":"Led2i"},"GPIO13":{"37":"PWM1"},"GPIO14":{"56":"Led1i"},"GPIO15":{"122":"Button1i"},"GPIO16":{"29":"Relay1i"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
00:29:48 CMD: Backlog Rule1; Rule2
00:29:48 MQT: /myhome/smart_light/stat/bedroom_dimmer/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":361,"Rules":"on switch3#state=2 do dimmer + endon on switch2#state=2 do dimmer - endon on switch2#state=3 do dimmer 20 endon on switch3#state=3 do dimmer 100 endon"}
00:29:49 MQT: /myhome/smart_light/stat/bedroom_dimmer/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":424,"Rules":"on Dimmer#State do publish /myhome/smart_light/stat/bedroom_dimmer/Dimmer %value% endon"}
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-18:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

raufis27, 2020-04-18:
I updated template, if you think that this info is not enough, please close this issue. I downgraded to previous release. Thank you.

closed 2020-04-18 by raufis27