arendst/Tasmota #8222: Problem with Shutter and Buttons

andur82, 2020-04-18:
### PROBLEM DESCRIPTION
When using the "shutterbutton1 1 toggle" command the response is always 0
After setting up a button input, after reboot the configuration is cleared.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Generic (Wemos D1 mini)
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
21:08:17 RSL: stat/tasmota_18E2A2/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
21:08:17 RSL: stat/tasmota_18E2A2/RESULT = {"Module":{"18":"Generic"}}
21:08:17 RSL: stat/tasmota_18E2A2/RESULT = {"GPIO0":{"0":"Nessuno"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"0":"Nessuno"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"0":"Nessuno"},"GPIO5":{"17":"Button1"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"37":"PWM1"},"GPIO13":{"42":"Counter1"},"GPIO14":{"22":"Relay2"},"GPIO15":{"0":"Nessuno"},"GPIO16":{"29":"Relay1i"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
21:11:30 RSL: stat/tasmota_18E2A2/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota_18E2A2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:56","StartupUTC":"2020-04-18T20:07:34","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-04-18T20:50:12","SaveCount":28,"SaveAddress":"F8000"}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:59:30","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"317/699"}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AnDuR Home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS4 = {"StatusMEM":{"ProgramSize":578,"Free":424,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":0,"Features":["00000410","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS5 = {"StatusNET":{"Hostname":"tasmota_18E2A2-0674","IPAddress":"192.168.69.222","Gateway":"192.168.69.250","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"5C:CF:7F:18:E2:A2","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_18E2A2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS7 = {"StatusTIM":{"UTC":"2020-04-18T20:11:30","Local":"2020-04-18T21:11:30","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:52","Sunset":"19:46"}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS10 = {"StatusSNS":{"Time":"2020-04-18T21:11:30","COUNTER":{"C1":12069},"Shutter1":{"Position":0,"Direction":0,"Target":0}}}
21:11:30 RSL: stat/tasmota_18E2A2/STATUS11 = {"StatusSTS":{"Time":"2020-04-18T21:11:30","Uptime":"0T00:03:56","UptimeSec":236,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"AnDuR Home","BSSId":"18:0F:76:40:81:68","Channel":10,"RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:14"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
21:09:22 CMD: shutterbutton1 1 toggle
21:09:22 RSL: stat/tasmota_18E2A2/RESULT = {"ShutterButton1":"1 0 - - - - - - - -"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
CMD: shutterbutton1 1 toggle

### EXPECTED BEHAVIOUR
I want to have the result "ShutterButton1":"1 t - - - - - - -" such as documentation

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-04-19:
I'm sorry but if you want it work according to documentation you should use the syntax from the said documentation 

`ShutterButton<x> <button> <func> <mqtt>`

closed 2020-04-19 by blakadder

andur82, 2020-04-19:
I've used both method, the one on the documentation example
ShutterButton1 1 toggle 
_Single press will toggle shutter, double press will move it to 50%. No hold action and no other shutter control by MQTT, <mqtt> is don't care here._ 
but the result is the same.

> ShutterButton<x> <button> <p1> <p2> <p3> <ph> <m1> <m2> <m3> <mh> <mi>

<button> 1..4: Button number, 0/-: disable buttons for this shutter
<p1> 0..100: single press position, t: toggle, -: disable
<p2> 0..100: double press position, t: toggle, -: disable
<p3> 0..100: tripple press position, t: toggle, -: disable
<ph> 0..100: hold press position, t: toggle, -: disable
<m1> 1: enable single press position MQTT broadcast, 0/-: disable
<m2> 1: enable double press position MQTT broadcast, 0/-: disable
<m3> 1: enable tripple press position MQTT broadcast, 0/-: disable
<mh> 1: enable hold press position MQTT broadcast, 0/-: disable
<mi> 1: enable MQTT broadcast to all shutter indices, 0/-: disable

in the parameter you wrote "t" as parameter for toggle but it doesn't work, it always take the "0" as parameter.

Take from [https://tasmota.github.io/docs/Blinds-and-Shutters/#button-control](url)