arendst/Tasmota #8226: [Zigbee] [Z2T] Sending MQTT commands to Zigbee devices

ST-Automation, 2020-04-19:
**Have you looked for this feature in other issues and in the docs?**  
I read the whole Zigbee documentation page and checked the issues on github, I dind't find what I was looking for. (https://tasmota.github.io/docs/Zigbee/).
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
On the Zigbee documentation page I would expect an explanation like: How to send MQTT-commands towards your Zigbee devices. 
The only explanation I find is how to send commands via the tasmota console using ZbSend (before ZigbeeSend on the old documentation). 
I'm wondering why there is no topic like e.g.: cmnd/z2t/ZbSend to send a strings like: { "device":"0x4773", "send":{"Power":"On"} } and that Z2T translates it to: ZbSend { "device":"0x4773", "send":{"Power":"On"} } as if you were typing the command in the tasmota console. Then I can read and write towards all my zigbee devices from NodeRed.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
If this function is available, then please explain it on the documentation page: https://tasmota.github.io/docs/Zigbee/#examples

If not, I would love to have this new simple feature.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
With the new documentation there is now an explanation on "Zigbee Binding", this can work but look to complicated and time consuming for every small change I would like to make in my setup. Via NodeRed and MQTT it is so much more convenient.
**Additional context**  
_Add any other context or screenshots about the feature request here._
/
**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-04-19:
I don't get your request.

Sending a MQTT message to the topic `cmnd/<your_device_topic>/ZbSend` with payload ` { "device":"0x4773", "send":{"Power":"On"} }` is precisely the way it works currently. Feel free to propose examples or clarifications in the documentation.

As for Zigbee Bindings, feel free to propose enhanced documentation.

meingraham, 2020-04-19:
Every Tasmota command, including zbSend, can be sent via MQTT as @s-hadinger has shown. This is explained in the Tasmota commands article. Please familiarize yourself with the Tasmota operational basics by reading that article.

closed 2020-04-19 by meingraham

ST-Automation, 2020-04-19:
I was already thinking it would be this simple, but I could find the right topic for MQTT, thank you for helping out, I got it working now. It would have helped me if this MQTT topic was given as an example in the documentation.

How can I contribute with proposing enhanced documentation? I don't think this is the way, making issues for every documentation proposal. I'm still finding my way a bit on github so...

Jason2866, 2020-04-19:
@ST-Automation  explained in docu https://tasmota.github.io/docs/Contributing/

spanout, 2020-10-06:
> I don't get your request.
>
> Sending a MQTT message to the topic `cmnd/<your_device_topic>/ZbSend` with payload ` { "device":"0x4773", "send":{"Power":"On"} }` is precisely the way it works currently. [...]

I struggled with this also...

The actual command is <your_device_topic>/cmnd/ZbSend

ST-Automation, 2020-10-06:
You should put cmnd in front, like this: cmnd/zigbeeGateway/ZbSend (ZigbeeSend instead of ZbSend for older versions of Tasmota)

In MQTT-settings in your Tasmota webinterface you define"Topic = %topic%" as in my example: **zigbeeGateway**
And "Full Topic (%prefix%/%topic%/)" should be: **%prefix%/%topic%/**

And then you just send **{ "device":"0x4773", "send":{"Power":"On"} }** as payload via MQTT, and then the light should turn on e.g.

Hopefully this helps

spanout, 2020-10-06:
Thank you... This does now make sense

sanchosk, 2021-01-17:
One suggestion - with SetOption112 the tasmota device correctly publishes the status and responses of the devices.
So now I have:
`tele/button1/SENSOR {"button1":{"Device":"0x9DC0","Name":"button1","0006!01":"","Power1":1,"Endpoint":1,"LinkQuality":37}}`
All is OK.
But I would expect the zigbee bridge to also listen to commands with similar style now. So instead of 
`cmnd/zigbeeTas/ZbSend {"Device":"lamp1","Send":{"power":2}}`
I'd like to send something like
`cmnd/zigbeeTas/lamp1 {"power" : 2}`
Is there some option to set it like that? It would simplify my nodeRed automation...

s-hadinger, 2021-01-17:
I get your point but it's not possible currently and would have some deep implications on Tasmota MQTT parsing. There is no plan to implement it.

sanchosk, 2021-01-17:
My other option would be to implement module in nodeRed where there is much more RAM and storage, I guess :)

avk999, 2021-02-16:
> One suggestion - with SetOption112 the tasmota device correctly publishes the status and responses of the devices.
> So now I have:
> `tele/button1/SENSOR [...]

That can be fixed by a simple shell (or python or whatever) script that reads the commands in the format you prefer and re-posts them converted? I have a shell script with mosquitto_sub-pub doing it for quite some time.