arendst/Tasmota #8227: Compilation of tasmota - Add info about TasmotaCompiler availability

bovirus, 2020-04-19:
@arendst 

**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

I propose you to add in home page doc about compilation section the info about TasmotaCompiler.

TasmotaCompiler is very interesting project that offer a way to self compile the tasmota firmware using a web interface without specific knowledge of compilation.

You can select 

github tasmota branch (and you can also definie a personal branch)
functions to enable in the firmware
language
pre-set Wifi settings

Info: https://github.com/benzino77/tasmocompiler

It's interesting way to self compile a bstable/development firmware not including not necessary functions.

Thanks.

Jason2866, 2020-04-19:
It is there https://tasmota.github.io/docs/Compile-your-build/

bovirus, 2020-04-19:
@Jason2866 

I believe that could be at least added info in home page / Compile section an info text that suggest to check doc for additional way to compile.

meingraham, 2020-04-19:
The docs step a user on how to set up Tasmota on a device starting from the ground up. There are a few references therein on cases where a user can/will compile the firmware. These references link to the article @Jason2866 provided. That article lists TasmoCompiler as an option. 

Besides arranging the information in a logical progression, there is not much more that can be done based on the countless number of topics that could be deemed important to users. This is the reason there is a search function - search for compiling, read that article, discover TasmoCompiler, follow that link, read about using it. Users should expect to have to read it they desire to use Tasmota properly.

closed 2020-04-19 by meingraham

bovirus, 2020-04-19:
@meingraham 

Sincerly speaking I didn't understand what is the negative point to add a simple info in Compile section that say

"if you want to compile please check doc info (link) to get more info about different way to compile source code.

Small, simple and direct.
And for me more useful than "Use search function".

Thanks.

Jason2866, 2020-04-19:
@bovirus  what you suggest is already there too.
https://tasmota.github.io/docs/Getting-Started/#compiling-tools-optional

[image]

meingraham, 2020-04-19:
@Jason2866 

Came here to post the very thing you did :wink:

So, @bovirus - "Getting Started" seems like where a user would go... to get started.

Then, the user sees (on the right) the progression of tasks on that journey... and what should the user see... "Compiling Tools". Follow the link and voila!

"Small, simple and direct."

TTFN