arendst/Tasmota #8228: LC-Tech 4 CH relay

TasmoFlash, 2020-04-19:
<Hi Guys
if one of you could integrate a device in Tasmota, try it now for one
week without success.
LC-Tech 4 CH relay
{"LCTech 4-Relay", // LC Technology Relay board 4 channel, based on ESP 01
      GPIO_USER, // GPIO00
      GPIO_RXD, // GPIO01 TX Serial RXD
      GPIO_USER, // GPIO02
      GPIO_TXD, // GPIO03 RX Serial TXD
      0,
      0,
      0, 0, 0, 0, 0, 0, // Flash connection
      0, 0, 0, 0, 0, 0
   }

thank you in advance

closed 2020-04-19 by close-issue-app[bot]

taste66, 2020-04-19:
https://tasmota.github.io/docs/devices/LC-Technology-WiFi-Relay/

TasmoFlash, 2020-04-20:
Danke hat geklappt super Service!!

https://tasmota.github.io/docs/devices/LC-Technology-WiFi-Relay/