arendst/Tasmota #8232: Trigger on SerialReceived/SSerialReceived parsing for reading in sensor value

Lightning1984, 2020-04-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes, for hours!!

**Is your feature request related to a problem? Please describe.**  
The basis for this function seems to be there since https://github.com/arendst/Tasmota/issues/4942 was implemented.

**Describe the solution you'd like**  
I have tried to configure a rule using  https://github.com/arendst/Tasmota/issues/4942 which works to trigger a static command, i.e. switch something on or off by a serial String. I would aditionally like to receive values of a sensor using this method, for example

on SerialReceived#Data=pressure:1234 do var2 %value% <-- var2 should be set to 1234

I do have full control over the other device connected via TTL serial so can let it respond in any desired way.

**Describe alternatives you've considered**  
I have looked into using the scripting function instead of rules, but couldn't even find any information on how to process serial data. I have also thought of using some of the metering interfaces but that does not seem to be a valid path either.

**Additional context**  
I'm trying to bring a compressor with a self made controller system into my KNX BUS and like to send the pressure value to the BUS additionally to being able to switch it on and off. Obviousely I could try to use MQTT to send the full serial Text to some other system where I could parse the Text, and then again generate a KNX telegram with the value but I'd like to keep this as simple as possible and would like to avoid any further components involved 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-19:
Please, post the output of the console when you are receiving a command from the serial. Thanks.

Jason2866, 2020-04-19:
Have you enabled serial bridge function before (via sending a dummy string) ?

meingraham, 2020-04-19:
https://tasmota.github.io/docs/Commands/#serial-bridge

> Information received by Tasmota over the serial bridge is captured automatically. Before data will be received, a properly formatted SerialSend<x> or SSerialSend<x> command must be executed. This must be done any time the device restarts (e.g., via a System#Boot triggered rule).

Lightning1984, 2020-04-19:
Hi,

thats all not really the problem, below you can see some sample output:

22:00:29 CMD: rule1 1
22:00:29 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":330,"Rules":"on SerialReceived#Data=on do backlog var1 %value%; power1 1 endon on SerialReceived#Data=off do backlog power1 0; var3 %value% endon on SerialReceived#Data=aaa do var2 %value% endon"}
22:00:32 RSL: RESULT = {"SerialReceived":on}
22:00:32 RUL: SERIALRECEIVED#DATA=ON performs "backlog var1 ON; power1 1"
22:00:32 RSL: RESULT = {"Var1":"ON"}
22:00:32 KNX: Output 1 = 1 sent to 0.6.3
22:00:32 RSL: RESULT = {"POWER1":"ON"}
22:00:32 RSL: POWER1 = ON
22:00:52 RSL: RESULT = {"SerialReceived":off}
22:00:52 RUL: SERIALRECEIVED#DATA=OFF performs "backlog power1 0; var3 OFF"
22:00:52 KNX: Output 1 = 0 sent to 0.6.3
22:00:52 RSL: RESULT = {"POWER1":"OFF"}
22:00:52 RSL: POWER1 = OFF
22:00:52 RSL: RESULT = {"Var3":"OFF"}
22:01:01 CMD: vars
22:01:01 RSL: RESULT = {"Command":"Unknown"}
22:01:03 CMD: var
22:01:03 RSL: RESULT = {"Var1":"ON","Var2":"AAA","Var3":"OFF","Var4":"","Var5":"","Var6":"","Var7":"","Var8":"","Var9":"","Var10":"","Var11":"","Var12":"","Var13":"","Var14":"","Var15":"","Var16":""}

As you can see above, triggering an action upon a known string received from Serial is working just fine and i can save that string in one of the variables but:

Let's say my compressor controller would be sending a serial string like this:
pressure:720
How would the rule need to look like? SerialReceived#Data=pressure:* or in regex SerialReceived#Data=pressure\:[0-9]{3}
I'd like to only save the variable part (720 in this case) as VAR<x> to then send it to the knx bus

ascillato, 2020-04-20:
Hi,

So far, parsing inside the DATA of the serialreceived is only supported if the data received is a JSON.

* Simplier case:

If you are going to send just one type of value, you can modify your device to send by serial **_just the number_** without the `pressure:` text at the begining.
Like this:
`22:00:32 RSL: RESULT = {"SerialReceived":10} `
and not this:
`22:00:32 RSL: RESULT = {"SerialReceived":pressure:10}`

So, in this simplier case, you can use the following rule:
```lua
rule1 on SerialReceived#Data do KNXTX_VAL1 %value% endon
```
Where the KNX address to send that value is the one defined in the KNX Menu of the Tasmota Web UI at the SLOT 1 **KNX_TX1** ( or any slot set in KNX Menu at KNX_TX )

* More complex case (received JSON STRING)

If you are going to send several types of values, you can modify your device to send by serial the data with JSON Encoding.
So from your pressure example, your device should send:

`{"pressure":"10"}`

and you should see in the console:

`22:00:32 RSL: RESULT = {"SerialReceived":{"pressure":"10"}} `

So, the rule needed should be:

```lua
rule1 on SerialReceived#pressure do KNXTX_VAL1 %value% endon
```
Where the KNX address to send that value is the one defined in the KNX Menu of the Tasmota Web UI at the SLOT 1 **KNX_TX1** ( or any slot set in KNX Menu at KNX_TX )

ascillato2, 2020-04-21:
Hi,

Any news on this?

Lightning1984, 2020-04-21:
Hi,

Thanks for the quick response! Your solution works like a charm. Could I just not find it or is this functionality not documented yet?

One thing I noticed though is that when storing the value in a variable and then using the same variable to send to knx in a backlog chained command, the value the variable had before the update is sent to the KNX bus:

`Rule1 "`
`Rule1 on SerialReceived#pressure do backlog var1 %value%; knxtx_val1 %var1% endon on SerialReceived#temp do backlog var2 %value%; knxtx_val2 %var2% endon on SerialReceived#status do backlog var3 %value%; knxtx_val3 %var3% endon on SerialReceived#error do backlog var4 %value%; knxtx_val4 %var3% endon`

22:46:43 RUL: SERIALRECEIVED#PRESSURE performs "backlog var1 10; knxtx_val1 10"
22:46:43 RSL: RESULT = {"Var1":"10"}
22:46:43 KNX: KNX TX 1 = 10.00 sent to 0.6.4
22:46:43 RSL: RESULT = {"KnxTx_Val1":"10.00"}
22:46:58 RSL: RESULT = {"SerialReceived":{"pressure":"20"}}
22:46:58 RUL: SERIALRECEIVED#PRESSURE performs "backlog var1 20; knxtx_val1 10"
22:46:58 RSL: RESULT = {"Var1":"20"}
22:46:59 KNX: KNX TX 1 = 10.00 sent to 0.6.4
22:46:59 RSL: RESULT = {"KnxTx_Val1":"10.00"}
22:47:10 RSL: RESULT = {"SerialReceived":{"pressure":"20"}}
22:47:10 RUL: SERIALRECEIVED#PRESSURE performs "backlog var1 20; knxtx_val1 20"
22:47:10 RSL: RESULT = {"Var1":"20"}
22:47:10 KNX: KNX TX 1 = 20.00 sent to 0.6.4
22:47:10 RSL: RESULT = {"KnxTx_Val1":"20.00"}
`

When doing it this way it just works fine:

`Rule1 "`
`Rule1 on SerialReceived#pressure do backlog var1 %value%; knxtx_val1 %value% endon on SerialReceived#temp do backlog var2 %value%; knxtx_val2 %value% endon on SerialReceived#status do backlog var3 %value%; knxtx_val3 %value% endon on SerialReceived#error do backlog var4 %value%; knxtx_val4 %value% endon`

best regards

closed 2020-04-21 by Lightning1984

ascillato, 2020-04-21:
> Could I just not find it or is this functionality not documented yet?
I remembered this usage from an old issue, so seems that we forgot to add that example to the docs. Could you add it please?

> One thing I noticed though is that when storing the value in a variable and then using the same variable to send to knx in a backlog chained command, the value the variable had before the update is sent

Yes, that is because the text %var1% is replaced with the value of VAR1 but before the backlog command is executed. The actual string passed to backlog command is:
if VAR is 5 and the value is 3, what backlog receives to then execute is
`backlog var1 3; knxtx_val1 5 endon`
So, your second approach is the correct one.

lalo-uy, 2020-12-07:
on SerialReceived#Data do  var1 %value% endon
do not trigger on   {"SerialReceived":{"A":"28-B7-52-79-A2-1-3-2E","T":"23.19"}}

I can capture A or T fields with on SerialReceived# or on SerialReceived#T 
   but not both or the whole string.
I want to perform a " publish tele/HUB/%var1%  %var2% "

lalo-uy, 2020-12-07:
I got it working with different rules

Rule1 ON system#boot DO Backlog baudrate 9600; seriallog 0; serialsend2 a endon
Rule2 ON SerialReceived#A do var1 %value% endon 
Rule3 ON SerialReceived#T do publish tele/HUB/%var1%  %value% endon

makedir, 2024-01-27:
@lalo-uy Hello. I also want to do something really simple with a ESP32 Tasmota device, listening on serial for something like var%value, and then onReceive send var=value as json to a MQTT broker. Can you explain a little bit on your last example? What is #A and #T ?