arendst/Tasmota #8233: mdns fails after wifi lost

kugelkopf123, 2020-04-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After the Wifi connection is lost, e.g. by restarting the router/access point, MDNS no longer works correctly. In all devices that were connected to this AP, I get an error message. 
```
01:31:13 DNS: fehlgeschlagen
01:31:14 DNS: fehlgeschlagen
01:31:15 DNS: fehlgeschlagen
01:31:16 DNS: fehlgeschlagen
01:31:17 WIF: Prüfe Verbindung...
01:31:17 DNS: fehlgeschlagen
01:31:18 DNS: fehlgeschlagen
01:31:19 DNS: fehlgeschlagen
01:31:20 DNS: fehlgeschlagen
01:31:21 DNS: fehlgeschlagen
01:31:22 DNS: fehlgeschlagen
01:31:23 DNS: fehlgeschlagen
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly2.5, SonoffBasic,SonoffDualR2,Shelly1
- [x] Tasmota binary firmware version number used: 8.2.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: MS VS-PlatformIO
- [x] Flashing tools used: PlatformIO > ESPTOOL.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
01:40:32 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
01:40:32 SRC: Backlog
01:40:32 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
01:40:32 MQT: stat/Shelly25_1/RESULT = {"NAME":"Shelly 2.5 test","GPIO":[56,0,19,0,21,127,0,0,0,126,0,22,0],"FLAG":2,"BASE":18}
01:40:32 DNS: fehlgeschlagen
01:40:32 SRC: Backlog
01:40:32 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
01:40:32 MQT: stat/Shelly25_1/RESULT = {"Module":{"0":"Shelly 2.5 test"}}
01:40:33 SRC: Backlog
01:40:33 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
01:40:33 MQT: stat/Shelly25_1/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"19":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"127":"Button2in"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"126":"Button1in"},"GPIO14":{"0":"None"},"GPIO15":{"22":"Relay2"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
01:42:24 MQT: stat/Shelly25_1/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":259,"Rules":"on energy#current[1]>0.600 do backlog var1=%value%;power1 0;power2 0; publish stat/%topic%/RESULT {"BLACKOUT1":"%var1%"} endon on energy#current[2]>0.600 do backlog var2=%value%;power1 0;power2 0;publish stat/%topic%/RESULT {"BLACKOUT2":"%var2%"} endon"}
01:42:24 SRC: Backlog
01:42:24 CMD: Gruppe 0, Index 2, Befehl "RULE", Daten ""
01:42:24 MQT: stat/Shelly25_1/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":370,"Rules":"on SHUTTER1#Position do backlog publish stat/%topic%/RESULT {"ShutterPosition1":%value%} endon"}
01:42:24 DNS: fehlgeschlagen
01:42:24 SRC: Backlog
01:42:24 CMD: Gruppe 0, Index 3, Befehl "RULE", Daten ""
01:42:24 MQT: stat/Shelly25_1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
01:43:23 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
01:43:23 MQT: stat/Shelly25_1/STATUS = {"Status":{"Module":0,"FriendlyName":["Shelly25_1","Shelly25_12"],"Topic":"Shelly25_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:43:23 MQT: stat/Shelly25_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://raspi4/tasmotaUpdates/shelly25.bin.gz","RestartReason":"Software/System restart","Uptime":"8T10:50:54","StartupUTC":"2020-04-11T12:52:29","Sleep":50,"CfgHolder":4617,"BootCount":1,"BCResetTime":"2020-04-08T00:37:06","SaveCount":557,"SaveAddress":"1F6000"}}
01:43:23 MQT: stat/Shelly25_1/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(tasmota)","BuildDateTime":"2020.04.08 00:19:07","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"418/699"}}
01:43:23 MQT: stat/Shelly25_1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Eiswuerstchen",""],"TelePeriod":300,"Resolution":"569B90C0","SetOption":["0020C80B","1405C80001000600003C4800000000000000","408000A8","00000000"]}}
01:43:23 MQT: stat/Shelly25_1/STATUS4 = {"StatusMEM":{"ProgramSize":481,"Free":1544,"Heap":21,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000407","8F000196","040003A1","00000004","010000C0","00000100","00000000"],"Drivers":"1,2,3,4,9,10,27","Sensors":"2,3"}}
01:43:23 MQT: stat/Shelly25_1/STATUS5 = {"StatusNET":{"Hostname":"Shelly25-1","IPAddress":"192.168.1.202","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:68:EC:63","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:43:23 MQT: stat/Shelly25_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.73","MqttPort":1883,"MqttClientMask":"Shelly25_1","MqttClient":"Shelly25_1","MqttUser":"tasmota","MqttCount":9,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:43:23 MQT: stat/Shelly25_1/STATUS7 = {"StatusTIM":{"UTC":"2020-04-19T23:43:23","Local":"2020-04-20T01:43:23","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:22","Sunset":"20:38"}}
01:43:23 MQT: stat/Shelly25_1/STATUS10 = {"StatusSNS":{"Time":"2020-04-20T01:43:23","ANALOG":{"Temperature":31.2},"Shutter1":{"Position":4,"Direction":0,"Target":4},"TempUnit":"C"}}
01:43:23 MQT: stat/Shelly25_1/STATUS11 = {"StatusSTS":{"Time":"2020-04-20T01:43:23","Uptime":"8T10:50:54","UptimeSec":730254,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":9,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Eiswuerstchen","BSSId":"E0:28:6D:22:3F:BC","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":4,"Downtime":"0T00:00:22"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
01:42:56 DNS: fehlgeschlagen
01:42:57 WIF: Prüfe Verbindung...
01:42:57 DNS: fehlgeschlagen
01:42:58 DNS: fehlgeschlagen
01:42:59 DNS: fehlgeschlagen
01:43:00 DNS: fehlgeschlagen
01:43:01 DNS: fehlgeschlagen
01:43:02 DNS: fehlgeschlagen
01:43:03 DNS: fehlgeschlagen
01:43:04 DNS: fehlgeschlagen

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile Tasmota with 
```
#define  USE_DISCOVERY      
#define WEBSERVER_ADVERTISE
#undef MQTT_HOST_DISCOVERY  
```
And switch the WIFI from the AP off for a Minute and then on again. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Just the normal MDNS behaviour. Perhaps an auto-restart of the MDNS Service.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
-

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
-

**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-04-19:
What happens if yo turn off `SetOption55` and then turn it on again?

Jason2866, 2020-04-20:
mDNS is a Mess by design.... mDNS generates more problems than it solves.
Why not using a real DNS server? Let the RPi do, or your Router.

The issue can probably not been solved in Tasmota since Tasmota uses the API from Arduino Core
for this function.

kugelkopf123, 2020-04-20:
> What happens if yo turn off `SetOption55` and then turn it on again?

If i do so, it reboots and everything is fine again. 

@Jason2866 
Im just reporting this issue its not essential for my life. Some kind of hey... There is a problem, perhaps this is nice to know.

effelle, 2020-04-20:
The problem is as for now there is no other option than restart the device.

kugelkopf123, 2020-04-20:
> The problem is as for now there is no other option than restart the device.

Thats okay for me. As is already said. Just wanted to report this behaviour.

closed 2020-04-20 by kugelkopf123

effelle, 2020-04-20:
And I thank you for that.

Jason2866, 2020-04-20:
@kugelkopf123  Thx, for reporting.
We have already discussed to drop completly support for mDNS
Since for most which are using it, it works Theo decided to keep,
knowing that  mDNS is generating strange issues.