arendst/Tasmota #8234: [Feature Request] Adding estimated costs the WebUI for PowerPlugs

chimcen, 2020-04-20:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Missing the possibility to see the estimated costs for the used electrical current.

**Describe the solution you'd like**  
dding estimated costs the WebUI for 
Is there a way to add in WebUI an overview for all costs incurred so far?

**Describe alternatives you've considered**  
...

**Additional context**  
Like this:
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-04-20:
Hi, 

Already available. 

Please see TARIFF command. 

Thanks

closed 2020-04-20 by ascillato2

chimcen, 2020-04-20:
Didn’t see this before - sry.

But can you help me by giving me an example how to configure this to see the costs like shown on my picture?