arendst/Tasmota #8236: SSerialReceived in HEX-Format doesn't form valid JSON response

mabsenger, 2020-04-20:
### PROBLEM DESCRIPTION
SSerialReceived JSON seems not be correctly formated. The Quotes in the Value are missing

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): NodeMCU
- [x ] Tasmota binary firmware version number used: 8.2.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: Tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
15:20:17 MQT: mytopic/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
15:20:17 MQT: mytopic/stat/RESULT = {"Module":{"0":"Generic"}}
15:20:17 MQT: mytopic/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"72":"SerBr Rx"},"GPIO13":{"0":"None"},"GPIO14":{"71":"SerBr Tx"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
15:21:27 MQT: mytopic/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":470,"Rules":"ON System#Boot DO SSerialSend5 test ENDON"}
15:21:27 MQT: mytopic/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
15:21:28 MQT: mytopic/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
15:23:27 MQT: mytopic/tele/RESULT = {"SSerialReceived":7E01020304057E}

```

### TO REPRODUCE
setup device as stated above, eg. configure module and serial pins, setup rule to execute SSerialSend5 data to get responses as HEX string and receive data via SerialBridge

### EXPECTED BEHAVIOUR
form a valid json in the result with quotes around received HEX-String:
15:23:27 MQT: mytopic/tele/RESULT = {"SSerialReceived":"7E01020304057E"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-20:
Duplicate see #8040 
Use latest development version

ascillato2, 2020-04-20:
Closing this issue as it is already reported and solved in the development branch. Sorry. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-20 by ascillato2