arendst/Tasmota #8237: Hiking DDS238-2 reporting all zeros

capputeam, 2020-04-20:
### PROBLEM DESCRIPTION
I have an Hiking DDS238-2 with Tasmota 8.2.0 (sensors) running. Even if the hiking display reports the correct values (Current, Voltage, Power, ....), the tasmota webgui (and MQTT) reports all zeros. Trying to setoption71 and setoption72 in all possible combination dont change the situation.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): HIKING DDS238-2_
- [x ] Tasmota binary firmware version number used: 8.2.0 (sensors)_____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
16:21:39 CMD: template
16:21:39 MQT: stat/Contatore01/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,0,0,255,255,255,255,255],"FLAG":15,"BASE":18}
16:21:42 CMD: module
16:21:42 MQT: stat/Contatore01/RESULT = {"Module":{"18":"Generic"}}
16:21:46 CMD: gpio 255
16:21:46 MQT: stat/Contatore01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"176":"DDS238-2 Tx"},"GPIO5":{"177":"DDS238-2 Rx"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

16:22:25 CMD: rule1
16:22:25 MQT: stat/Contatore01/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
16:22:38 CMD: rule2
16:22:38 MQT: stat/Contatore01/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
16:22:40 CMD: rule3
16:22:40 MQT: stat/Contatore01/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
  16:23:15 CMD: status 0
16:23:15 MQT: stat/Contatore01/STATUS = {"Status":{"Module":18,"FriendlyName":["Contatore01"],"Topic":"Contatore01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:23:15 MQT: stat/Contatore01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T00:36:59","StartupUTC":"2020-04-20T14:46:16","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2020-04-14T14:50:54","SaveCount":45,"SaveAddress":"F4000"}}
16:23:15 MQT: stat/Contatore01/STATUS2 = {"StatusFWR":{"Version":"8.2.0(sensors)","BuildDateTime":"2020-03-20T14:48:25","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"364/699"}}
16:23:15 MQT: stat/Contatore01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Matrix",""],"TelePeriod":300,"Resolution":"559F80C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00600000","00000000"]}}
16:23:15 MQT: stat/Contatore01/STATUS4 = {"StatusMEM":{"ProgramSize":597,"Free":404,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8FDAA387","0415A3A4","B7FFBFCD","01DA9BC4","24367CC7","00000002"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64"}}
16:23:15 MQT: stat/Contatore01/STATUS5 = {"StatusNET":{"Hostname":"Contatore01-5606","IPAddress":"192.168.1.106","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:38:D5:E6","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
16:23:15 MQT: stat/Contatore01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.31","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_38D5E6","MqttUser":"paolo","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:23:15 MQT: stat/Contatore01/STATUS7 = {"StatusTIM":{"UTC":"2020-04-20T15:23:15","Local":"2020-04-20T16:23:15","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:49"}}
16:23:15 MQT: stat/Contatore01/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:23:15 MQT: stat/Contatore01/STATUS10 = {"StatusSNS":{"Time":"2020-04-20T16:23:15","ENERGY":{"TotalStartTime":"2020-04-14T14:46:47","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}}
16:23:15 MQT: stat/Contatore01/STATUS11 = {"StatusSTS":{"Time":"2020-04-20T16:23:15","Uptime":"0T00:36:59","UptimeSec":2219,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Matrix","BSSId":"C4:04:15:1A:E6:B6","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Tasmota webgui report the same values showed by hicking display.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-04-20:
Hi, please, set weblog to 4 for more debug information and then post the output of the console to see the serial comms between your device and Tasmota.

capputeam, 2020-04-21:
Hi, this is the output (I tried with all combination of setoption71 and setoption72 and the output is always the same):

09:26:11 WIF: Checking connection...
09:26:31 WIF: Checking connection...
09:26:31 MQT: tele/Contatore01/STATE = {"Time":"2020-04-21T09:26:31","Uptime":"0T17:40:15","UptimeSec":63615,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Matrix","BSSId":"C4:04:15:1A:E6:B6","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:08"}}
09:26:31 MQT: tele/Contatore01/SENSOR = {"Time":"2020-04-21T09:26:31","ENERGY":{"TotalStartTime":"2020-04-14T14:46:47","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0.000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}
09:26:51 WIF: Checking connection...
09:27:11 WIF: Checking connection...
09:27:31 WIF: Checking connection...
09:27:51 WIF: Checking connection...
09:28:10 WIF: Checking connection...
09:28:31 WIF: Checking connection...
09:28:51 WIF: Checking connection...
09:29:11 WIF: Checking connection...
09:29:31 WIF: Checking connection...
09:29:51 WIF: Checking connection...
09:30:11 WIF: Checking connection...
09:30:31 WIF: Checking connection...
09:30:51 WIF: Checking connection...
09:31:11 WIF: Checking connection...
09:31:31 WIF: Checking connection...
09:31:31 MQT: tele/Contatore01/STATE = {"Time":"2020-04-21T09:31:31","Uptime":"0T17:45:15","UptimeSec":63915,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Matrix","BSSId":"C4:04:15:1A:E6:B6","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:08"}}
09:31:31 MQT: tele/Contatore01/SENSOR = {"Time":"2020-04-21T09:31:31","ENERGY":{"TotalStartTime":"2020-04-14T14:46:47","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0.000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}
09:31:51 WIF: Checking connection...
09:32:11 WIF: Checking connection...
09:32:31 WIF: Checking connection...
09:32:51 WIF: Checking connection...
09:33:11 WIF: Checking connection...
09:33:31 WIF: Checking connection...
09:33:51 WIF: Checking connection...
09:34:11 WIF: Checking connection...
09:34:31 WIF: Checking connection...
09:34:51 WIF: Checking connection...
09:35:11 WIF: Checking connection...
09:35:31 WIF: Checking connection...
09:35:51 WIF: Checking connection...
09:36:11 WIF: Checking connection...
09:36:31 WIF: Checking connection...
09:36:31 MQT: tele/Contatore01/STATE = {"Time":"2020-04-21T09:36:31","Uptime":"0T17:50:15","UptimeSec":64215,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Matrix","BSSId":"C4:04:15:1A:E6:B6","Channel":11,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:08"}}
09:36:31 MQT: tele/Contatore01/SENSOR = {"Time":"2020-04-21T09:36:31","ENERGY":{"TotalStartTime":"2020-04-14T14:46:47","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0.000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}
09:36:51 WIF: Checking connection...
09:37:11 WIF: Checking connection...
09:37:31 WIF: Checking connection...
09:37:51 WIF: Checking connection...
09:38:11 WIF: Checking connection...
09:38:31 WIF: Checking connection...
09:38:51 WIF: Checking connection...
09:39:11 WIF: Checking connection...
09:39:31 WIF: Checking connection...
09:39:51 WIF: Checking connection...
09:40:11 WIF: Checking connection...
09:40:31 WIF: Checking connection...
09:40:51 WIF: Checking connection...
09:41:11 WIF: Checking connection...
09:41:31 WIF: Checking connection...
09:41:31 MQT: tele/Contatore01/STATE = {"Time":"2020-04-21T09:41:31","Uptime":"0T17:55:15","UptimeSec":64515,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Matrix","BSSId":"C4:04:15:1A:E6:B6","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:08"}}
09:41:31 MQT: tele/Contatore01/SENSOR = {"Time":"2020-04-21T09:41:31","ENERGY":{"TotalStartTime":"2020-04-14T14:46:47","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0.000,"Power":0.000,"ApparentPower":0.000,"ReactivePower":0.000,"Factor":0.00,"Voltage":0.000,"Current":0.000}}
09:41:51 WIF: Checking connection...
09:42:11 WIF: Checking connection...
09:42:31 WIF: Checking connection...
09:42:51 WIF: Checking connection...
09:43:11 WIF: Checking connection...
09:43:31 WIF: Checking connection...

ascillato, 2020-04-21:
From your log, your DDS238 is not sending anything to Tasmota. Please check your wirings and the power of your DDS238.

capputeam, 2020-04-21:
The wiring seems to be ok, the LCD display on the DDS238 is reporting correct values for Current, Voltage, Power consumption (I checked with a multimeter).
[image]

arendst, 2020-04-21:
It's the wiring between the DDS238 AND a Tasmota device!

Is there any wiring between the DDS238 and Tasmota using the modbus protocol?

ascillato, 2020-04-21:
If you have connected the communication wires from DDS238 to TASMOTA as you have posted before ( ` "GPIO4":{"176":"DDS238-2 Tx"},"GPIO5":{"177":"DDS238-2 Rx"} ` ) may be your wires are scrambled. Try swapping TX and RX.

capputeam, 2020-04-21:
Nothing happens swapping GPIO4 and GPIO5.
Then I tried other wiring, I found this:

17:14:37 CMD: gpio 255
17:14:37 SRC: WebConsole from 192.168.1.67
17:14:37 CMD: Group 0, Index 1, Command "GPIO", Data "255"
17:14:37 MQT: stat/Contatore01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"176":"DDS238-2 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"177":"DDS238-2 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
17:14:57 DMP: 
17:14:57 DBG: DDS2382 response error 7
17:14:57 WIF: Checking connection...
17:15:17 WIF: Checking connection...
17:15:37 WIF: Checking connection...
17:15:55 DMP: 01 01 00 01 01 12 55 AA 03 07 00 08 12 02 00 04 00 00 02 37 62 55 AA 03 07 00 08 13 02 00 04 00 00 03 C7 F4 55 AA 03 07 00
17:15:55 DBG: DDS2382 response error 9
17:15:56 DMP: 
17:15:56 DBG: DDS2382 response error 7
17:15:57 DMP: 
17:15:57 DBG: DDS2382 response error 7

capputeam, 2020-04-23:
I started supposing that my device is the same or similar to the one used in https://github.com/arendst/Tasmota/pull/6384 where the support for DDS238-2 has been introduced. It was a Hiking DDS238-2 ZN/S https://it.aliexpress.com/item/32812065998.html insteat I have a Hiking DDS238-2 WIFI https://it.aliexpress.com/item/4000571797301.html?spm=a2g0s.9042311.0.0.7fda4c4djuKqyA . 

I found wiring configuration googleing around (using GPIO4 and 5), not sure whether  it is the same used during the initial integration phase with the DDS232-2 ZN/S or not.

Maybe I'm working with wrong assumptions.

capputeam, 2020-04-25:
After a lot of test and disassebling the device, I found that it is based on the TYWE3S module connected to a main board based on the MCU G80F923 (from Chinese company Greenchips).
The TYWE3S module is connected only by 6 pin with the main board: TX, RX, GPIO16, GPIO14, GND and VCC.
The only wiring combination that reports something is when I use the following configuration.
[image]
But the only thing that I see in the console is 

DDS2382 response error 7

Any idea about this error? Or how I can resolve it?

arendst, 2020-04-26:
I'm not sure if I understand you correctly but it was never the plan to replace the code within the DDS device. Tasmota expects a serial connected DDS to communicate with modbus protocol to a tasmotized esp8266 device.

capputeam, 2020-04-28:
Yes, and that is what it is.
TYWE3S is the component with tasmota firmware, the main board is the one with the microcontroller who make the job (power metering).
They are connected through 6 wires that (from the TYWE3S spec) are : TX, RX, GPIO16, GPIO14, GND and VCC.
With the configuration above (using TX and RX) I got an error "DDS2382 response error 7" on the console and conseguently no values are collected by tasmota and sent to MQTT topic (all zeros).

arendst, 2020-04-28:
As I tried to explain. THAT IS NOT SUPPORTED IN TASMOTA!!!

The way EXTERNAL energy monitoring devices are handled is by using an ESP8266 device connected by a serial connection to a device capable of communicating via the Modbus protocol.

Your solution doesn't use Modbus protocol but a dedicated protocol between the TyWE3S and the microcontroller. That protocol is not supported by Tasmota.

Initially your TYWE3S contained software capabel of Modbus communication using RS485 connectors as shown below:

[image]

Those connectors were supposed to connect to an external ESP8266 device running Tasmota and configured for DDS2382. You should not have changed the software on the internal TYWE3S.

capputeam, 2020-04-30:
Ok, thanks for your support.

closed 2020-04-30 by capputeam

thiagoarreguy, 2020-06-05:
> I started supposing that my device is the same or similar to the one used in #6384 where the support for DDS238-2 has been introduced. It was a Hiking DDS238-2 ZN/S [...]

Try to buy this: https://www.aliexpress.com/item/4000530641386.html

capputeam, 2020-06-05:
I partially solved the issue using the following configuration

[image]

only few parameters are available (total power, instant power, voltage, current).

But it works.

thiagoarreguy, 2020-06-05:
> I partially solved the issue using the following configuration
> 
> [image]
> 
> only few parameters are available (total power, instant power, voltage, current).
> 
> But it works.

But you used tasmota.bin instead tasmota-sensors.bin

Why dont you use http://thehackbox.org/tasmota/tasmota-sensors.bin ?

capputeam, 2020-06-05:
Because with Tuya MCU module it is not needed.

tasmota-sensors is needed if you want to use DDS238-2 wiring.

inorhan, 2020-06-15:
> 
> 
> I partially solved the issue using the following configuration
> 
> [image]
> 
> only few parameters are available (total power, instant power, voltage, current).
> 
> But it works.

Hello thiagoarreguy.

 I also have a dds238-2 wifi, and I'm using your approach. 
I have the relay ON, but Current =0, Power=0 (I have a load), and Voltage appears (with a very high value) then disappears at all.
I used: Backlog TuyaMCU 33,20; TuyaMCU 32,18; TuyaMCU 31,19.
I think I'm missing some kind of value correction /10 /100 ... right? But don't know how to do it.

RESULT = {"TuyaReceived":{"Data":"55AA030700081402000400037922C9","Cmnd":7,"CmndData":"1402000400037922","DpType2Id20":227618,"20":{"DpId":20,"DpIdType":2,"DpIdData":"00037922"}}}

37922C9  ->  37922  ->hex a dec ->  227618   ->227vac

What commands did you use?

Thanks!

inorhan, 2020-08-18:
Hello. 
Has someone configure the "dds238-2 wifi" ?
I used: Backlog TuyaMCU 33,20; TuyaMCU 32,18; TuyaMCU 31,19. ... but tasmota it's not reading the right values.

inorhan, 2020-09-03:
Hi again.
Added the template for dds2812-2 wifi, here https://templates.blakadder.com/index.html.
TuyaMCU 33,20 it's not working correctly, so I used hassio for getting the Volts. If someone knows how to do it in Tasmota please answer back.

tasmota:
{"NAME":"dds2382wifi","GPIO":[0,107,0,108,0,0,0,0,0,0,56,0,17],"FLAG":0,"BASE":1}
Backlog TuyaMCU 33,20; TuyaMCU 32,18; TuyaMCU 31,19; SetOption19 1; SetOption53 1; SetOption21 1; SetOption66 1;

sensors.yaml:

  - platform: mqtt
    name: "Automatico dds2382wifi de tuya"
    state_topic: "dds2382wifi/tele/RESULT"
    value_template: >- 
      {% set message = value_json.TuyaReceived['20'].DpIdData | int(base=16) %}
      {{ (message|float) / 1000.0 }}
    unit_of_measurement: 'V'
    icon: mdi:alpha-v-circle-outline

capputeam, 2020-09-11:
Hi hinorhan,

I use a rule defined in that way:

CMD: Rule1 on System#Boot do RuleTimer1 10 endon on Rules#Timer=1 do backlog Baudrate 9600; SerialSend5 55AA0008000007; RuleTimer1 10 endon

the response is good to me and voltage is always fine.

inorhan, 2020-09-11:
> 
> 
> Hi hinorhan,
> 
> I use a rule defined in that way:
> 
> CMD: Rule1 on System#Boot do RuleTimer1 10 endon on Rules#Timer=1 do backlog Baudrate 9600; SerialSend5 55AA0008000007; RuleTimer1 10 endon
> 
> the response is good to me and voltage is always fine.

Thanks capputeam, but I still have the same results.
I don´t know why, but in three diferent DDS238-2 Wifi, Voltage shows 3178 VAC or similar (it's not getting the right resolution or it's doing an incorrect HEX->DEC conversion in the volts).

16:35:01 MQT: tele/hola/RESULT = {"TuyaReceived":{"Data":"55AA030700081402000400037C24CE","Cmnd":7,"CmndData":"1402000400037C24","DpType2Id20":228388,"20":{"DpId":20,"DpIdType":2,"DpIdData":"00037C24"}}}
16:35:01 TYA: fnId=33 is set for dpId=20
16:35:01 TYA: Rx ID=20 Voltage=31780

00037C24 HEX -> 228388 DEC -> 228,388VAC (correct) -> tasmota shows 3178 VAC

Gui576, 2020-09-16:
Hello All,

Thanks for the update, I have successfully get DDS238-2 WIfi working with your instruction.

But , if I enable MQTT I get all value at 0 even in the webgui ?

Without MQTT it's working  : 

[image]

Do you have an idea ?

inorhan, 2020-09-16:
> 
> 
> Hello All,
> 
> Thanks for the update, I have successfully get DDS238-2 WIfi working with your instruction.
> 
> But , if I enable MQTT I get all value at 0 even in the webgui ?
> 
> Without MQTT it's working :
> 
> [image]
> 
> Do you have an idea ?

Hi. Works for me with mqtt, maybe we have different hardware versions.
Do you have SetOption66 enable ? If so try disabling it.

sebitnt, 2020-10-19:
Hi!
I finally got my DDS238-2 WiFi working (a bit) with Tasmota. I unsoldered the original WiFi-Module and exchanged it with an ESP12-F module because OTA flashing was not possible and I wanted the possibility to revert back.
It did not seem to correctly interpret the measured values with the newest Tasmota 8.5 on it. I got all zero readings back than.  (In the console I did not even get something like " TYA: Rx ID=20 Voltage=2315", but raw values were received fine.) After downgrading to 8.4.0 the Voltage, Power and Current is shown fine.
Now I am wondering how you guys got the power factor etc. values working. Is there anything I have to do further?
Greetings
Sebi

sebitnt, 2020-10-19:
Well, some problems are selv-solving :-) The power factor seems to be calculated by Tasmota on its own and of course is shown when a load is being connected.
But two questions remain for me: Is there some way of getting the values for positive and negative energy and the frequency directly over the TuyaMCU serial line?
Can somebody say anything about DpID's 9 an 101? 9 seems to be some kind of counter. If you set it to 0xf for example, it will count down to 0. But I have absolutely no clue about ID 101.

Gui576, 2020-10-23:
Hello,

On my side I get an issue with mqtt I was unable to solve it ...

So all is working fine, I have flash sucessfully with tuya convert, maybe you have a new firmware release who block the flash.

On my side on web interface all is working => 

[image]

krikk, 2020-11-26:
> Can somebody say anything about DpID's 9 an 101? 9 seems to be some kind of counter. If you set it to 0xf for example, it will count down to 0. But I have absolutely no clue about ID 101.

i am pretty sure that 101 is Total kWh:  i have a value of 30 (0x1e) on ID101 which means 0.30 kWh Total ...and the value counts up, so now i need to find a way to configure this in tasmota..

inorhan, 2020-11-26:
> i am pretty sure that 101 is Total kWh: i have a value of 30 (0x1e) on ID101 which means 0.30 kWh Total ...and the value counts up, so now i need to find a way to configure this in tasmota..

No idea about DpID's 9. Maybe the indication for + or - watts ?

If it helps you, this is the way to do DpID's 101 in Hassio:

  - platform: mqtt
    name: "Total Energy kWh"
    unique_id: "Total Energy kWh breaker1"
    state_topic: "tele/tasmotaenergy/RESULT"
    value_template: >- 
      {% set message = value_json.TuyaReceived['101'].DpIdData | int(base=16) %}
      {{ (message|float) / 10.0 }}
    unit_of_measurement: 'kWh'
    icon: mdi:alpha-k-circle-outline

sebitnt, 2020-11-26:
> i am pretty sure that 101 is Total kWh: i have a value of 30 (0x1e) on ID101 which means 0.30 kWh Total ...and the value counts up, so now i need to find a way to configure this in tasmota..

Yes, you are totally right! I have just checked it on my meter. My value is 0x4DE4 (19940) and the meter is showing 199.4 kWh.
I tested to set SetOption72 to 1 but it did not overwrite the total energy. Any ideas?

noisemaker00, 2020-12-15:
I bought one of these power meter, any updates about flashing with Tasmota?
I opened the device and unsoldered the TYWE3S and temporaly replaced with an ESP8266 dev board to make things easier.
The TYWE3S module has 6 pads to connect to another MCU, these connection (accordin with TYWE3S pinout) are:
VCC
GND
RX
TX
GPIO-14
GPIO-16
So can I use this template to flash the ESP8266?
[https://templates.blakadder.com/hiking_DDS238-2.html](https://templates.blakadder.com/hiking_DDS238-2.html)

krikk, 2020-12-15:
i did the same, see here: https://github.com/krikk/Hiking-DDS238-2-WIFI-Din-Rail-Energy-Meter-flashing-Tasmota

noisemaker00, 2020-12-31:
according to your experience on this device, is possible to add a buzzer and play a "beep" sound when tot power consumption is greater than a fixed value? Maybe with Tasmota rules?

noisemaker00, 2021-01-07:
I tried to activate MQTT, but then all values turned to zero. The I tried to press the "Reset Configuration" but when the microcontroller restart I hear a click from the device and now the red light is on (according to manual it means the relay is off).
How can I turn it on again?
Can I perform this via tasmota console?

krikk, 2021-01-08:
@noisemaker00: i don't know which version what you are using but with recent versions of tasmota nearly everything is working, for a full guide see my instructions posted above...

serpini1, 2021-05-13:
I have a problem with this device, after tasmotizer it and change module settings, on the main page, there isn't unit of measurement
[image]
There isn't voltage, current, power.... Why there isn't any information?
This is my module config
[image]
Thanks!

juanmito, 2021-09-15:
Hi.

I'm having same problem. I've tested all solutions provided here, and nothing result :(.

When I use the option "weblog 4", i only see this messages:

`:13:28 APP: Boot Count 12
20:13:28 CFG: Saved to flash at FB, Count 33, Bytes 4096
20:13:29 TYA: Send "55aa00000000ff"
20:13:32 RSL: tasmota/discovery/A4E57C0BD449/config = {"ip":"192.168.1.234","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota_0BD449-5193","mac":"A4E57C0BD449","md":"Hiking DDS238-2 WIFI","ty":1,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.2.0","t":"tasmota_0BD449","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1}
20:13:32 RSL: tasmota/discovery/A4E57C0BD449/sensors = {"sn":{"Time":"2021-09-15T20:13:32"},"ver":1}
20:13:40 TYA: Send "55aa00000000ff"
20:13:41 WIF: Checking connection...
20:13:51 TYA: Send "55aa00000000ff"
20:14:01 WIF: Checking connection...
20:14:02 TYA: Send "55aa00000000ff"
20:14:13 TYA: Send "55aa00000000ff"
20:14:21 WIF: Checking connection...
20:14:24 TYA: Send "55aa00000000ff"
20:14:35 TYA: Send "55aa00000000ff"
20:14:41 WIF: Checking connection...
20:14:46 TYA: Send "55aa00000000ff"
20:14:57 TYA: Send "55aa00000000ff"
20:15:01 WIF: Checking connection...
20:15:08 TYA: Send "55aa00000000ff"
20:15:19 TYA: Send "55aa00000000ff"
20:15:21 WIF: Checking connection...`

Any approach? 

Thanks!

g6094199, 2022-05-10:
is see that the frequency which is shown in the display is missing in tasmota.
is it possible to add this data, too?

olonsoft, 2022-10-23:
Hello,
 I have the same problem as @serpini1 
I don't see any values. But the terminal shows values coming:
[image]
[image]

Any help, please!