arendst/Tasmota #8247: Buzzer with PWM

DD8HL, 2020-04-21:
I wanted to use a passive Buzzer wich needs PWM to function, but realized that the "buzzer" function doesnt allow the use of pwm. Would be pretty useful for all passive buzzers if those two could be combined somehow, otherwise I have to think about a routine to turn pwm on and off more or less manually.

closed 2020-04-21 by close-issue-app[bot]

DD8HL, 2020-04-21:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I want to use a passive buzzer with tasmota
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
the possibility to combinde pwm with buzzer
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
create a rule that turns pwm on and off with delays
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

obilginer, 2020-06-03:
i also need solution for this issue