arendst/Tasmota #8250: Dimmer and color values lost on wifi reconnect

DavinKD, 2020-04-22:
### PROBLEM DESCRIPTION
When wifi reconnects (AP rebooted, etc), the stored Dimmer and color values are lost.  Therefore, when the light is turned back on, it does not use the previous color/brightness when it was last turned off.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Color bulb
- [ ] Tasmota binary firmware version number used: 8.2.0.3
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:50:42 MQT: stat/3074/RESULT = {"NAME":"RGB Switch","GPIO":[50,255,52,255,39,38,255,255,49,17,37,255,255],"FLAG":0,"BASE":18}
19:50:42 MQT: stat/3074/RESULT = {"Module":{"0":"RGB Switch"}}
19:50:42 MQT: stat/3074/RESULT = {"GPIO0":{"50":"PWM5i"},"GPIO1":{"0":"None"},"GPIO2":{"52":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"39":"PWM3"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"49":"PWM4i"},"GPIO13":{"17":"Button1"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

No rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

19:51:53 MQT: stat/3074/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"3074","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:51:53 MQT: stat/3074/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"BedroomLamps","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"13T07:52:31","StartupUTC":"2020-04-08T18:59:22","Sleep":50,"CfgHolder":4617,"BootCount":53,"BCResetTime":"2020-02-11T15:34:33","SaveCount":4182,"SaveAddress":"FA000"}}
19:51:53 MQT: stat/3074/STATUS2 = {"StatusFWR":{"Version":"8.2.0.3(ef9ffaa-tasmota)","BuildDateTime":"2020-04-08T17:00:48","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"341/699"}}
19:51:53 MQT: stat/3074/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dameron_smart",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000080","00000008"]}}
19:51:53 MQT: stat/3074/STATUS4 = {"StatusMEM":{"ProgramSize":580,"Free":420,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:51:53 MQT: stat/3074/STATUS5 = {"StatusNET":{"Hostname":"3074-3074","IPAddress":"192.168.0.240","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"80:7D:3A:1A:CC:02","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
19:51:53 MQT: stat/3074/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1ACC02","MqttUser":"DVES_USER","MqttCount":5,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:51:53 MQT: stat/3074/STATUS7 = {"StatusTIM":{"UTC":"2020-04-22T02:51:53","Local":"2020-04-21T19:51:53","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-07:00","Sunrise":"21:46","Sunset":"11:51"}}
19:51:53 MQT: stat/3074/STATUS10 = {"StatusSNS":{"Time":"2020-04-21T19:51:53"}}
19:51:53 MQT: stat/3074/STATUS11 = {"StatusSTS":{"Time":"2020-04-21T19:51:53","Uptime":"13T07:52:31","UptimeSec":1151551,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,0,0","White":0,"CT":500,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dameron_smart","BSSId":"76:83:C2:B1:A0:12","Channel":6,"RSSI":100,"Signal":-46,"LinkCount":3,"Downtime":"0T00:00:22"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Not console output, but logs from my MQTT server.
9:59:55 AM : Payload:  3074 - [{"Time":"2020-04-21T09:59:54","Uptime":"12T22:00:32","UptimeSec":1116032,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":20,"Color":"3300000000","HSBColor":"0,100,20","White":0,"CT":500,"Channel":[20,0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dameron_smart","BSSId":"76:83:C2:B1:A0:12","Channel":6,"RSSI":100,"Signal":-45,"LinkCount":3,"Downtime":"0T00:00:22"}}]

**Access Point was rebooted between these two tele periods

10:04:54 AM : Payload:  3074 - [{"Time":"2020-04-21T10:04:54","Uptime":"12T22:05:32","UptimeSec":1116332,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"POWER":"OFF","Dimmer":0,"Color":"0000000000","HSBColor":"0,0,0","White":0,"CT":500,"Channel":[0,0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":5,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dameron_smart","BSSId":"76:83:C2:B1:A0:12","Channel":6,"RSSI":100,"Signal":-44,"LinkCount":3,"Downtime":"0T00:00:22"}}]

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1.  Set dimmer to 20, etc
2.  Turn off light
3.  Reboot AP
4.  Check dimmer level, it will now be 0.  Color, CT, etc is affected as well.

### EXPECTED BEHAVIOUR
Dimmer, Color, CT, etc should be retained.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-04-22:
Will look at it.

ascillato, 2020-04-22:
Your wificonfig is 2, that means that when your wifi disconnects it will reset and start the Access Point mode. So, your lost of dimmer settings are intended because your config set to do that.

Use instead the default of wificonfig 4 that is retry the wifi connection without restarting

s-hadinger, 2020-04-22:
Correct, I cannot reproduce with Wificonfig 4

DavinKD, 2020-04-22:
Is there no way to preserve settings across reboot with wificonfig 2?

ascillato, 2020-04-22:
No, because it is a reset. It is the same as unplug and plug again.

If you want, you can use rules OR you can move the control to your home automation software, in that case you can set all the commands to be retained

ascillato, 2020-04-22:
Or just use wificonfig 4

DavinKD, 2020-04-22:
Thanks for the info.  Closing issue.

closed 2020-04-22 by DavinKD