arendst/Tasmota #8260: Please add VEML7700 sensor

nsitt, 2020-04-23:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Hi, I tried to adapt existing I2C sensors to cover the VEML7700 but was not successful. I am giving up. There already exists a adafruit library, so I guess that someone who is more skilled may be able to add the VEML7700 support quite easily.
Thank you for your assistance!

ascillato, 2020-04-23:
Hi, This request is duplicated. It was already requested but nobody had provided a pull request. See https://github.com/arendst/Tasmota/issues/5720

ascillato2, 2020-04-23:
May @subdancer can give some help or code?

ascillato2, 2020-04-23:
@nsitt a working Tasmota driver for this can be found at https://github.com/subdancer/Tasmota/tree/master

As you have this sensor, you can use this code to test and provide a PR if you want

ascillato2, 2020-04-26:
@nsitt 

Hi, any news on this? Do you need any help?

ascillato, 2020-04-30:
@nsitt

Hi, any news on this? Do you need any help?

device111, 2020-05-01:
Hi,
i had the VEML6075.
I have begun to implement the VEML7700 and the VEML6075 with the Adafruit Libary.
on my Fork is a test with both Sensors on actual tasmota dev. (not full completed)
tests with the VEML7700 are welcome. I have no hardware for the 7700.

nsitt, 2020-05-03:
@device111 I compiled it and tried to load it on a sonoff basic. Getting an error" Upload Failed: Upload buffer miscompare" tasmota-minimal.bin was loaded first. Working fine when loading other binaries after tasmota-minimal.bin . Guessing something is eating too much memory?

ascillato, 2020-05-03:
> Upload Failed: Upload buffer miscompare

That means that the file you are uploading is bigger than the available space. This shouldn't happens if the minimal is loaded. Is your compiled version bigger than 660KB?

device111, 2020-05-03:
You must deactivate  most Sensors. The 6075 and 7700 with the Libary use a lot of space. Try to load the compressed bin after load minimal.
I have terminated my Fork and upload a Version with native 6075 and 7700 with libary.

device111, 2020-05-03:
New dev-Fork is Online. Now, the VEML6075 is without libary and the 7700 with Adafruit lib.

nsitt, 2020-05-04:
I tested the new dev version. It is showing VEML7700 data, however it looks as if VEML7700 code is recognized even when VEML6070 is active although these are on different addresses:

This is what I have on the I2C bus:
VEML7700 at 0x10
BMP280 at 0x38 0x39
VEML6070 at 0x76

13:48:04 CMD: i2cscan
13:48:04 MQT: stat/tasmota/RESULT = {"I2CScan":"Device(s) found at 0x10 0x38 0x39 0x76"}

 Here you can see the view when both VEML7700 and VEML6070 are attached
[image]
Here you can see the view when VEML6070 is attached
[image]

nsitt, 2020-05-04:
correction: 
VEML6070 is at 0x38, 0x39. 
BMP280 at 0x76

device111, 2020-05-04:
can you make a screenshot with only 6070 attached  and an I2Cscan on Console?

device111, 2020-05-04:
one question...why do you have different Screenshots? one is with Version 8.2.0 and one is with actual 8.2.0.6...

nsitt, 2020-05-04:
@device111 I am surprised, too. All I did is to turn the sonoff off, disconnect Veml7700 and turn sonoff on again. I did it again and now it is showing 8.2.0.6 in both screens. 

Screenshots were taken within 65 seconds.

All sensors connected:
[image]

Veml6070 disconnected:

[image]

Providing 2 other screenshots:
1. Only BMP280 connected. VEML7700 is showing data?
[image]

The last screenshot is with only VEML6070 connected (7700 and BMP280 disconnected) 

[image]

23:36:28 CMD: i2cscan
23:36:28 MQT: stat/tasmota/RESULT = {"I2CScan":"Device(s) found at 0x38 0x39"}

To me it seems as if 7700 is not working properly. I am trying to code it myself without library using the examples given but I it is going slowly.

device111, 2020-05-05:
ok, i have seen an issue in Libary. Now is fixed.
Try it.

nsitt, 2020-05-05:
Tested the new code:
08:13:33 CMD: i2cscan
08:13:33 MQT: stat/tasmota/RESULT = {"I2CScan":"Device(s) found at 0x10 0x38 0x39 0x76"}

All sensors attached:
[image]

7700 only:
[image]

To me it seems 6070 and 7700 can not coexist. 

Martin, I think we should synch up offline and see if we can fix in direct communication, seems we are both from Bavaria. email: tasmota@how2.de

nsitt, 2020-05-05:
After some back and forth over email, Martin compiled 8.2.0 with the VEML7700 code and this fixed the issues. For unknown reasons VEML6070 is broken in 8.2.0.6

[image]

ascillato2, 2020-05-10:
@nsitt 

Hi, any news on this? Can you provide a PR for adding this sensor to Tasmota? Thanks.

device111, 2020-05-10:
Hi Adrian,

i must do some checks and then i make a PR in the next view days.

ascillato2, 2020-05-16:
Closing this issue as a PR has been submitted for review at https://github.com/arendst/Tasmota/pull/8432

Thanks for sharing your ideas and work. Very appreciated.

closed 2020-05-16 by ascillato2