arendst/Tasmota #8262: GUI - Create GUI strings using sub-string (result not so good for translation)

bovirus, 2020-04-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Some part of GUI sued compiste string result of join of sub-string.

- Firmware ugrade window

- Start upgrade etxt button in web server firmware upgrade. seems depend by this string

#define D_START_UPGRADE "Start upgrade"

- Start upgrade for local firmware window seems the composition of two sub-string

#define D_START "Start"
#define D_UPGRADE "upgrade"

Why don't use 

#define D_START_UPGRADE "Start upgrade"

also for local file firmware upgrade?

- Start restore

Start restore is a compition of 

#define D_START "Start"
#define D_RESTORE "restore"

Why don't use a speifc string?

- Uplaod Sucecsfully (in green) after firmware upgrade

Seemd the result of combination of

#define D_UPLOAD "Upload"
#define D_SUCCESSFUL "Successful"

why don't sue instead

#define D_UPLOAD_DONE "Upload done"

Thanks.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expaceted less possible sue of comination string that can be good for English but not so good for other languages.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2020-04-23:
Its a waste of code. You're welcome!

closed 2020-04-23 by blakadder

bovirus, 2020-04-23:
@bovirus 

Could be a waste of code if you have only English UI.

But if you decide to have a multilingual UI the UI is very buggy like now to combine string to create string.

You get ridiculous result in language that are not English.

And it's very bad to like you did to close the issues without propoisng a real solution.

It's mroe common for you to close the iseeu, because you are not inteersted or you are not able to do it.

Becasue you thinkk only about the cdoe.
But your UI is based on "age of stone" technology.

And sincerly I'm quite unsatisfied by your level of udnerstanding of real needs.

because you didn't read and think only about English.
And then plwase remove all not English languAge.

Please don't offfnd the intelligence of Italian users and the italian ruels and gramamr.

What do you think If I create a program in Italian and I make the Englih user interface combining single word in English?

Please explin me why you shloudl vreate combine string (Start upgrade) when the same string is already availble.

**_This one is waste of space.
You use space for do ssomething that is already donne._**

blakadder, 2020-04-23:
I'm sorry but all the examples you posted were in english and you didn't give any useful information on the impact on other languages.

Yes, this is not Apple where user experience is paramount, You seem to miss the fact everything is operating on a "stone age" processor hence the "stone age" technology.

"Real needs" differ amongst people, this project is open source and you're free to fork it and create something more suitable to your needs

English is my 3rd language.

I didn't even know your nationality nor do I care much about it, knowledge of languages isn't about where you're from but how disciplined in learning you are

Go for it! As I already mentioned, it is open source.

Creating a new string takes up valuable flash and ram space. Once again, you're working with below arm level processor and very limited (i)ram.

If you find the interface buggy please post a detailed bug reports using the provided template,

effelle, 2020-04-23:
@bovirus 
You should first learn how our code works.
Let's take `D_START` for example, it is used on `support.ino`, `xdrv_01_webserver.ino`, `xdrv_10_scripter.ino`, `xdrv_11_knx.ino`, `xdrv_23_zigbee_converters.ino` and `xsns_66_iAQ.ino` and is obviously needed on all the translation files. 
As you can see is not an easy change at all and our goal is to reach a compromise between information and functionality, where functionality takes precedence.
Also, Tasmota was created for devices controlled by an automation system like Home Assistant, OpenHab or Domoticz.  
Sure, you can use a Tasmotized device standalone, but is not what we have in mind for it.
Honestly the interface it's probably one of the last things that worries us.

bovirus, 2020-04-23:
@blakadder 

In which language I have to explain a problem about string than English.?
I'm not talking about the translation of Italian is right or not.
I'm talikng about source code and how developer use strings.

Sometimes use specific string

Sometims Use single word to combine a string.
Could be OK in English but is not good if you are a multi-linguage UI.

Like use for the same string "Start upgrade" in two different button in the same window (Firmware upgrade) 

one time use specific "Start upgarde" string

One time use string "Start " + string "upgrade"

The result using multi.-lingual interface dind't match.
You have two different translations.

Then for the second button you should ude the same string "Start upgrsde" already availsble and used for first button.

This is I know as save space/memory.

blakadder, 2020-04-23:
that's why its the translator's job to adapt to the existing constraints, like translating any other work. Do you change Shakespeare's sonets from iambic pentameter into something you like more?

bovirus, 2020-04-23:
@blakadder 

I'm so sorry but you expanation is out of context.
We are not talking about paragraph but at low level (words)

The Shakespear sonet in Italy are published in English?
I don' believe.
But there are professional translator author that as knowledge iof Shakespear that translate it.

I don't beleiev that an ptofessional Italian translator of Shakespear sonet get word by word and translate it from English to Italian.

Because in this case are out of context.

Because English and Italian are different and sue grammatical rules (ex. Italian has male/female that English not has).

I think that for anyone is not logical to use for a GUI to create a string combining single word translated.
Could be OK for English but not for not English language.
Or you are not agree about it?

In the same time pelase explain me why you have to use a combine string "Start" + "upgrade" when "Start upgrade" is available as string.

blakadder, 2020-04-23:
I'm not sure if you're joking or being serious now? Maybe its the language barrier...

> In the same time please explain me why you have to use a combine string "Start" + "upgrade" when "Start upgrade" is available as string.

It has already been explained https://github.com/arendst/Tasmota/issues/8262#issuecomment-618519540

In short: it will not be done since its requiring more code space (despite what you **think** creating more strings is not space saving) and a breaking change for entire codebase and other language translations.

bovirus, 2020-04-23:
@blakadder 

You said  that me where you using

D_START + D_UPGRADE 

you cannot use

D_START_UPGRADE 

I don't believe.
If you use in the same window for the same button in otehr position a unqiue word what kind of problem could be to use the same unique word also for second button with exactly the same meaning?

blakadder, 2020-04-23:
since your beliefs and thoughts do not correlate with facts noone can help you until you align those two. arrivederci!

bovirus, 2020-04-23:
If when ir eported the problem you red teh code probably could fix without so many problem

**_- file ttasmota\xdrv_01_webserve.ino_**

Button right ("Start upgrade")

  "<br><button type='submit'>" D_START_UPGRADE "</button></form>" 

Button wrong (combined string)

<button type='submit' onclick='eb(\"f1\").style.display=\"none\";eb(\"f2\").style.display=\"block\";this.form.submit();'>" D_START " %s</button>

if you replace with

<button type='submit' onclick='eb(\"f1\").style.display=\"none\";eb(\"f2\").style.display=\"block\";this.form.submit();'>"  D_START_UPGRADE" </button>

could be usful?

barbudor, 2020-04-23:
Dear @bovirus 

You suggest to replace
```<button type='submit' onclick='eb(\"f1\").style.display=\"none\";eb(\"f2\").style.display=\"block\";this.form.submit();'>" D_START " %s</button>```

by 
```<button type='submit' onclick='eb(\"f1\").style.display=\"none\";eb(\"f2\").style.display=\"block\";this.form.submit();'>"  D_START_UPGRADE" </button>```

But what about the Start Restore then ?
This would require to create another string as 
```<button type='submit' onclick='eb(\"f1\").style.display=\"none\";eb(\"f2\").style.display=\"block\";this.form.submit();'>"  D_START_RESTORE " </button>```

which would use twice the memory.

Have you made all those proposed changes in a fork and proven that this will overall save code size ?

meingraham, 2020-04-23:
https://www.youtube.com/watch?v=OQqeTEYBFfM

I think @bovirus may be sitting in both seats :wink:

bovirus, 2020-04-23:
@meingraham 

Thanks for your positive contribuite!!!
Of course are you are not interested at all at this problem because your language is English.

To save as much as possible the space/memory and make a puzzle with words and this the result.

Probably the best way is to leave the UI only in English.
Because the UI source code is designed think only about English language.

barbudor, 2020-04-23:
@bovirus none of those that answered you today are english native speakers.
We are french, german, italian, slovak, dutch, ... professional developers

So if your solution is _actually_ really reducing memory, just make the PR to prove it.
This is how OpenSource works.
Act, not speak

meingraham, 2020-04-23:
@barbudor - This should answer the question :wink:

- Ecuadorian mother, American father, Parisian sister, Honduran sister
- Residences: California, Madrid, Iowa, Quito, North Carolina, Georgia, Munich, South Carolina

So, native English but fluent also in Spanish.

@bovirus - knowing a language (even multiples) and _communication_ are two different things.

bovirus, 2020-04-23:
@barbudor 

I spent too many time to explain something that it's logic.

I create a fork from original tasmota project
Replace the Italian language with many strings revised and corrected and out of context.
Then built the new firmware with tasmocompiler.
I Flashed firmware. All is OK, And benw strings in place,
Space used: 15 bytes plus original firmware (I saved a lot of space removing original ripetitive strings...).
I reduce a lot my current memory (26KB) of 0.05%.

Now I can say: I want and I do it.

Thanks so many people help me a lot.