arendst/Tasmota #8263: Turning ON or OFF the light with the Rotary Dial

poofyteddy, 2020-04-23:
**Have you looked for this feature in other issues and in the docs?**  
Yes, and i didn't find much, i guess the product isn't used by many ?
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No issue, just an improvement request
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Right now, you turn the light on and off by pushing the button, it work well but light up at the last saved brightness
What i would like it the same behavior than the original Mi Desk lamp firmware have.

- light is off, turning the rotary dial up turn on the light at 1% brightness
- light is on and at 1% brightness, turning the dial down one last time turn the light off

That way, no longer blinded at 2am in the morning because the last on state was at 60%
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I'v considered rule. I managed to turn off the light when brightness was 1% but (and maybe i didn't look hard enough) i didn't find any way to turn it back on. Plus it was jenky and i would like to keep rule space for other stuff.

**Additional context**  
_Add any other context or screenshots about the feature request here._
I try to look how to do it myself, but big project like this one overwhelm me.
i think it could be trivial to add some extra condition to the RotaryHandler in support_rotary.ino, but i didn't find something as simple as the "turn light on" and "turn light off" function to call :(

I'm willing to help with debug and test as much as possible if someone want to try to do this without the hardware in hand :)
**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-05-20:
I will take a look if it's a small patch

s-hadinger, 2020-06-15:
I still need to find time to add this one. It should be easy but I can't test it.

poofyteddy, 2020-06-15:
> but I can't test it.

Hi,
Thank's for still thinking about this. I've try again but can't find the correct way to turn on a light ;(
Once again, i'm fully available to test this, as many time as needed. I need this feature in order to flash my second light, who is actually the one on my side of the bed ;D

poofyteddy, 2020-07-12:
they have been some huge work on the support_rotary.ino recently.
I'm exited, but i'm also more lost than ever now that the code have became more complex.
Thank you for assigning this issue to yourself arendst. I am not capable to do this :(

closed 2020-07-12 by ascillato2

poofyteddy, 2020-07-13:
Thankyou for this,
I am doing some power wiring in the house right now and i take me all of my time. As soon as i'm done with that i'll upgrade my light to test.

poofyteddy, 2020-10-25:
I am sorry to come back to this month after it's closure, but i didn't had enough time before.
I the change that have been made will turn on the dimmer at the last known %.
What i was looking for was for it to turn on a 1%

I was digging into the code, but only found option 77 to disable turning off the light, i didn't find where the code turned on the light if the rotary was increased but the light was off.

Could you please point me to where this is done in the code ?
I am guessing i can quickly make a PR with a if and a new option.

arendst, 2020-10-25:
If you looked in the rotary code you'll see the only command send on a rotary turn is a ``LightDimmerOffset``. if the light is off it will send an offset related to the last dimmer state as with any normal physical dimmer or dimmer command.

If you want differen behaviour I suggest you start digging there (line 201) and find other functions to be available from the light library (xdrv_04_light.ini). 

I suspect you want something like this:
```
test if light is off and if so and dial is turned force a dimmer 1 call else execute current ``LightDimmerOffset`` call
```

arendst, 2020-10-25:
A fix is coming up shortly.

poofyteddy, 2020-10-25:
Damn you !! I both love you very much for this crazy quick add-on, but i didn't want to bother you with something i could have done ... especially on a Sunday.

Well thank you very much i will now read how you did it, and what function from the xdrv_light you used :)

poofyteddy, 2020-10-25:
It work very well :heart: 
I am not quite sure what does `LightGetDimmer` do because it' call getDimmer, and it look like it return a new "empty" dimmer ?
Then you add a +1 for the 1%

Anyway, i will now flash solder and flash my second lamp since you've made Tasmota the best firmware for it !

arendst, 2020-10-25:
``LightGetDimmer`` return the current dimmer value. To make the dimmer set to just 1 you'll need to substract the curent dimmer value with the current value (=0) and add just 1. Sound gibberish and it is.

poofyteddy, 2020-10-25:
Ho ! i missed the - in front of the function ! ok got it !
Thank you again