arendst/Tasmota #8266: User-configurable number of pulsetimers to use with PCF8574 chip.

gufonero, 2020-04-23:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
The number of Pulsetimers in Tasmota 8.2 is limited to a hard-coded value of 8, defined by the MAX_PULSETIMERS constant in the tasmota.h file. I need to increase it in order to control 16 relays driven by two PCF8574 IO expanders. The pulsetime functionality is used as a watchdog safeguard, to prevent the relays staying in the ON state indefinitely in case the MQTT OFF message does not reach the ESP due to WiFi failure.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
A user-friendly way to increase the number of pulsetimers. The concrete proposal is a MAX_PULSETIMERS preprocessor variable that can be defined in user_config_override.h

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Tried to compile a custom version changing the value of the MAX_PULSETIMERS constant in the tasmota.h file to 16, and changing also the  CFG_HOLDER value. The ESP starts right after flashing, but on reboot it is unable to connect to the WiFi network. 

This looks like a bug (see more details below), but it also may be that there more changes needed to increase the number of pulsetimers. I have found no documentation on this topic and I would appreciate any directions to make it work. 

In any case I believe that the proposed change would be useful for people needing to drive more than 8 relays with a single ESP.

**Additional context**  
_Add any other context or screenshots about the feature request here._

Details on the functioning of the customized 8.2 version built with MAX_PULSETIMERS=16.

1. The firmware starts correctly after erasing / flashing, and both the console and the web UI are available.

2. The decode-config utility, when run right after this first boot, reports a length error (4112 = 4096 + 16). 

```
ERROR 4 (@2955): Number of bytes read does ot match - read 4112, expected 4096 byte
```

3. After rebooting (from the web UI) the serial interface shows that the ESP tries to connect to WiFi with wrong settings (garbage SSID name and mode). The serial interface shows the following:

```
00:00:00 CFG: Loaded from flash at FB, Count 1
00:00:00 SRC: Restart
00:00:00 Project tasmota  Version 8.2.0(tasmota)-STAGE
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP118  in mode 11N as ...
00:00:01 CFG: Saved to flash at FA, Count 3, Bytes 4112
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
```

My understanding is that increasing the number of pulsetimers makes the config data structure 16 bytes bigger, and I suspect that the firmware is mixing up both versions, but have not been able to understand what is happening precisely.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-24:
You are right, change to 16 does corrupt settings area.  Compability breaking change needed

Jason2866, 2020-04-24:
Theo is doing a lot of changes under the hood at the moment. Stay tuned.

gufonero, 2020-04-24:
@Jason2866 Thank you very much for the quick answer. Just a clarification: the number of relays is not the problem; the current version works fine with 16 relays. The problem is related only to the pulsetime functionality.

Glad to know that there will be a solution, even if it needs a hardware upgrade to ESP32. Will try to get a couple of them. Any hint on the availability time?

In the meantime, can you suggest any workaround for the intended watchdog functionality, or how to compile a custom version that works, even if compatibility is broken?

Jason2866, 2020-04-24:
Not easy, since settings are stored in a restricted area where no place is left.
If you change the nukber of relais (= size), something goes over the border.
Long story short answer. Not possible without great changes in code.

gufonero, 2020-04-25:
@Jason2866 Thank you for the feedback. This was a helpful hint; I had not realized that the reason for the settings garbling was lack of room in the settings area.

Based on your remark, I was able to successfully build a custom version with 16 pulsetimers. I made room for them by disabling Domoticz support (which I don't use) and removing from the SYSCFG structure some unneded Domoticz-related variables. This involved only minimal changes to the tasmota.h and the settings.h files (as well as defining a specific CFG_HOLDER value).

I know that modifying files that are not designed for user configuration is not an ideal solution, but considering that this is only needed for one specific relay controller and that the changes are clean and very localized, I think it is the best alternative for my needs. 

Thank you for your help. Shall I close the issue? I think the requested functionality can still be useful for other people, even if it cannot be addressed until some future release.

Jason2866, 2020-04-25:
Great you found a solution for your needs. The beauty of Open Source software.
Please close. Your solution still can be found with a search in issues.

closed 2020-04-25 by ascillato2

fflnvb, 2022-02-19:
@gufonero I'm glad to stumble upon your issue and would like to know if you can share the changes inside settings.h and tasmota.h. I tried to simply reduce free_f61 space in order to compile it properly, but didn't get it to work by now.

gufonero, 2022-02-25:
@fflnvb Sure I'm happy to share the changes I made, but please note that they are based on Tasmota v8.2, later upgraded to v8.5.1. They might not be appropriate for later versions because if I remember correctly there were some changes in the config data area layout after Tasmota v.9. 

The general idea is to increase the number of pulsetimers defined by MAX_PULSETIMERS from 8 to 16. For this to work you need to remove something else that takes the same size from the config data area so that its size stays the same. I decided to remove some Domoticz-related data since I don't use Domoticz. All of this is controlled by two #define variables USE_CUSTOM_MULTI_PCF8574=1 and USER_MAX_PULSETIMERS=16 which are defined in an ad-hoc environment inside platformio_override.ini.

You also need to change the value of the CFG_HOLDER variable to something different from the default.

Here are the relevant parts of the setup. 

### settings.h

Disable the domoticz data areas as follows

``` 
uint16_t      domoticz_update_timer;     // 340
uint16_t      pwm_range;                 // 342
#ifndef USE_CUSTOM_MULTI_PCF8574            // Default Tasmota configuration
unsigned long domoticz_relay_idx[MAX_DOMOTICZ_IDX];  // 344
unsigned long domoticz_key_idx[MAX_DOMOTICZ_IDX];    // 354
#endif
unsigned long energy_power_calibration;  // 364
unsigned long energy_voltage_calibration;  // 368
unsigned long energy_current_calibration;  // 36C

``` 

```
char          serial_delimiter;          // 451
uint8_t       seriallog_level;           // 452
uint8_t       sleep;                     // 453
#ifndef USE_CUSTOM_MULTI_PCF8574            // Default Tasmota configuration
uint16_t      domoticz_switch_idx[MAX_DOMOTICZ_IDX];      // 454
uint16_t      domoticz_sensor_idx[MAX_DOMOTICZ_SNS_IDX];  // 45C
#endif
uint8_t       module;                    // 474
uint8_t       ws_color[4][3];            // 475
uint8_t       ws_width[3];               // 481
    
```

### tasmota.h

Disable the domoticz data areas as follows

```
const uint8_t MAX_PWMS = 5;                 // Max number of PWM channels
const uint8_t MAX_COUNTERS = 4;             // Max number of counter sensors
const uint8_t MAX_TIMERS = 16;              // Max number of Timers
#ifndef USE_CUSTOM_MULTI_PCF8574            // Original Tasmota configuration
#pragma message "Var USE_CUSTOM_MULTI_PCF8574 is not defined"
const uint8_t MAX_PULSETIMERS = 8;          // Max number of supported pulse timers
const uint8_t MAX_DOMOTICZ_IDX = 4;         // Max number of Domoticz device, key and switch indices
const uint8_t MAX_DOMOTICZ_SNS_IDX = 12;    // Max number of Domoticz sensors indices
#else                                       // Custom configuration (variable pulsetimers, no Doticz support)
#pragma message "Var USE_CUSTOM_MULTI_PCF8574 is defined"
const uint8_t MAX_PULSETIMERS = USER_MAX_PULSETIMERS;          // Max number of supported pulse timers
#endif
const uint8_t MAX_KNX_GA = 10;              // Max number of KNX Group Addresses to read that can be set
const uint8_t MAX_KNX_CB = 10;              // Max number of KNX Group Addresses to write that can be set
const uint8_t MAX_XNRG_DRIVERS = 32;        // Max number of allowed energy drivers
 
```

### platformio_override.ini

Create a new environment as follows

```
[env:tasmota-pcf8574]
build_flags                = ${common.build_flags}
                            -D USE_CUSTOM_MULTI_PCF8574=1
                            -D USER_MAX_PULSETIMERS=16
```

### user_config_override.h

Add the following to manage the built-in CFG_HOLDER and 

```
#undef  CFG_HOLDER
#ifdef USE_CUSTOM_MULTI_PCF8574    // Multiple PCF8574 IO expanders with pulsetime functionality (breaks SYSCFG compatibility)
#define CFG_HOLDER        8026     // [Reset 1] Change this value to load SECTION1 configuration parameters to flash
#define USE_PCF8574
//#define USER_MAX_PULSETIMERS 16
#undef USE_DOMOTICZ
#else                              // Default Tasmota configuration
#define CFG_HOLDER        8001     // [Reset 1] Change this value to load SECTION1 configuration parameters to flash
+#endif
```

Hope this helps. Please let me know if you need something else.

arendst, 2022-02-25:
I'll add support for 32 pulsetimers today.

gufonero, 2022-02-25:
@arendst Nice enhancement, Theo. Thank you very much!