arendst/Tasmota #8275: Better messages (or docs) for "MQT: Connect failed" error

fdcastel, 2020-04-25:
**Have you looked for this feature in other issues and in the docs?**  
Yes. It appears that this information used to be in https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview#status-codes ([source](https://github.com/arendst/Tasmota/issues/6428#issuecomment-531564560))

However, since the migration for the new documentation site, I cannot found it anymore. There is nothing in the [MQTT section](https://tasmota.github.io/docs/MQTT/) neither in the [MQTT commands](https://tasmota.github.io/docs/Commands/#mqtt) documentation.

**Is your feature request related to a problem? Please describe.**  
Yes. One of my devices is not connecting to MQTT with the error message

`"MQT: Connect failed to xxxx:1883, rc 5. Retry in 10 sec"`

being a novice user, I can just wonder what the problem is. Actually, it took some time until I noticed that "rc 5" was something informative.

**Describe the solution you'd like**  
A better error message describing the real problem, without cryptic codes. Like

`"MQT: Connect failed to xxxx:1883 (invalid password). Retry in 10 sec"`

Alternatively, a table in the documentation explaining what the codes mean would be an acceptable compromise :wink:.

**Describe alternatives you've considered**  
Looked at tasmota [source code](https://github.com/arendst/Tasmota/blob/9d838c001c5e6a9c40a470e7c6caec785827e6a2/tasmota/xdrv_02_mqtt.ino#L703). Found [this table](https://pubsubclient.knolleary.net/api.html#state).

Yes. It solved my problem. But now I'm worried that other users may lost the same amount of time I did looking for this info.

**Additional context**  
BTW: a lot of useful information was lost with the deactivation of the GitHub Wiki. I already crossed with several links like this one which now simply redirect to the main GitHub project page. Please consider reactivate the old wiki, even if in some kind of "read-only" mode.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-25:
You are invited to improve the documentation. It is Open Source too and users can contribute.
How is explained in the docu 😀

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-25 by ascillato2

device111, 2020-04-25:
I also don't like the output of error Number.
i think, we could do this in a better Log message with text. i will release a PR.

device111, 2020-04-25:
PR is Online: #8279

Jason2866, 2020-04-25:
Imho, Space waste of flash

ascillato, 2020-04-26:
@fdcastel 

Hi, please, help us and add this table to the docs. Thanks.

hheimbuerger, 2024-08-12:
Seems like there has been a regression? Now in 2024, I'm getting this again:

```
08:11:36.306 MQT: Connect failed to localhost:1883, rc -2. Retry in 60 sec
```

Using "Tasmota 14.1.0 (release-tasmota32)". But it's my very first time having any contact with Tasmota and I've done zero further research, so take with a grain of salt.

OP, thanks for linking to [the table](https://pubsubclient.knolleary.net/api.html#state), that provided me with what I needed.

fdcastel, 2024-08-13:
@hheimbuerger I don't think this is a regression. 

This issue was closed because the table containing the MQTT error codes was added to [the docs](https://tasmota.github.io/docs/MQTT/#return-codes-rc) (via #543).

It seems Tasmota developers chose not to change the error messages to reduce the firmware size (a sensible decision, IMHO).

hheimbuerger, 2024-08-13:
@fdcastel You're right, my bad. I saw the merged docs PR and thought it was a code commit.

I wish these ten strings would appear in the logs, but I cannot judge how tight your space limitations are.

I've resolved my issue. I was just confusing my own network architecture. 🙄 Thanks!