arendst/Tasmota #8276: I2C GPIO pins not working in tasmota_template.h

sebeksd, 2020-04-25:
### PROBLEM DESCRIPTION
I was configuring my own base configuration through "user_config_override.h" and all is working fine.
Because not all settings can be overridden with that file I was forced to edit "tasmota_template.h" (at least for now) to edit pins/gpio settings. After editing "WEMOS"/"Generic" my Buttons/LED/PWM are working fine and they disappear from "Module configuration" (WEB) and they are visible in "Template parameters" with correct values. Problem begin to show when I changed two additional GPIO to I2C. I2C pins are not working, they are not appearing in both WEB menus.
After issuing I2CScan I get "{"Command":"Error"}". After manually selecting correct pins I2C start working. I've tried to search for the issue but I'm not familiar with Tasmota enough to find where problem is. It looks like GPIO_I2C_SDA, GPIO_I2C_SCL are completely ignored in base configuration.
I've attached all necessary files (config override and template).
```
 { "Lamp",         // WEMOS - Any ESP8266/ESP8285 device like WeMos and NodeMCU hardware (ESP8266)
     GPIO_KEY1_NP,        // GPIO00 D3 Wemos Button Shield
     GPIO_USER,        // GPIO01 TX Serial RXD
     GPIO_USER,        // GPIO02 D4 Wemos DHT Shield
     GPIO_USER,        // GPIO03 RX Serial TXD and Optional sensor
     GPIO_PWM1,        // GPIO04 D2 Wemos I2C SDA
     GPIO_SWT1,        // GPIO05 D1 Wemos I2C SCL / Wemos Relay Shield (0 = Off, 1 = On) / Wemos WS2812B RGB led Shield
                       // GPIO06 (SD_CLK   Flash)
                       // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
                       // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     GPIO_USER,        // GPIO09 (SD_DATA2 Flash QIO or ESP8285)
     GPIO_USER,        // GPIO10 (SD_DATA3 Flash QIO or ESP8285)
                       // GPIO11 (SD_CMD   Flash)
     GPIO_I2C_SDA,        // GPIO12 D6
     GPIO_USER,        // GPIO13 D7
     GPIO_I2C_SCL,        // GPIO14 D5
     GPIO_USER,        // GPIO15 D8
     GPIO_LED1_INV,        // GPIO16 D0 Wemos Wake
     ADC0_USER         // ADC0 A0 Analog input
  }
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 8.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Visual Studio Code + PlatformIO
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:02:08 RSL: qiot/things/admin/stat/LampaPokoj/RESULT = {"NAME":"Lamp","GPIO":[90,255,255,255,37,9,255,255,255,255,255,255,56],"FLAG":15,"BASE":18}
20:02:08 RSL: qiot/things/admin/stat/LampaPokoj/RESULT = {"Module":{"18":"Lamp"}}
20:02:08 RSL: qiot/things/admin/stat/LampaPokoj/RESULT = {"GPIO0":{"90":"Button1n"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"9":"Switch1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"56":"Led1i"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:51:30 RSL: qiot/things/admin/stat/LampaPokoj/STATUS = {"Status":{"Module":18,"FriendlyName":["Lampa pokoj"],"Topic":"LampaPokoj","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Set template GPIO to I2C

### EXPECTED BEHAVIOUR
1. GPIO set to I2C appear in "Template parameters" as I2C, 
2. GPIO used for I2C dissapear from "Module parameters", 
3. I2C working 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-25:
No need for using config_override.h or changing tasmota_template.h.
You can define all GPIOs you need via the Template function.
See https://tasmota.github.io/docs/Templates/

sebeksd, 2020-04-25:
Yes I know, I'm want to include everything in single bin file. For now I'm mostly experimenting so there is a chance I will change mind mind and use different route.
Right now my goal is to create gti branch/stash/patch/commit with full configuration of every Tasmota device I have, so in the end if I need to build/reconfigure something I will have it in one place. For example Using WEBUI or/and commands I need to remember to use few SetOptionsX, Templates, Pins and so on. Of course I can just backup my current configuration which is fine (in 90% cases) but like I said I'm experimenting and I like to have everything in one place, human readable (source files instead of config backup), and have a diff (git) of my changes done in the past.  

I'm aware that using "tasmota_template.h" is not "advertised" method but I created this ticket because this issue could potentially be something "larger" and reappear in future in more "correct" situation.

ascillato2, 2020-04-25:
> After issuing I2CScan I get "{"Command":"Error"}".

That means that I2C was not enabled in your build.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-25 by ascillato2

sebeksd, 2020-04-25:
"#define USE_I2C" is in my_user_config.h and in user_config_override.h.
Code "Wire.begi.." in "void GpioInit(void)" is called. Also disabling USE_I2C mean that configuring I2C GPIO through WEB would not work but it do.

I will try to use Chat like you mentioned but tomorrow. Thanks.

sebeksd, 2020-05-02:
For some reason It start working as I was hoping, I2C pins are loaded from template I've changed.
For why it wasn't working in the first place I can only guess that it was related to not erasing flash before upload, I'm new to PlatformIO and I wasn't paying attention to If erase flash on upload is done by default or not. 

To summaries: Changing "tasmota_template.h" can be used to configure GPIO in firmware.

ascillato, 2020-06-24:
Another new method is as https://github.com/arendst/Tasmota/pull/8766

Just uncomment line 56 of file my_user_config.h and set there the module template you want