arendst/Tasmota #8282: Infrared sensing and receiving

betonishard, 2020-04-26:
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am on latest version v8.2.0
When I use tasmora-ir.bin my infrared receiver and sender always show in the console always the correct protocol e.g. "PROTOCOL:SAMSUNG" when I use the default tasmota.bin I always get  "PROTOCOL: UNKNOWN". I can not exactly recall from which version on this was not working, however it did work. 
I don't have enough knowledge about this to say whether it is a bug, hence my question. I would suspect that there is a reason why tasmota-ir.bin is brought to life. However I do think that in this case SAMSUNG is quite a "regular" device.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Example: Unknown
00:00:42 MQT: xxx/tele/RESULT = {"Time":"1970-01-01T00:00:42","IrReceived":{"Protocol":"UNKNOWN","Bits":34,"Hash":"0x68733A46"}}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Install tasmota.bin, hook up a receiver and check console. Now install tasmota-ir.bin and it works.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-26:
Samsung Protocol is not included in tasmota.bin.
You have to use tasmota-ir

betonishard, 2020-04-26:
@Jason2866 , thanks for your quick response. Do you happen to know since when this has been done like this? So which version?

Maybe an idea would be to have this in a SetOption. 

Thanks.!!

Jason2866, 2020-04-26:
Dont remember, was a change (update) in the used IR library some month ago.
No, a SetOption is not possible, since every protocol needs flash space, thats why tasmota-ir exists

arendst, 2020-04-26:
See release notes https://github.com/arendst/Tasmota/releases/tag/v7.1.0

closed 2020-04-26 by ascillato2