arendst/Tasmota #8286: App for controlling sonoff devices from remote

BB1982, 2020-04-26:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  
Hi everyone
There are tons of valid reasons to flash Tasmota on Sonoff devices, but missing the possibility to operate a switch (or other features) while far from home network is definitely a big issue. Is there any way to access the web interface from remote or to use an app to send commans/receive status leveraging, as an example, MQTT? Obviously there are solutions like port forwarding (dangerous!) or personal hubs running on devices like Raspberry, but in both cases you introduce complications.
In my case, I was planning to use a cloud broker (MyQttHub) and I'd find quite convenient if an app could be designed to operate directly with such broker.
Regards
Bruno

Jason2866, 2020-04-26:
There is no Open Source App for Tasmota. There is no easy secure way to control Tasmota
out of its LAN. It is designed to be used with a local mqtt.
There is the supported possibility to connect secured to AWS. 
If you plan to build a APP, doing via AWS IoT is a way

schrej-zz, 2020-04-26:
Well,
why not connect to local net via VPN?
Joerg

closed 2020-04-26 by ascillato2