arendst/Tasmota #8287: backlog + interlock not working as expected

issacg, 2020-04-26:
### PROBLEM DESCRIPTION
Interlock and backlog appear not to be working together.  When running the following command `Backlog Interlock 1,2; Interlock ON` I see the following output:
```
17:50:03 CMD: Backlog Interlock 1,2; Interlock ON
17:50:03 MQT: stat/mcutest/RESULT = {"Interlock":"OFF"}
17:50:03 MQT: stat/mcutest/RESULT = {"Interlock":"OFF"}
```

Further probing confirms...
```
17:52:45 CMD: interlock
17:52:45 MQT: stat/mcutest/RESULT = {"Interlock":"OFF"}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Tested on Shelly 2.5 and on generic NodeMCU
- [X] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Platform.io
- [ ] Flashing tools used: _____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255` (taken from the Shelly device): 
```
17:55:17 MQT: stat/SHUTTER_LR/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
17:55:17 MQT: stat/SHUTTER_LR/RESULT = {"Module":{"0":"Shelly 2.5"}}
17:55:18 MQT: stat/SHUTTER_LR/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relay2"},"GPIO16":{"156":"ADE7953 IRQ"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The backlog commands were taken from [the docs site](https://tasmota.github.io/docs/Blinds-and-Shutters/#shutter-modes) - my assumption had been that it would work as expected.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-26:
Which Tasmota version are you using?

issacg, 2020-04-26:
Marking as closed.  Further testing shows that the above may not be completely correct.  I'll continue investigating and re-open if needed.

issacg, 2020-04-26:
Re-opening this to let other maintainers decide if this is actually a bug or not.

Further testing of the above led me to discover inconsistencies, as I implied in the previous comment.  What's actually happening is that in my "real world scenario" I was sending one long backlog which included both the module/template setup and then, later in the backlog, the interlock commands.

What was happening is that the interlock 1,2 was being saved, but the `interlock on` was being ignored because the default module (sonoff basic) only had a single relay.

It would be really nifty if the backlog allowed for a command to explicitly allow rebooting in the middle on condition that it would continue the backlog after trying to re-connect (regardless of whether that succeeds, in case the current config state disallowed that).  But that's probably a feature request, and not a bug.

Can a more seasoned maintainer confirm and/or give a second opinion, please?

arendst, 2020-04-26:
Regarding a reboot in the middle of a backlog that was removed over 18 month ago and replaced by a hold until all commands in the backlog have been executed.

Impleneting your wish would be a major task as the backlog needs to be stored in between the restart.

Considering to low volume of use of a lot of code to handle this I'll have to conclude it is as designed and won't be implemented any time son.

You'll have to redefine your backlog use.

ascillato2, 2020-04-26:
Docs corrected.

Thanks for reporting.

closed 2020-04-26 by ascillato2

issacg, 2020-04-29:
@arendst when looking for an alternate solution to this, I thought of setting one time rules to run at `ON System#Boot` sort of like this:
```
BACKLOG (do other stuff); Interlock 1,2; RULE1 ON System#Boot DO BACKLOG Interlock On; (do other stuff?); Rule1 '' ENDON; Restart 1
```
But then I ran into #3785 which seems not to have a good solution...  Wondering if this would be a good reason to allow for some way to embed nested backlog commands (which would be a new issue) or if it's not a good line of thinking to even explore?

Thanks for your input!