arendst/Tasmota #8290: SwitchX#state where X>1 always return %value% = 2

mvincm, 2020-04-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When you use more than one Switch the returned value is always "2". Does not matter if the real state is HIGH or LOW. Switch1 works just fine. "Status 0" also shows the correct state `{"StatusSNS":{"Time":"2020-04-26T23:45:59","Switch1":"ON","Switch6":"ON","DS18B20-1":{"Id":"03049779D469","Temperature":19.1},"DS18B20-2":{"Id":"03049779D518","Temperature":19.1},"TempUnit":"C"}}`. Talking about Switch6 in my case.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 NodeMCU v3
- [ ] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
23:45:01 RSL: stat/tasmota_D71B5C/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:45:01 RSL: stat/tasmota_D71B5C/RESULT = {"Module":{"0":"Generic"}}
23:45:01 RSL: stat/tasmota_D71B5C/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"14":"Switch6"},"GPIO13":{"9":"Switch1"},"GPIO14":{"0":"None"},"GPIO15":{"4":"DS18x20"},"GPIO16":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
23:47:31 RSL: stat/tasmota_D71B5C/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":476,"Rules":"on Switch6#state=0 do var1 20 endon"}
23:47:31 RSL: stat/tasmota_D71B5C/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
23:47:31 RSL: stat/tasmota_D71B5C/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```
23:48:04 RSL: stat/tasmota_D71B5C/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_D71B5C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"test_top","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:52","StartupUTC":"2020-04-26T22:12:12","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-04-26T15:00:23","SaveCount":136,"SaveAddress":"F4000"}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"362/699"}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NewOne24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS5 = {"StatusNET":{"Hostname":"esp8266-tasmota","IPAddress":"192.168.24.132","Gateway":"192.168.24.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.24.31","Mac":"5C:CF:7F:D7:1B:5C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D71B5C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS7 = {"StatusTIM":{"UTC":"2020-04-26T22:48:05","Local":"2020-04-26T23:48:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:37","Sunset":"19:58"}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS10 = {"StatusSNS":{"Time":"2020-04-26T23:48:05","Switch1":"ON","Switch6":"ON","DS18B20-1":{"Id":"03049779D469","Temperature":19.1},"DS18B20-2":{"Id":"03049779D518","Temperature":19.1},"TempUnit":"C"}}
23:48:05 RSL: stat/tasmota_D71B5C/STATUS11 = {"StatusSTS":{"Time":"2020-04-26T23:48:05","Uptime":"0T00:35:53","UptimeSec":2153,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"NewOne24","BSSId":"EC:1A:59:83:82:58","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
23:51:02 CMD: Rule2 1
23:51:02 RSL: stat/tasmota_D71B5C/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":445,"Rules":"on Switch6#state do Publish cmnd/custom-topic/SWITCH %value% endon"}
23:51:10 RUL: SWITCH6#STATE performs "Publish cmnd/custom-topic/SWITCH 2"
23:51:10 RSL: cmnd/custom-topic/SWITCH = 2
```

### TO REPRODUCE
Just configure the template to use more than one switch.

### EXPECTED BEHAVIOUR
%value% should be "0" or "1" not "2"

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-04-26:
Hi,

The value 2 means TOGGLE and that output will depend on which `switchmode` did you use.

If you want to be "1" or "0" depending on the physical state of the switch number 6, you need to write in the Tasmota console: `switchmode6 1`

Available Switchmodes from the [docs at commands](https://tasmota.github.io/docs/Commands/):

[image]

More info on this topic: https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode

Closing as it is not a Tasmota Bug.

Please, if you need further assistance on Tasmota usage, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-04-26 by ascillato2

mvincm, 2020-04-27:
face palm.... Sorry for that... 

Thx!