arendst/Tasmota #8293: script syntax is not explained fully in the docs

prabbit237, 2020-04-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The page at https://tasmota.github.io/docs/Scripting-Language/ shows \ being used in many of the examples but, unless I'm just missing it, nowhere does it show when the \ is needed or exactly what it does. Also it's using other characters, such as &gt;, in non-obvious ways.

There are places where it refers to \n, \r & \t and I know what those are. But it also shows in the examples something like:
````
if temp\30  
and hum\70  
then =>print damn hot!  
endif
````
Ok, it looks like the \ is the same as &gt; (noting that &gt; isn't listed in the comparison operators.) But then it shows:
````
Console Commands

script <n \<n: 0 = switch script off; 1 = switch script on
script <cmdline execute \<cmdline
````
Ok, so it's used as the normal escape character? Or was it meant to be an escape in the docs whre it should be displayed as "execute <cmdline"? (Several other places look like it's being used that way, such as "=> <command Execute \<command recursion disabled"

But then we find the following example:
````
if upsecs%5==0  
then =\print %upsecs%  (every 5 seconds)  
endif

; not recommended for reliable timers  
timer+=1  
if timer\=5  
then =>print 5 seconds over (may be)  
timer=0  
endif
````
Ok, the = starts the command. But what is the \ for in HERE? It's not a substitute for &gt; and it's not escaping the "p" so what **IS** it doing and why does it need to be there? We also see it in the "if timer\=5 " so, does this mean ">="? And then there's the "then =>print" line. Why does the first part use "then =>print" but the second part uses "then =>print"???????????

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [ ] Tasmota binary firmware version number used: 8.2.0.5
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: platformio
- [x] Flashing tools used: platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
[13:14:59] stat/tasmota_CB0B3C/RESULT {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
[13:14:59] stat/tasmota_CB0B3C/RESULT {"Module":{"18":"Generic"}}
[13:14:59] stat/tasmota_CB0B3C/RESULT {"GPIO0":{"5":"I2C SCL"},"GPIO1":{"0":"None"},"GPIO2":{"6":"I2C SDA"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"4":"DS18x20"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
[13:15:33] stat/tasmota_CB0B3C/STATUS {"Status":{"Module":18,"FriendlyName":["mikes thermostat"],"Topic":"tasmota_CB0B3C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
[13:15:33] stat/tasmota_CB0B3C/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:38:50","StartupUTC":"2020-04-27T16:36:43","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2020-04-27T12:30:10","SaveCount":48,"SaveAddress":"FB000"}}
[13:15:33] stat/tasmota_CB0B3C/STATUS2 {"StatusFWR":{"Version":"8.2.0.5(tasmota)","BuildDateTime":"2020-04-27T12:02:15","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"372/699"}}
[13:15:33] stat/tasmota_CB0B3C/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["01009109","2805C8000100060000005A00000000000000","00000000","00000000"]}}
[13:15:33] stat/tasmota_CB0B3C/STATUS4 {"StatusMEM":{"ProgramSize":571,"Free":432,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","01E087B6","04180121","00000045","000000C0","00000000","00000000"],"Drivers":"1,2,3,4,8,9,10,12,13,20,21","Sensors":"1,2,3,5"}}
[13:15:33] stat/tasmota_CB0B3C/STATUS5 {"StatusNET":{"Hostname":"tasmota_CB0B3C-2876","IPAddress":"192.168.1.145","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:CB:0B:3C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
[13:15:33] stat/tasmota_CB0B3C/STATUS6 {"StatusMQT":{"MqttHost":"192.168.1.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CB0B3C","MqttUser":"mike","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
[13:15:33] stat/tasmota_CB0B3C/STATUS7 {"StatusTIM":{"UTC":"2020-04-27T17:15:34","Local":"2020-04-27T18:15:34","StartDST":"2020-03-08T02:00:00","EndDST":"2020-11-01T03:00:00","Timezone":"+01:00"}}
[13:15:33] stat/tasmota_CB0B3C/STATUS10 {"StatusSNS":{"Time":"2020-04-27T18:15:34","DS18B20":{"Id":"031197799755","Temperature":70.0},"TempUnit":"F"}}
[13:15:34] stat/tasmota_CB0B3C/STATUS11 {"StatusSTS":{"Time":"2020-04-27T18:15:34","Uptime":"0T00:38:51","UptimeSec":2331,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT","BSSId":"7E:D2:94:56:B1:F1","Channel":11,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

kugelkopf123, 2020-04-27:
I think it is a problem sinds the Update to the new doc system. There are many missing “>” as far as I know. It need to be checked and corrected.

gemu2015, 2020-04-27:
in the meantime i went through and corrected many formatting issues which occurred when the doc system changed though i may still have overlooked some.

ascillato2, 2020-04-27:
@prabbit237 Thanks for Reporting.
@gemu2015 Thanks for fixing.

closed 2020-04-27 by ascillato2

prabbit237, 2020-04-28:
Ok, now it's pretty much consistent and using the structure I would have assumed. Thanx. But under the comparison operators you have:

````
Comparison operators ==, !=, `, =, <, <=` 
````
I assume >= and > should be in there and the ` marks are in error? And based on what you were saying about it missing some > symbols, I think the following two spots need to be changed

````
Console Commands

script <n \<n: 0 = switch script off; 1 = switch script on
script <cmdline execute \<cmdline
````

should be:

````
Console Commands

script <n> <n>: 0 = switch script off; 1 = switch script on
script <cmdline> execute <cmdline>
````

====================

````
Commands
=> <command Execute \<command recursion disabled
+> <command Execute \<command recursion enabled
-> <command Execute \<command - do not send MQTT or log messages (i.e., silent execute - useful to reduce traffic)
````

should be:

````
Commands
=> <command> Execute <command> recursion disabled
+> <command> Execute <command> recursion enabled
-> <command> Execute <command> - do not send MQTT or log messages (i.e., silent execute - useful to reduce traffic)
````

kugelkopf123, 2020-04-28:
Please help us to make it better. Simply click on the Pen Icon on top of the wiki page to edit it and start a PullRequest. =)