arendst/Tasmota #8295: AM2301 not working but seems to work as SI7021

Siggemada, 2020-04-27:
### PROBLEM DESCRIPTION
AM2301 reports null as sensor value. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic):TH16
- [x ] Tasmota binary firmware version number used: ___8.2__
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: apt-get esptool on ubuntu 18

19:29:44 CMD: Backlog Template; Module; GPIO 255
19:29:44 RSL: stat/tasmota_510196/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:29:44 RSL: stat/tasmota_510196/RESULT = {"Module":{"4":"Sonoff TH"}}
19:29:44 RSL: stat/tasmota_510196/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"2":"AM2301"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}04
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

19:29:44 CMD: Backlog Template; Module; GPIO 255
19:29:44 RSL: stat/tasmota_510196/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:29:44 RSL: stat/tasmota_510196/RESULT = {"Module":{"4":"Sonoff TH"}}
19:29:44 RSL: stat/tasmota_510196/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"2":"AM2301"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}04
```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
19:31:22 CMD: Backlog Rule1; Rule2; Rule3
19:31:23 RSL: stat/tasmota_510196/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
19:31:23 RSL: stat/tasmota_510196/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
19:31:23 RSL: stat/tasmota_510196/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:31:50 CMD: Status 0
19:31:50 RSL: stat/tasmota_510196/STATUS = {"Status":{"Module":4,"FriendlyName":["Tasmota"],"Topic":"tasmota_510196","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:31:50 RSL: stat/tasmota_510196/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:48","StartupUTC":"2020-04-27T18:25:02","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2020-04-26T12:59:55","SaveCount":34,"SaveAddress":"FA000"}}
19:31:50 RSL: stat/tasmota_510196/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"335/699"}}
19:31:50 RSL: stat/tasmota_510196/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Q","W"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:31:50 RSL: stat/tasmota_510196/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
19:31:50 RSL: stat/tasmota_510196/STATUS5 = {"StatusNET":{"Hostname":"tasmota_510196-0406","IPAddress":"E","Gateway":"R","Subnetmask":"T","DNSServer":"Y","Mac":"1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:31:51 RSL: stat/tasmota_510196/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_510196","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:31:51 RSL: stat/tasmota_510196/STATUS7 = {"StatusTIM":{"UTC":"2020-04-27T18:31:51","Local":"2020-04-27T19:31:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:35","Sunset":"20:00"}}
19:31:51 RSL: stat/tasmota_510196/STATUS10 = {"StatusSNS":{"Time":"2020-04-27T19:31:51","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}}
19:31:51 RSL: stat/tasmota_510196/STATUS11 = {"StatusSTS":{"Time":"2020-04-27T19:31:51","Uptime":"0T00:06:49","UptimeSec":409,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Q","BSSId":"W","Channel":8,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
19:35:14 CFG: CR 335/699
19:35:14 LOG: SerialLog 2, WebLog 4, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 300
19:35:14 HTP: Configuration
19:35:14 DHT: Timeout waiting for start signal low pulse
19:35:15 CFG: Saved to flash at F9, Count 35, Bytes 4096
19:35:16 RSL: tele/tasmota_510196/STATE = {"Time":"2020-04-27T19:35:16","Uptime":"0T00:10:14","UptimeSec":614,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Q","BSSId":"W","Channel":8,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}
19:35:16 RSL: tele/tasmota_510196/SENSOR = {"Time":"2020-04-27T19:35:16","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":0.0},"TempUnit":"C"}
19:35:16 DHT: Timeout waiting for start signal low pulse
19:35:17 HTP: Main Menu
19:35:18 DHT: Timeout waiting for start signal low pulse
19:35:18 HTP: Console
19:35:20 DHT: Timeout waiting for start signal low pulse
19:35:22 DHT: Timeout waiting for start signal low pulse
19:35:24 DHT: Timeout waiting for start signal low pulse

```

### TO REPRODUCE
Configure AM2301 sensor as AM2301 sensor

### EXPECTED BEHAVIOUR
AM2301 sensor configured as AM2301 sensor should give values not null.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I have just received 5 Sonoff TH16 with AM2301 sensors. At first I thought it was a wiring problem but all 5 combined with 5 different AM2301 sensors all gave null values.

So I tried to configure as SI7021, and that seems to work. The sensors are clearly marked as AS2301 both on the package and on the sensor shell itself. All Sonoff branded.

Console output here:
00:00:00 CFG: Loaded from flash at F8, Count 36
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 335/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Tasmota Version 8.2.0(tasmota)-STAGE
00:00:00 DHT: (v5) Sensors found 1
....
19:37:30 NTP: UTC 2020-04-27T18:37:29, DST 2020-03-29T02:00:00, STD 2020-10-25T03:00:00
19:37:30 RSL: tele/tasmota_510196/INFO1 = {"Module":"Sonoff TH","Version":"8.2.0(tasmota)","FallbackTopic":"cmnd/DVES_510196_fb/","GroupTopic":"cmnd/tasmotas/"}
19:37:30 RSL: tele/tasmota_510196/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_510196-0406","IPAddress":"Q"}
19:37:30 RSL: tele/tasmota_510196/INFO3 = {"RestartReason":"Software/System restart"}
19:37:30 RSL: stat/tasmota_510196/RESULT = {"POWER":"OFF"}
19:37:30 RSL: stat/tasmota_510196/POWER = OFF
19:37:31 APP: Boot Count 17
19:37:31 CFG: Saved to flash at F7, Count 37, Bytes 4096
19:37:34 RSL: tele/tasmota_510196/STATE = {"Time":"2020-04-27T19:37:34","Uptime":"0T00:00:13","UptimeSec":13,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"W","BSSId":"E","Channel":8,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:07"}}
19:37:34 RSL: tele/tasmota_510196/SENSOR = {"Time":"2020-04-27T19:37:34","SI7021":{"Temperature":21.8,"Humidity":24.3,"DewPoint":0.5},"TempUnit":"C"}
19:37:49 WIF: Checking connection...

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-04-27:
Yes, sonoff uses special SI7021 sensors labeled as AS2301
See https://tasmota.github.io/docs/Components/
Behaviour as designed

closed 2020-04-27 by ascillato2

inv1sible, 2021-02-06:
Thanks for this issue. Got a AM2301 labeled sensor that is working, when configured as SI7021 otherwise only null