arendst/Tasmota #8297: SetOption11 not working for SwitchMode 8

jensuffhaus, 2020-04-28:
### PROBLEM DESCRIPTION
I am trying to use SetOption11 to swap the primary/secondary functions on my SonoffMini so that the normal single toggle send the mqtt message and the actual double toggle, toggles the relay. I have SwitchMode 8 configured as my physical button is a toggle switch and not a momentary button. SwitchMode11 has no effect at all.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Mini
- [x] Tasmota binary firmware version number used: 8.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA TasmoAdmin
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:29:04 MQT: SM836101C514/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:29:05 MQT: SM836101C514/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
12:29:05 MQT: SM836101C514/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"9":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
12:30:46 MQT: SM836101C514/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["SM836101C514"],"Topic":"SM836101C514","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"0000","SaveData":1,"SaveState":1,"SwitchTopic":"SM836101C514","SwitchMode":[8,8,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:30:46 MQT: SM836101C514/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.0.40:9541/data/firmwares/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:24:33","StartupUTC":"2020-04-28T10:06:13","Sleep":50,"CfgHolder":4617,"BootCount":89,"BCResetTime":"2020-04-28T10:59:21","SaveCount":909,"SaveAddress":"F5000"}}
12:30:46 MQT: SM836101C514/stat/STATUS2 = {"StatusFWR":{"Version":"8.2.0(tasmota)","BuildDateTime":"2020-03-20T14:45:23","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"378/699"}}
12:30:46 MQT: SM836101C514/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UFFHAUS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["C00A8809","0505C8000100060000005A00000000000000","00000200","00000000"]}}
12:30:46 MQT: SM836101C514/stat/STATUS4 = {"StatusMEM":{"ProgramSize":577,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
12:30:46 MQT: SM836101C514/stat/STATUS5 = {"StatusNET":{"Hostname":"SM836101C514-3698","IPAddress":"192.168.0.102","Gateway":"192.168.0.100","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"D8:F1:5B:DF:CE:72","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:30:46 MQT: SM836101C514/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DFCE72","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:30:46 MQT: SM836101C514/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-04-28T11:30:46","Local":"2020-04-28T12:30:46","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:33","Sunset":"20:01"}}
12:30:46 MQT: SM836101C514/stat/STATUS10 = {"StatusSNS":{"Time":"2020-04-28T12:30:46","Switch1":"OFF"}}
12:30:46 MQT: SM836101C514/stat/STATUS11 = {"StatusSTS":{"Time":"2020-04-28T12:30:46","Uptime":"0T01:24:33","UptimeSec":5073,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"*****","BSSId":"CC:2D:E0:62:47:0E","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Configure "switchmode1 8" and "switchtopic1 1", now a single toggle will toggle the relay and a double toggle will issue an mqtt command. Now set "setoption11 1" and try again. Same behavior.

### EXPECTED BEHAVIOUR
SetOption11 is expected to swap the physical triggers and resulting actions but no change occurs.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
It is presumed and highly likely that the same bug applies to switchmode 9 and 10 as these were also added separately not so long ago at the same time as 8.

**(Please, remember to close the issue when the problem has been addressed)**

jensuffhaus, 2020-04-28:
Just some more detail on how I am trying to use the feature (setoption11) ...

I plan to use mqtt to send a command to tasmota to change setoption11 either 1 or 0. This will be based on events/time/sensors in HASS. The idea is that during daytime the primary functions will be used more while at night the secondary functions will be more used. Using this option people will be able to just keep performing the same action day and night but the resulting response will be appropriate to the time of day rather than having to expect people to change what they do. 

If a user insists on doing the other function they can perfrom the "hold" action.

I know this can be achieved with rules and using mqtt for both and having everything go through HASS but this creates a point of total failure if WiFi or mqtt go down, Using setoption11 keeps the logic and control local and at worst will be stuck on the latest setting when things failed, but the user will still be able to use the switch and control both functions locally without mqtt.

effelle, 2020-04-28:
So you are trying to use a setoption for a button for `switch1`? 
If yes, are you aware `SetOption11` is for Buttons and not for switches? 
Set it as a button2 not switch1 and you will have what you need.

jensuffhaus, 2020-04-28:
Why would setoption11 not apply to switches considering that several switchmodes are meant to have a primary and secondary function (5,6,8,9,10). As I understand setoption11 it is to swap the primary and secondary function, and this holds true for switches as well so it should be applied.

Also, for latching switches using switchmode 8, 9 or 10 I cannot configure them as buttons because they would be always "long hold". this is why these switchmodes were created to allow a double toggle for a secondary function.

I do think setoption11 (or another option) should allow the primary and secondary function of a SWITCH to be swapped ...

Day: setoption11 0
- Primary Action (press, toggle) = Primary Function (toggle relay)
- Secondary Action (hold, double toggle) = Secondary function (mqtt)

Night: setoption11 1
- Primary Action (press, toggle) = Secondary function (mqtt)
- Secondary Action (hold, double toggle) = Primary Function (toggle relay)

ascillato2, 2020-04-30:
Seems like adding a setoption for this interchange of features for switches would make a very complicated solution.

I think that is better to solve your use case by the use of rules. 

Let me write an example

ascillato, 2020-04-30:
Set your GPIO as SWITCH1 and `SWITCHMODE1 8`

# VAR1

`VAR1` is going to be used as a 0 - DAY / 1 - NIGHT

* Day: `var1` = 0

Primary Action (press, toggle of the key) = Primary Function (toggle relay)
Secondary Action (hold, double toggle of the key) = Secondary function (mqtt)

* Night: `var1` = 1

Primary Action (press, toggle of key) = Secondary function (mqtt)
Secondary Action (hold, double toggle of the key) = Primary Function (toggle relay)

# TIMERS

* Set timer1 to trigger a RULE when it reaches SUNSET
* Set timer2 to trigger a RULE when it reaches SUNRISE

# RULES

* Rule1: when your device starts, these rules will check the time to discriminate DAY and NIGHT and put that into VAR1. Also this rule-set, contains the rules to change the state of VAR1 when the TIMERS 1 and 2 trigger at sunset and sunrise.
* Rule2: will be the fuctions of switch1

```lua
rule1 1
rule1 on switch1#boot do var1 0 endon on time#initialized do backlog event checksunrise=%time%;event checksunset=%time% endon on event#checksunrise<%sunrise% do var1 1 endon on event#checksunset>%sunset% do var1 1 endon on Clock#Timer=1 do var1 1 endon on Clock#Timer=2 do var1 0 endon

rule2 1
rule2 on switch1#state=2 do event function1=%var1% endon on event#function1=0 do power1 2 endon on event#function1=1 do publish cmnd/TOPIC/CUSTOM 1 endon on switch1#state=3 do event function2=%var1% endon on event#function2=1 do power1 2 endon on event#function2=0 do publish cmnd/TOPIC/CUSTOM 1 endon
```

closed 2020-04-30 by ascillato2

jensuffhaus, 2020-05-01:
@ascillato2 Thanks for the input and example. It makes perfect sense and I am beginning to better understand tasmota.

I have it mostly implemented but for some reason tasmota always sends the default hold message as well as the one sent in the rule, so it sends 2 mqtt messages. I tried "switchtopic 0" and "switchtopic 1" and it send both mqtt's for both cases. Is that intended?

I thought that the rule would stop the built-in message being sent. This is the only issue keeping this from functioning as the lights now flash on/off. 

I can change the topic and ignore the default message but was hoping to keep things standard to limit nuances in the configs.